abhiTronix/raspberry-pi-cross-compilers #1: build scripts / docker image ?

jcelerier, 2018-09-06:
Hello,
would you happen to have the build scripts that you used to build these compilers ? 

Thanks for the great work !

abhiTronix, 2018-09-06:
@jcelerier Thanks for your interest, I may work on the script by tomorrow. But _**Building Docker image may take some time**_. However, Any PRs are most welcomed! :)

**Also I will change source files in next update to some other website as I'm currently using github's LFS support bandwidth to upload files to github, whose limit is now over.**

jcelerier, 2018-09-06:
no problem, I have scripts almost ready but I miss the raspberry sysroot, that's how I stumbled on your projects.
Here's what I use to build gcc : 

```bash
BINUTILS=binutils-2.31.1
MPC=mpc-1.0.3
MPFR=mpfr-3.1.4
GMP=gmp-6.1.2
GCC=gcc-8.2.0
ISL=isl-0.18
CLOOG=cloog-0.18.4

wget -nv "http://isl.gforge.inria.fr/$ISL.tar.bz2"
wget -nv "ftp://gcc.gnu.org/pub/gcc/releases/$GCC/$GCC.tar.xz"
wget -nv "http://ftp.gnu.org/gnu/gmp/$GMP.tar.bz2"
wget -nv "ftp://ftp.gnu.org/gnu/mpc/$MPC.tar.gz"
wget -nv "ftp://gcc.gnu.org/pub/gcc/infrastructure/$MPFR.tar.bz2"
wget -nv "http://ftp.gnu.org/gnu/binutils/$BINUTILS.tar.bz2"
wget -nv "http://www.bastoul.net/cloog/pages/download/count.php3?url=./$CLOOG.tar.gz" -O $CLOOG.tar.gz

tar xaf "$GMP.tar.bz2"
tar xaf "$MPC.tar.gz"
tar xaf "$MPFR.tar.bz2"
tar xaf "$ISL.tar.bz2"
tar xaf "$CLOOG.tar.gz"
tar xaf "$BINUTILS.tar.bz2"
tar xaf "$GCC.tar.xz"

mkdir combined
(
  cd combined
  ln -s ../$GCC/* .
  ln -s ../$BINUTILS/* .
  ln -s ../$GMP gmp
  ln -s ../$MPC mpc
  ln -s ../$MPFR mpfr
  ln -s ../$ISL isl
  ln -s ../$CLOOG cloog
)
```

followed by

```
mkdir gcc-build
(
  cd gcc-build
  # export CFLAGS="-O3 -march=armv7-a -g0"
  # export CXXFLAGS="-O3 -march=armv7-a -g0"
  ../combined/configure \
        --enable-languages=c,c++,lto \
        --enable-gold \
        --enable-plugins \
        --enable-plugin \
        --disable-multilib \
        --disable-nls \
        --enable-werror=no \
        --enable-threads \
        --enable-lto \
        --with-float=hard \
        --with-arch=armv7-a \
        --with-fpu=neon \
        --build=arm-linux-gnueabihf \
        --with-build-config=bootstrap-lto \
        --enable-stage1-checking=release \
        --target=arm-linux-gnueabihf \
        --prefix=/opt/gcc-arm-cross
  make -j$(nproc) 
  make install-strip
)
```

abhiTronix, 2018-09-06:
>  I have scripts almost ready but I miss the raspberry sysroot, that's how I stumbled on your projects.

@jcelerier, Thanks for this but the main goal of this repo is to Skip the compiling part(_Zero effort_) in the first place. Binaries provided are already compiled_(**Precompiled** as mentioned in the title)_ with Rpi supported compiler flags, You can, however, if you require any additional language support/feature, you are always welcomed.

jcelerier, 2018-09-06:
> Yes, but sharing it may defeat the purpose,

why ? let's put it bluntly: the point of having scripts / dockerfiles is that other people will still be able to produce newer binaries easily if, e.g. tomorrow you get hit by a bus and GCC 9 comes out. Besides, with a script, you can setup continuous integration so that binaries are built automatically from the latest GCC source for instance instead of doing it yourself. 

Thanks for the link, it's what I needed in any case !

abhiTronix, 2018-09-06:
> why ? let's put it bluntly: the point of having scripts / dockerfiles is that other people will still be able to produce newer binaries easily if, e.g. tomorrow you get hit by a bus and GCC 9 comes [...]

@jcelerier Yes, you're correct, But I created this repo. to simplify things, So that others may not have to compile/recompile same stuff everytime they need binaries. Some good scripts are available all over the internet, but it required on the go tweaking and stuff to reach the desired output according to my experience. Also, not all GCC compilers flags/compiling is similar, so it adds to the complexity of maintaining different version's wiki rather than uploading the all-in-one finished binary.

I will try to release the latest/recent binaries as soon as possible. Thanks for suggesting this.Goodluck

abhiTronix, 2019-10-02:
**I'm converting this repository for building and releasing automated cli docker builds in upcoming days. Kindly help me in achieving this goal. Thank you.**

closed 2020-02-05 by abhiTronix