abhiTronix/raspberry-pi-cross-compilers #3: cross-compiler 8.2.0 doesn't have {sysroot}/usr/include in its header search path

avilleret, 2019-01-10:
with your 8.2.0 compiler I got : 

```
$ echo | LANG=en_US.UTF-8 arm-linux-gnueabihf-gcc --sysroot=/home/antoine/dev/ofnode/sysroot -E -Wp,-v -
ignoring nonexistent directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/sys-include"
ignoring duplicate directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/include"
ignoring duplicate directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/include-fixed"
ignoring nonexistent directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/sys-include"
ignoring duplicate directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/include
 /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/include-fixed
 /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/include
End of search list.
# 1 "<stdin>"
# 1 "<built-in>"
# 1 "<command-line>"
# 31 "<command-line>"
# 1 "/opt/cross-pi-gcc-8.2.0/arm-linux-gnueabihf/include/stdc-predef.h" 1 3
# 32 "<command-line>" 2
# 1 "<stdin>"
```

while with the 8.1.0 I installed from archlinux repo I have :

```
$  echo | arm-linux-gnueabihf-gcc --sysroot=~/dev/ofnode/sysroot -E -Wp,-v -
ignoring nonexistent directory "~/dev/ofnode/sysroot/usr/arm-linux-gnueabihf/include"
ignoring nonexistent directory "~/dev/ofnode/sysroot/usr/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/arm-linux-gnueabihf/8.1.0/include
 /usr/lib/gcc/arm-linux-gnueabihf/8.1.0/include-fixed
 /usr/lib/gcc/arm-linux-gnueabihf/8.1.0/../../../../arm-linux-gnueabihf/include
End of search list.
# 1 "<stdin>"
# 1 "<built-in>"
# 1 "<command-line>"
# 31 "<command-line>"
# 1 "/usr/arm-linux-gnueabihf/include/stdc-predef.h" 1 3
# 32 "<command-line>" 2
# 1 "<stdin>"
[antoine@antoine-pc ~]$  echo | arm-linux-gnueabihf-gcc --sysroot=/home/antoine/dev/ofnode/sysroot -E -Wp,-v -
ignoring nonexistent directory "/home/antoine/dev/ofnode/sysroot/usr/arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/arm-linux-gnueabihf/8.1.0/include
 /usr/lib/gcc/arm-linux-gnueabihf/8.1.0/include-fixed
 /usr/lib/gcc/arm-linux-gnueabihf/8.1.0/../../../../arm-linux-gnueabihf/include
 /home/antoine/dev/ofnode/sysroot/usr/include
End of search list.
# 1 "<stdin>"
# 1 "<built-in>"
# 1 "<command-line>"
# 31 "<command-line>"
# 1 "/usr/arm-linux-gnueabihf/include/stdc-predef.h" 1 3
# 32 "<command-line>" 2
# 1 "<stdin>"
```

Please note that in the second case, `#include <...> search starts here:` lists `/home/antoine/dev/ofnode/sysroot/usr/include` while the first (8.2.0) doesn't.

While I can manually add the include path with a `-I` flag, it makes this compiler hard to use with CMake which strips off "${CMAKE_SYSROOT}/usr/include".
see related issue here : https://gitlab.kitware.com/cmake/cmake/issues/17966

abhiTronix, 2019-01-10:
@avilleret  Can share the `GCC 8.1.0` link that you tried from archlinux repo?

avilleret, 2019-01-10:
it's in AUR repo : https://aur.archlinux.org/packages/arm-linux-gnueabihf-gcc/

abhiTronix, 2019-01-10:
@avilleret  I thought you're using custom Cross-Compiler similar this repo. ,

The main reason behind this bug/error is that the Official Cross Compiler install files directly to the system root directory, i.e `/usr/` or `/usr/local/` and Cmake is made explicitly to work with such configurations only and not for custom paths/configurations. Therefore custom Cross-Compiler(like provided here) fails to work in this scenario as its default path is `/opt/` not the root directory. 

So unfortunately, if you want to use **GCC 8.2.0** Cross-Compiler on your system, **you yourself have compile & installed it from source, to the root directory of your machine.**

abhiTronix, 2019-01-10:
@avilleret  They're working on a fix here(in progress): https://gitlab.kitware.com/cmake/cmake/merge_requests/2716

avilleret, 2019-01-10:
it is indeed possible to use your Crosscompiler build with CMake (and I thanks you a lot for sharing)
I have to disable the CMake's implicit directory feature by injecting the following lines into my toolchain file

```
unset(CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES)
unset(CMAKE_CXX_IMPLICIT_INCLUDE_DIRECTORIES)
```

I was unsure if its a bug (or a feature) of CMake or if it's a lake of configuration on your side.
Now I'm fixed :-)
Thanks for the quick reply, thanks for the explanation, and thanks again for all this wonderful work.

closed 2019-01-10 by abhiTronix