abhiTronix/raspberry-pi-cross-compilers #4: Hardcoded paths in linker scripts

mitza-oci, 2019-01-25:
The following linker scripts assume the crosscompiler is installed in `/opt/cross-pi-gcc`
```
cross-pi-gcc-8.2.0/arm-linux-gnueabihf/lib/libpthread.so
cross-pi-gcc-8.2.0/arm-linux-gnueabihf/lib/libc.so
```

mitza-oci, 2019-02-01:
The idea of eliminating hardcoded paths is that a user without write permissions in `/opt` could still use it (for example, in the home directory).

abhiTronix, 2019-02-02:
@mitza-oci That might result in broken binding, if not careful. I'll consider implementing this feature in the next release. Related PR is always welcomed!

mitza-oci, 2019-02-05:
> That might result in many broken binding, if not carefully done. I'll consider implementing this feature in the next release.

For now I'd suggest updating the instructions, since they do not currently result in a working toolchain.

> Related PR is always welcomed!

Not sure how to go about this, since the repository doesn't seem to contain the scripts used to produce the crosscompilers.

abhiTronix, 2019-03-02:
@mitza-oci I tried to eliminate from Cross-Compiler GCC Binaries by building them from scratch but it always turns out to be broken at the end. Also, I can't find any example of such implementation anywhere online. Therefore this feature is abandoned for the future. Feel free to point me in the right direction if you find any related info. Goodluck

abhiTronix, 2019-05-24:
@raitraak-rrk @mitza-oci, See https://github.com/abhiTronix/raspberry-pi-cross-compilers/issues/8#issuecomment-495691678

closed 2019-05-28 by abhiTronix

abhiTronix, 2019-05-28:
Solved in v2.7