abhiTronix/raspberry-pi-cross-compilers #6: Ln directions do not show ln from /opt but only from /usr

jimcruse, 2019-01-26:
The link from to final should be from /opt not /usr

abhiTronix, 2019-01-26:
@jimcruse Are you using Native binaries(.i.e for Raspberry Pi) or Cross-Compiler(i.e. for Linux)?

jimcruse, 2019-01-26:
Native for pi

jimcruse, 2019-01-26:
> The link from to final should be from /opt not /usr
>
> *What issue you are facing from linking to /usr?*
>
> In readme:
>
> Manage Links as below:(Native Compiler Only)
> source .bashrc
> sudo ln -s /usr/include/arm-linux-gnueabihf/sys /usr/include/sys
> sudo ln -s /usr/include/arm-linux-gnueabihf/bits /usr/include/bits
> sudo ln -s /usr/include/arm-linux-gnueabihf/gnu /usr/include/gnu
> sudo ln -s /usr/include/arm-linux-gnueabihf/asm /usr/include/asm
> sudo ln -s /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
> sudo ln -s /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
> sudo ln -s /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
>
> These links are created to replace old(default) GCC 6.3.0 i.e. available
> from apt command. *Please specify your problem?*

From /usr/lib/arm-linux-gnueabihf/crtn.o should be /opt .....

abhiTronix, 2019-01-26:
@jimcruse, 

> From /usr/lib/arm-linux-gnueabihf/crtn.o should be /opt .....

You are understanding this wrong,

As mentioned, In Readme:
```
Manage Links as below:(Native Compiler Only)
source .bashrc
sudo ln -s /usr/include/arm-linux-gnueabihf/sys /usr/include/sys
sudo ln -s /usr/include/arm-linux-gnueabihf/bits /usr/include/bits
sudo ln -s /usr/include/arm-linux-gnueabihf/gnu /usr/include/gnu
sudo ln -s /usr/include/arm-linux-gnueabihf/asm /usr/include/asm
sudo ln -s /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
sudo ln -s /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
sudo ln -s /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
```
> Both from and to are /usr but nothing is /opt where the new files are

That is because my GCC files expect those files to be in `/usr/include location` not `/usr/include/arm-linux-gnueabihf`or similiarliarly for libararies `/usr/lib location` not `/usr/lib/arm-linux-gnueabihf` and **There is no need to create `/opt` files symbolic links beause it is already in system path** with step in Readme as follows:
```Properly configure paths as below(Permanently by adding it to your .bashrc):
echo 'export PATH=/opt/<extracted folder-name e.g gcc-8.2.0>/bin:$PATH' >> .bashrc
echo 'export LD_LIBRARY_PATH=/opt/<extracted folder-name e.g gcc-8.2.0>/lib:$LD_LIBRARY_PATH' >> .bashrc
```

closed 2019-01-26 by abhiTronix

jimcruse, 2019-01-26:
After redoing the setup

root@raspberrypi:/home/pi/gcc_all/cross-pi-gcc-8.2.0/bin# ./arm-linux-gnu
eabihf-
gcc-8.2.0
bash: ./arm-linux-gnueabihf-gcc-8.2.0: cannot execute binary file: Exec
format error
root@raspberrypi:/home/pi/gcc_all/cross-pi-gcc-8.2.0/bin# exit

Closed #6