abhiTronix/raspberry-pi-cross-compilers #7: Exec format error when trying execute gcc-8.2.0

jimcruse, 2019-01-26:
After all the setup I now get an exec format error
I now get an: root@raspberrypi:/home/pi/gcc_all/cross-pi-gcc-8.2.0/bin# .arm-linux-gnueabihf- gcc-8.2.0 bash: ./arm-linux-gnueabihf-gcc-8.2.0: cannot execute binary file: Exec format error

closed 2019-01-30 by abhiTronix

jimcruse, 2019-02-15:
I reinstalled everything from scratch and got it working, had to relink g++ to /opt/gcc-8.2.0/bin/arm-linux-gnueabihf-g++

abhiTronix, 2019-02-20:
> I reinstalled everything from scratch and got it working, had to relink g++ to /opt/gcc-8.2.0/bin/arm-linux-gnueabihf-g++

@jimcruse This Bug is fixed in commit https://github.com/abhiTronix/raspberry-pi-cross-compilers/commit/645352297e1074e5a0ab27c354ace6a1914b8467 . **Kindly Try ReadMe section again. Thankyou**

jimcruse, 2019-02-20:
> 6453522

I'll try next week, I broke my raspberry pi