abhiTronix/raspberry-pi-cross-compilers #8: limits.h broken

mitza-oci, 2019-02-15:
Expected behavior (using host system's GCC):
```
$ echo -e "#include <limits.h>\n_POSIX_PATH_MAX" | gcc - -E -P
256
```
Observed behavior with cross-compiler from this project:
```
$ echo -e "#include <limits.h>\n_POSIX_PATH_MAX" | /opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc - -E -P
_POSIX_PATH_MAX
```
It looks like limits.h from include-fixed is not including the real limits.h.

abhiTronix, 2019-02-16:
Nope, it isn't

Here is the verbose output of your command(*enabled by `-v` flag*):
```
COLLECT_GCC=/opt/cross-pi-gcc-8.2.0/bin/arm-linux-gnueabihf-gcc
Target: arm-linux-gnueabihf
Configured with: ../gcc-8.2.0/configure --prefix=/opt/cross-pi-gcc --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --program-prefix=arm-linux-gnueabihf- --enable-shared --with-cpu=cortex-a7 --with-fpu=neon-vfpv4 --with-float=hard --enable-threads=posix --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-target-system-zlib --disable-sjlj-exceptions --enable-checking=release --enable-multiarch --disable-multilib
Thread model: posix
gcc version 8.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-E' '-v' '-P' '-mcpu=cortex-a7' '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7ve+simd'
 /opt/cross-pi-gcc-8.2.0/bin/../libexec/gcc/arm-linux-gnueabihf/8.2.0/cc1 -E -quiet -v -P -imultilib . -imultiarch arm-linux-gnueabihf -iprefix /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/ - -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4 -mtls-dialect=gnu -marm -march=armv7ve+simd
ignoring nonexistent directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/sys-include"
ignoring duplicate directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/include"
ignoring duplicate directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/include-fixed"
ignoring nonexistent directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/sys-include"
ignoring duplicate directory "/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/include
 /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/include-fixed
 /opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/include
End of search list.
COMPILER_PATH=/opt/cross-pi-gcc-8.2.0/bin/../libexec/gcc/arm-linux-gnueabihf/8.2.0/:/opt/cross-pi-gcc-8.2.0/bin/../libexec/gcc/arm-linux-gnueabihf/:/opt/cross-pi-gcc-8.2.0/bin/../libexec/gcc/:/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/:/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/:/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/:/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/lib/
COLLECT_GCC_OPTIONS='-E' '-v' '-P' '-mcpu=cortex-a7' '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7ve+simd'
```
**where `/opt/cross-pi-gcc-8.2.0/bin/../lib/gcc/arm-linux-gnueabihf/8.2.0/../../../../arm-linux-gnueabihf/include` is the correct file path, where the actual `limits.h` file is located.**, *Otherwise it might have raised the error such as file not found or linked.*
<br>

### So, coming back to your query:
>Observed behavior with cross-compiler from this project:

```
$ echo -e "#include <limits.h>\n_POSIX_PATH_MAX" | /opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc - -E -P
_POSIX_PATH_MAX
```
*Actually the `_POSIX_PATH_MAX` is now located in `posix1_lim.h` file, not in `limit.h` . But it get activate if `__USE_POSIX` flag is enabled/defined in limit.h. But I don't find a way to activate it, or maybe I'm misunderstanding it, or It is changed in GCC 8.2.0 only.*

### Solution:

So, you can achive expected behaviour in our cross-compiler using following modified command(***Don't forget to change <cross-pi-gcc-8.2.0> to your cross-compiler folder***):
```
$ echo -e "#include <bits/posix1_lim.h>\n_POSIX_PATH_MAX" | /opt/<cross-pi-gcc-8.2.0>/bin/arm-linux-gnueabihf-gcc - -E -P
256
```   
Hope that helps!

abhiTronix, 2019-02-18:
@mitza-oci  I'll consider it solved then!

mitza-oci, 2019-02-18:
It's not solved, the cross-compiler fails to provide the required system headers.  Building common open-source packages fails (https://github.com/google/googletest/blob/d850e144710e330070b756c009749dc7a7302301/googletest/src/gtest-filepath.cc#L55)
Compare to the official Raspberry Pi toolchain at https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64

abhiTronix, 2019-02-18:
@mitza-oci That's weird that my briefly explained reply doesn't make sense to you. Kindly re-read it, the answer is right there.

mitza-oci, 2019-02-18:
If the solution involves changing 3rd-party source code, that's not going to work.  When including exactly `<limits.h>`, the preprocessor macro `_POSIX_PATH_MAX` should be defined.  I'm looking for either flags we can pass to the compiler to make this work, or changes to the toolchain itself.

abhiTronix, 2019-02-18:
@mitza-oci Yes, Solution surely involves changing the third-party source code it requires including correct Header files to make them work correctly. _Since GCC 8.2.0 source may have changed and not yet supported by the third-party binaries you want to use. **Kindly raise an issue regarding this issue in the third-party binaries to support new GCC 8.2.0 compilers binaries.**_

Being said that, you may want to switch to **GCC 6.3.0 cross-compiler binaries** instead, available in this repo. here: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compilers/

abhiTronix, 2019-02-18:
@mitza-oci Also,
Compare to the official Raspberry Pi toolchain at https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64
are ancient relics, Kindly avoid it , otherwise it's upto you. Goodluck!

raitraak-rrk, 2019-05-18:
Getting the same issue with PATH_MAX.
gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf provides the define in limits.h. Why is this compiler build acting differently?

abhiTronix, 2019-05-19:
> Getting the same issue with PATH_MAX.
> gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf provides the define in limits.h. Why is this compiler build acting differently?

@raitraak-rrk Are you getting this error with my binaries or someone else's? 

This is an issue, as I explained [earlier](https://github.com/abhiTronix/raspberry-pi-cross-compilers/issues/8#issuecomment-464309203), is with the newer versions of GCC. since **_source files have been changed in the recent releases_**, I don't know from which version this change has been made. So to be on safer side, You need to use GCC 6.3.0 to avoid this error.

raitraak-rrk, 2019-05-19:
Hi,

tested with your 8.3 and the mentioned Arm 8.3(gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf ) release. Only appears with your binary. The issue is PATH_MAX not being defined in limits.h (or the wrong limits.h is being used).

Test code:

#include <limits.h>
constexpr auto pathmax {PATH_MAX};

User "uray" seems to have a similar issue in the cross-compiler guide comments:
https://preshing.com/20141119/how-to-build-a-gcc-cross-compiler/

abhiTronix, 2019-05-19:
@raitraak-rrk Can you test my GCC 9.1.0 binary too and revert?

abhiTronix, 2019-05-19:
@raitraak-rrk I thought of manually defining PATH_MAX but PATH_MAX variable is different for different operating systems like for windows it's 260, on Linux, it's 4096, on OpenBSD, it's 1024.

More information here: http://insanecoding.blogspot.com/2007/11/pathmax-simply-isnt.html

raitraak-rrk, 2019-05-19:
Same behaviour with your 9.1.

PATH_MAX is defined in linux/limits.h so pretty sure this issue is related how the compiler build is configured. Some threads mention "configuration magic" related to PATH_MAX.

https://bugs.chromium.org/p/nativeclient/issues/detail?id=3190

Personally, I don't really care if the value is valid, but the missing definition breaks external library code (e.g. GoogleTest).

abhiTronix, 2019-05-19:
@raitraak-rrk definition is not missing, It is not being accessed properly. 

> Some threads mention "configuration magic" related to PATH_MAX.
> 
> https://bugs.chromium.org/p/nativeclient/issues/detail?id=3190

Can you summarize this configration here?

raitraak-rrk, 2019-05-19:
Sorry mate, I have no experience in building compilers :) You are way ahead of me. Maybe post to SO for help?

Cheers

abhiTronix, 2019-05-19:
@raitraak-rrk Thanks anyway, I'm working on it. If you find something else related to this issue, kindly report it here!

abhiTronix, 2019-05-19:
@raitraak-rrk replace `#include <limit.h>` with `#include <linux/limits.h>`, does it works?

raitraak-rrk, 2019-05-19:
Yes. As I mentioned PATH_MAX is defined in linux/limits.h. But somehow it should get to the "normal" limits.h too.

abhiTronix, 2019-05-19:
@raitraak-rrk Problem located! 

These binaries build around  glibc-2.24 that is somehow using its own header files in which `limits.h` do not define `PATH_MAX` but it is rather defined by  `pathmax.h` file which is the actual cause for this issue.

**The official glibc document says I quote:**

> For PATH_MAX, Gnulib provides a module pathmax with a header file "pathmax.h". It defines PATH_MAX to a constant on platforms with a file name length limit. 

See here: https://www.gnu.org/software/gnulib/manual/html_node/limits_002eh.html

abhiTronix, 2019-05-24:
@raitraak-rrk Sorry for the delay. **The `limit.h` issue is now resolved.** 

_To confirm this kindly Stress test this [Cross-Compiler 6.3.0 binary](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/GCC%206.3.0/Raspberry%20Pi%201%2C%20Zero/cross-gcc-6.3.0-pi_0-1.tar.gz/download) that I have uploaded to Sourceforge for testing purposes. Kindly run all google test you know and test if it breaks or gives any error. Also, try to find any Hardcoded paths on the way, See issue #4. Please give you full review here._ 

**Once confirmed, I will push all binaries to the server, Thankyou in Advanced.**

abhiTronix, 2019-05-24:
@raitraak-rrk @mitza-oci, Apologies for the delay, I have resolved this issue.

**Kindly download this [cross-compiler 6.3.0 binary](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/GCC%206.3.0/Raspberry%20Pi%201%2C%20Zero/cross-gcc-6.3.0-pi_0-1.tar.gz/download) and Stress Test it with Google Tests and report the results here. Also kindly search if all hardcoded path are removed in this binary, See issue #4.**

Once confirmed it works, I will upload the rest of the cross-compilers binaries. Good luck!

raitraak-rrk, 2019-05-25:
Hi,
thanks for the update. Tested the binaries with the simple code snippet and it compiles without issues.
#include <limits.h>
constexpr auto pathmax {PATH_MAX};

Not sure what you mean by stress testing. Unfortunately my codebase doesn't compile with gcc6, so cannot verify googletest compilation. As long as the issue was only with PATH_MAX it should be fine.

No idea how to verify the other issue.

Cheers

abhiTronix, 2019-05-25:
@raitraak-rrk Thank you for the update. 

> Not sure what you mean by stress testing. Unfortunately my codebase doesn't compile with gcc6, so cannot verify googletest compilation. 
- From Stress Test I meant Can you compile a complex code and see if it passes a series of Google tests?
- Secondly, can you tell which GCC Cross-Compiler version works on your machine? I've uploaded [version 8.3.0 here](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/GCC%208.3.0/Raspberry%20Pi%202%2C%203/cross-gcc-8.3.0-pi_2-3.tar.gz/download). Is that ok or you want version 9.1.0?

> No idea how to verify the other issue.

It is very simple, just extract toolchain and cd into `bin` directory(`cross-gcc-8.3.0-pi_0-1/bin`) and paste this command:
```sh
./arm-linux-gnueabihf-gcc -print-search-dirs | sed '/^lib/b 1;d;:1;s,/[^/.][^/]*/\.\./,/,;t 1;s,:[^=]*=,:;,;s,;,;  ,g' | tr \; \\012
```
 and see if any folder says not found or through any error. 

**Revert the output here if there is any error. Good luck!**

raitraak-rrk, 2019-05-26:
Can you please upload 8.3 or 9.1 version for Model 3B+. That's the only Pi I have.

Output from the provided command looks clean:
`rait@Aq-Home:/opt/cross-pi-gcc-6.3.0-0/bin$ ./arm-linux-gnueabihf-gcc -print-search-dirs | sed '/^lib/b 1;d;:1;s,/[^/.][^/]*/\.\./,/,;t 1;s,:[^=]*=,:;,;s,;,;  ,g' | tr \; \\012
libraries:
  /opt/cross-pi-gcc-6.3.0-0/lib/gcc/arm-linux-gnueabihf/6.3.0/:/opt/cross-pi-gcc-6.3.0-0/lib/gcc/:/opt/cross-pi-gcc-6.3.0-0/arm-linux-gnueabihf/lib/arm-linux-gnueabihf/6.3.0/:/opt/cross-pi-gcc-6.3.0-0/arm-linux-gnueabihf/lib/:/opt/cross-pi-gcc-6.3.0-0/arm-linux-gnueabihf/libc/lib/arm-linux-gnueabihf/6.3.0/:/opt/cross-pi-gcc-6.3.0-0/arm-linux-gnueabihf/libc/lib/:/opt/cross-pi-gcc-6.3.0-0/arm-linux-gnueabihf/libc/usr/lib/arm-linux-gnueabihf/6.3.0/:/opt/cross-pi-gcc-6.3.0-0/arm-linux-gnueabihf/libc/usr/lib/`

abhiTronix, 2019-05-26:
@raitraak-rrk thanks for helping out,

> Can you please upload 8.3 or 9.1 version for Model 3B+. That's the only Pi I have.

You don't need Raspberry Pi for this purpose since you dealing with Cross-Compilers here, but you'll need a Linux machine only in this case. But **I've uploaded the  Cross Compilers [version 8.3.0](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/GCC%208.3.0/Raspberry%20Pi%203A%2B%2C%203B%2B/cross-gcc-8.3.0-pi_3%2B.tar.gz/download) and [version 9.1.0](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/GCC%209.1.0/Raspberry%20Pi%203A%2B%2C%203B%2B/cross-gcc-9.1.0-pi_3%2B.tar.gz/download) for the Raspberry Pi Model 3B+ as you desired. You can run Google Tests easily now!**

> Output from the provided command looks clean:
> rait@Aq-Home:/opt/cross-pi-gcc-6.3.0-0/bin$ ./arm-linux-gnueabihf-gcc -print-search-dirs | sed '/^lib/b 1;d;:1;s,/[^/.][^/]*/\.\./,/,;t [...]

Thanks for the update but **DO NOT** move binaries to `/opt` folder position. Just download in any folder such as default `home/<user>/Downloads`, extract it there and run the previous command i.e given above in `/bin` folder. **Remember, It must NOT contain any path that says `/opt/cross-pi-gcc-6.3.0-0/...something path...` when you run this command but instead paths must show the `home/<user>/Downloads/cross-pi-gcc-8.3.0-2/..something path ..` . If it does show any `/opt/cross-pi-gcc-6.3.0-0/...something paths..` at any line then it is a FAILURE for hardcoded path.** 

**Report the output here. Goodluck!**

raitraak-rrk, 2019-05-28:
`rait@Aq-Home:~/gcc/cross-pi-gcc-9.1.0-2/bin$ ./arm-linux-gnueabihf-gcc -print-search-dirs | sed '/^lib/b 1;d;:1;s,/[^/.][^/]*/\.\./,/,;t 1;s,:[^=]*=,:;,;s,;,;  ,g' | tr \; \\012

libraries:
  /home/rait/gcc/cross-pi-gcc-9.1.0-2/lib/gcc/arm-linux-gnueabihf/9.1.0/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/lib/gcc/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/arm-linux-gnueabihf/lib/arm-linux-gnueabihf/9.1.0/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/arm-linux-gnueabihf/lib/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/arm-linux-gnueabihf/libc/lib/arm-linux-gnueabihf/9.1.0/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/arm-linux-gnueabihf/libc/lib/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/arm-linux-gnueabihf/libc/usr/lib/arm-linux-gnueabihf/9.1.0/:/home/rait/gcc/cross-pi-gcc-9.1.0-2/arm-linux-gnueabihf/libc/usr/lib/`

raitraak-rrk, 2019-05-28:
GoogleTest also seems to build now with 9.1.

However, when trying to run on target (Pi 3B+) getting:
`/tmp/LibraryTest: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version GLIBCXX_3.4.26 not found (required by /tmp/LibraryTest)`

Also installed the native version on the Pi but this didn't help. Any ideas?

abhiTronix, 2019-05-28:
> GoogleTest also seems to build now with 9.1.
> 
> However, when trying to run on target (Pi 3B+) getting:
> /tmp/LibraryTest: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version GLIBCXX_3.4.26 not found (required by /tmp/LibraryTest)

> Also installed the native version on the Pi but this didn't help. Any ideas?

@raitraak-rrk  Thank you so much for the feedback. **You forgot to pass `export LD_LIBRARY_PATH=/opt/myGCCLibs` before running tests. Kindly set this flag and rerun tests.**

raitraak-rrk, 2019-05-28:
Seems VisualGDB project was broken somehow. All good now. Tests and app running and debugging on target. Thanks mate!

abhiTronix, 2019-05-28:
@raitraak-rrk, Again, thank you a lot for your time.

This issue is Resolved in v2.7

jcelerier, 2019-06-11:
is the issue solved in the files here ? https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Native-Compiler%20Toolchains/GCC%209.1.0/Raspberry%20Pi%202%2C%203/

with a simple `foo.c` : 

    #include <limits.h>
    char c[PATH_MAX]; 

I still get the undefined symbol error.

jcelerier, 2019-06-11:
> Can you please share your system details here? Such as output of uname -a and your current Raspberry Pi Model on which you are using these binaries?

I'm using them on a docker raspberry pi image (schachr/raspbian-stretch, armv7l)

> Did you strictly follow the instructions given in Wiki Section for Native Compiler Toolchains?

yup

> warning Finally, Remember Native Compiler Toolchains will only work with Raspberry Pi 32-bit ARM OS only(such as Raspbian), Do not use them on an AMD/Intel machine!

sure, the compiler works for most things, it's just limits.h which is the problem.

abhiTronix, 2019-06-11:
@jcelerier Thank you for this valuable information. I'm working on fixing this problem, once done, I'll inform you here.

abhiTronix, 2019-06-19:
@jcelerier Thank you for your patience.

 **I've eliminated this bug from latest GCC 9.1.0 Native Toolchain builds.**

**Kindly download these Bug-Free [Native compilers binaries from here](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Native-Compiler%20Toolchains/GCC%209.1.0/) and install them with [updated Wiki instructions](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Native-Compiler:-Installation-Instructions#raspberry-pi-native-compiler-toolchain-installation-instruction-) and thereby Stress Test it. _Also, share your test results here, I'm waiting for your response._**  

**Your test example `foo.c` :**
```c++
#include <limits.h>
char c[PATH_MAX]; 
```
**should work now. Cheers!**

_Once confirmed, that bug is eliminated, I'll upload rest of 8.3.0 bug-free GCC Native binaries. Goodluck_

jcelerier, 2019-06-19:
Great ! testing asap. Thanks for your time !

jcelerier, 2019-06-19:
hmmm, the new native compiler behaves weirdly : it has a sysroot in `/opt/native-pi-gcc-9.1.0-1/arm-linux-gnueabihf/libc/usr`, with a libc, etc... but it does not really make sense for a native compiler since it should use the glibc, libdl, etc that are in /usr/lib/ and /usr/lib/arm-linux-gnueabihf

abhiTronix, 2019-06-20:
> hmmm, the new native compiler behaves weirdly : it has a sysroot in /opt/native-pi-gcc-9.1.0-1/arm-linux-gnueabihf/libc/usr, with a libc, etc... but it does not really make sense for a native compiler since it should use the glibc, libdl, etc that are in /usr/lib/ and /usr/lib/arm-linux-gnueabihf

@jcelerier  Nope, you are wrong. **That is not a weird behaviour but the desired one since I've finally made these toolchain binaries completely standalone. It now DOES NOT depend on your system  `glibc or libdl etc` or anything but provides it's own `glibc etc.` files that reside in `libc` sysroot folder.** 

### Reason for doing that?

During my tests, I found many inconsistency in the location of important dependent files that reside in Raspbian OS and any Linux machine on which these native binaries are cross-compiled. **These inconsistencies tend to broke the required symlinks and thereby working of these binaries. For example take the location of `sys` Folder in Raspian OS, it is `/usr/include/arm-linux-gnueabihf/sys` but on any Linux(x64) system it is `/usr/include/x86_64-linux-gnu/sys`. This bug was the reason why my previous binaries were throwing undefined symbol error.** 

@jcelerier It doesn't matter where the files are located as long as they working as expected. **The real question that I asked was "_Are these binaries working/passing the Test or not?_" Can you confirm it?**

jcelerier, 2019-06-20:
Hello,
> That is not a weird behaviour but the desired one since I've finally made these toolchain binaries completely standalone.

I understand that this is useful for cross-compiler toolchains, but I don't understand the point for native toolchains since those are going to be run on a raspbian system anyways. 

> provides it's own glibc etc.

this means that if I want to distribute binaries built with your toolchain, I also have to ship your glibc, etc along with my app, which may break in plenty of places (https://stackoverflow.com/questions/847179/multiple-glibc-libraries-on-a-single-host/851229#851229)

> The real question that I asked was "Are these binaries working/passing the Test or not?"

well, the answer is that I cannot tell you because now my build breaks way earlier because none of the libraries installed in my raspbian system are found.

abhiTronix, 2019-06-20:
> I understand that this is useful for cross-compiler toolchains, but I don't understand the point for native toolchains since those are going to be run on a raspbian system anyways.

- First of all these native binaries are cross-compiled using the Cross-Compiler available with this repository. So when we compile these binaries we need important system header files and libs to make it work correctly. As I said the location of header files differs in ARM and AMD/Intel sytem, thereby it causes conflict with the path of these header files. 
- Secondly when these files are compiled, the symlinks and path are coded onto the executable which cannot be changed at runtime.

**Hence because of these two main reasons I made this decision to isolate paths of selected runtime heaer files and libs.**

> provides it's own glibc etc.
> 
> this means that if I want to distribute binaries built with your toolchain, I also have to ship your glibc, etc along with my app, which may break in plenty of places (https://stackoverflow.com/questions/847179/multiple-glibc-libraries-on-a-single-host/851229#851229)

Lmao 😆 , I don't think you know how a compiler works? Do you ship GCC Compilers too with your apps? If you do, then its like you ship Android Studio with Android App, which you clearly don't!

**Glibc** _aka GNU C Library project provides the core libraries for the GNU system and GNU/Linux systems, as well as many other systems that use Linux as the kernel. These libraries provide critical APIs including ISO C11, POSIX.1-2008, BSD, OS-specific APIs and more. These APIs include such foundational facilities as open, read, write, malloc, printf, getaddrinfo, dlopen, pthread_create, crypt, login, exit and more._

**You DO NOT need any glibc or anything else with your app. Glibc is a part of GCC GNU Project, need to make GCC work. Glibc has nothing to do with what GCC compiles or what its output is!**

The link you shared clearly states: "_It is very possible to have multiple versions of glibc on the same system (we do that every day). However, you need to know that glibc consists of many pieces (200+ shared libraries) which all must match._"  **That's why I'm providing my own correct Glibc files in a closed environment so that it won't conflict with your system files.**

> well, the answer is that I cannot tell you because now my build breaks way earlier because none of the libraries installed in my raspbian system (openssl, libgles, etc...) are found.

**You are clearly making this up, Don't you?** Remember My GCC binaries need no openssl, libgles, etc... or whatever. **These binaries are standalone and can work directly on a freshly installed Raspbian OS without any dependency. Let's see what's the error you are getting, can you post the whole system logs/ error here?**

jcelerier, 2019-06-20:
> You are clearly making this up, Don't you? Remember My GCC binaries need no openssl, libgles, etc... or whatever. 

Yes, that's not what I am saying.
Your GCC binaries only have their own sysroot in their include & link path, and links *my* software with *your* libc headers, glibc & sysroot which is different that the one that comes with Raspbian. It should pick up the glibc in /usr/lib so that my software can be relocated on another raspberry without having to copy your glibc along. 

> My GCC binaries are standalone and certainly can work directly on a freshly installed Raspbian OS without any dependency. Let's see what's the error you are getting, can you post the whole system logs/ error here?

here is a compiler invocation with -v : 

https://termbin.com/i3j3

notice how /usr/include and /usr/include/arm-linux-gnueabihf appears *nowhere*, so your compiler cannot pick up the system libs.

Here is a dockerfile that reproduces the issue : 

```
FROM schachr/raspbian-stretch

COPY qemu-arm-static /usr/bin/qemu-arm-static

RUN (ln -s /usr/include/arm-linux-gnueabihf/sys /usr/include/sys; \
 ln -s /usr/include/arm-linux-gnueabihf/bits /usr/include/bits; \
 ln -s /usr/include/arm-linux-gnueabihf/gnu /usr/include/gnu; \
 ln -s /usr/include/arm-linux-gnueabihf/asm /usr/include/asm; \   
 ln -s /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o; \  
 ln -s /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o; \   
 ln -s /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o; \
 )
 
RUN apt -y update && apt -y install build-essential git wget bzip2 bison flex ruby python perl texinfo debhelper chrpath sharutils libffi-dev patchutils diffstat xz-utils libedit-dev procps zlib1g-dev  "^libxcb.*" libx11-xcb-dev libglu1-mesa-dev libxrender-dev libxi-dev gperf libxslt-dev libasound2-dev libfontconfig1-dev libx11-dev libxext-dev libxfixes-dev libxcb1-dev libxcb-glx0-dev libdbus-1-dev libssl-dev openssl libgles2-mesa-dev && apt -y remove gcc-6 g++-6 && \
     wget https://github.com/OSSIA/sdk/releases/download/embedded-sdk1/native-gcc-9.1.0-pi_2-3.tar.gz -O /opt/native-gcc.tar.gz && (rm -rf /usr/local; cd /opt; tar xaf native-gcc.tar.gz  )
 ```

this installs many dev packages, but your compiler cannot find them : 

```
root@adb089c1c48c:/# cat foo.cpp 
#include <openssl/ssl.h>
int main()
{
}
root@adb089c1c48c:/# /opt/native-pi-gcc-9.1.0-1/bin/g++-9.1.0 foo.cpp 
foo.cpp:1:10: fatal error: openssl/ssl.h: No such file or directory
    1 | #include <openssl/ssl.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

(openssl/ssl.h being in /usr/include of the raspberry)

Likewise here is a preprocessed version of a simple c file which includes <stdio.h> :  https://termbin.com/kq0u

notice how all the includes are sourced from "/opt/native-pi-gcc-9.1.0-1/arm-linux-gnueabihf/libc/usr/include/stdio.h" - instead, it should go use Raspbian's libc because they may not have the same symbols .

abhiTronix, 2019-06-20:
@jcelerier First of all Apologies for these bugs and conflict between us for these binaries and thank you for sharing this debugging data. In reality, we are both correct and justify what is our opinion, but **the main problem that led to this conflict between us was that unfortuntely _I used the wrong build flag while compiling final GCC native binaries that I uploaded to Sourceforge. But instead on my test Raspberry Pi machine, I used the correct build flag for my native binaries. That's why I wasn't experiencing any error/bugs on my end with these native binaries as you do._** 

**The wrong flag that I used was `--with-build-sysroot` instead of correct `--with-native-system-header-dir` flag for my native binaries which lead to these bugs/errors. This was a big silly mistake from my end. Appolizies for that.**

**I've uploaded the corrected GCC 9.1.0 Native Binary [here](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Native-Compiler%20Toolchains/GCC%209.1.0/Raspberry%20Pi%202%2C%203/native-gcc-9.1.0-pi_2-3.tar.gz/download) with correct build flag this time. Kindly re-download these binaries and re-run the tests. I'm confident you won't recieve any error this time. Goodluck!**

jcelerier, 2019-06-21:
Hello, 
I just tested but this version doesn't have any headers packaged in - so simple programs such as

     #include <string>

do not compile.

The compiler returns the following : 

```
# g++ -v test.cpp 
Using built-in specs.
COLLECT_GCC=/opt/native-pi-gcc-9.1.0-1/bin/g++
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --build=x86_64-pc-linux-gnu --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf --with-sysroot=/arm-linux-gnueabihf/libc --with-native-system-header-dir=/opt/native-pi-gcc-9.1.0-1/arm-linux-gnueabihf/libc --enable-languages=c,c++,fortran --with-arch=armv7-a --with-fpu=neon-vfpv4 --with-float=hard --disable-multilib --program-suffix=-9.1.0
Thread model: posix
gcc version 9.1.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/native-pi-gcc-9.1.0-1/bin/../libexec/gcc/arm-linux-gnueabihf/9.1.0/cc1plus -quiet -v -iprefix /opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/ -isysroot /opt/native-pi-gcc-9.1.0-1/bin/../arm-linux-gnueabihf/libc -D_GNU_SOURCE test.cpp -quiet -dumpbase test.cpp -mfloat-abi=hard -mfpu=neon-vfpv4 -mtls-dialect=gnu -marm -march=armv7-a+neon-vfpv4 -auxbase test -version -o /tmp/ccCwYXBv.s
GNU C++14 (GCC) version 9.1.0 (arm-linux-gnueabihf)
	compiled by GNU C version 9.1.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../include/c++/9.1.0"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../include/c++/9.1.0/arm-linux-gnueabihf"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../include/c++/9.1.0/backward"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/include"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/include-fixed"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/include"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../include/c++/9.1.0"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../include/c++/9.1.0/arm-linux-gnueabihf"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../include/c++/9.1.0/backward"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/9.1.0/include"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../arm-linux-gnueabihf/libc/usr/local/include"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/9.1.0/include-fixed"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/include"
ignoring nonexistent directory "/opt/native-pi-gcc-9.1.0-1/bin/../arm-linux-gnueabihf/libc/opt/native-pi-gcc-9.1.0-1/arm-linux-gnueabihf/libc"
#include "..." search starts here:
#include <...> search starts here:
End of search list.
GNU C++14 (GCC) version 9.1.0 (arm-linux-gnueabihf)
	compiled by GNU C version 9.1.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: f6f989d3b086e434dc27bd601d90af60
cc1plus: error: no include path in which to search for stdc-predef.h
test.cpp:1:23: error: no include path in which to search for type_traits
    1 | #include <type_traits>
      |                       ^

```

jcelerier, 2019-06-21:
I've ended up building mine, here's what its configure line looks like : 

```
root@d769aeb0970e:/usr/local/bin# ./arm-linux-gnueabihf-gcc-9 foo.c -v
Using built-in specs.
COLLECT_GCC=./arm-linux-gnueabihf-gcc-9
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../combined/configure --enable-languages=c,c++,lto --enable-gold --enable-plugins --enable-plugin --disable-multilib --disable-nls --enable-werror=no --enable-threads --enable-lto --with-float=hard --with-cpu=cortex-a7 --with-fpu=neon-vfpv4 --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --disable-bootstrap --enable-stage1-checking=release --with-sysroot=/ --with-default-libstdcxx-abi=new --disable-libitm --disable-libquadmath --enable-multiarch --disable-sjlj-exceptions --enable-gnu-unique-object --without-included-gettext --with-arch-directory=arm --program-suffix=-9 --program-prefix=arm-linux-gnueabihf- --target=arm-linux-gnueabihf
Thread model: posix
gcc version 9.1.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-mcpu=cortex-a7' '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7ve+simd'
 /usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/cc1 -quiet -v -imultilib . -imultiarch arm-linux-gnueabihf foo.c -quiet -dumpbase foo.c -mcpu=cortex-a7 -mfloat-abi=hard -mfpu=neon-vfpv4 -mtls-dialect=gnu -marm -march=armv7ve+simd -auxbase foo -version -o /tmp/ccz4VlV2.s
GNU C17 (GCC) version 9.1.0 (arm-linux-gnueabihf)
	compiled by GNU C version 6.3.0 20170516, GMP version 6.1.2, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/local/include/arm-linux-gnueabihf"
ignoring nonexistent directory "/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/include
 /usr/local/include
 /usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/include-fixed
 /usr/include/arm-linux-gnueabihf
 /usr/include
End of search list.
GNU C17 (GCC) version 9.1.0 (arm-linux-gnueabihf)
	compiled by GNU C version 6.3.0 20170516, GMP version 6.1.2, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: dcb0f0085577e8800cd1318f038382b3
COLLECT_GCC_OPTIONS='-v' '-mcpu=cortex-a7' '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7ve+simd'
 /usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/bin/as -v -march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -meabi=5 -o /tmp/cc1wO6u5.o /tmp/ccz4VlV2.s
GNU assembler version 2.32 (arm-linux-gnueabihf) using BFD version (GNU Binutils) 2.32
COMPILER_PATH=/usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/:/usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/:/usr/local/libexec/gcc/arm-linux-gnueabihf/:/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/:/usr/local/lib/gcc/arm-linux-gnueabihf/:/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/:/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/lib/:/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../../:/lib/arm-linux-gnueabihf/:/lib/:/usr/lib/arm-linux-gnueabihf/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-mcpu=cortex-a7' '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7ve+simd'
 /usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/collect2 -plugin /usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/liblto_plugin.so -plugin-opt=/usr/local/libexec/gcc/arm-linux-gnueabihf/9.1.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccP8NDNd.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --sysroot=/ --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X -m armelf_linux_eabi /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/arm-linux-gnueabihf/crti.o /usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/crtbegin.o -L/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0 -L/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../../../arm-linux-gnueabihf/lib -L/usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/../../.. -L/lib/arm-linux-gnueabihf -L/usr/lib/arm-linux-gnueabihf /tmp/cc1wO6u5.o -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/local/lib/gcc/arm-linux-gnueabihf/9.1.0/crtend.o /usr/lib/arm-linux-gnueabihf/crtn.o
COLLECT_GCC_OPTIONS='-v' '-mcpu=cortex-a7' '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7ve+simd'
```

abhiTronix, 2019-06-21:
> I just tested but this version doesn't have any libstdc++ headers packaged 

@jcelerier **It is not he the problem and it not nessary to have specific version of libstdc++. These GCC native binaries uses system libstdc++ for compiling code. you can easily confirm this as follow:**
```shell
pi@raspberrypi:~ $ /home/pi/native-pi-gcc-9.1.0-1/arm-linux-gnueabihf/libc/sbin/ldconfig -p | grep stdc++
    libstdc++.so.6 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
```
&nbsp; 

---

> so simple programs such as
> 
>  #include <string>
> 
> do not compile.

@jcelerier **This is not the error with my GCC Binaries only but official GCC distro also throws the similar error as above. You see the output with GCC Raspbian 6.3.0-18+rpi1+deb9u1 installed from official Raspbian distro:**
```shell
pi@raspberrypi:~ $ gcc -v test2.c
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 6.3.0-18+rpi1+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) 
COLLECT_GCC_OPTIONS='-v' '-march=armv6' '-mfloat-abi=hard' '-mfpu=vfp' '-mtls-dialect=gnu'
 /usr/lib/gcc/arm-linux-gnueabihf/6/cc1 -quiet -v -imultilib . -imultiarch arm-linux-gnueabihf test2.c -quiet -dumpbase test2.c -march=armv6 -mfloat-abi=hard -mfpu=vfp -mtls-dialect=gnu -auxbase test2 -version -o /tmp/ccvAD3ax.s
GNU C11 (Raspbian 6.3.0-18+rpi1+deb9u1) version 6.3.0 20170516 (arm-linux-gnueabihf)
    compiled by GNU C version 6.3.0 20170516, GMP version 6.1.2, MPFR version 3.1.5, MPC version 1.0.3, isl version 0.15
GGC heuristics: --param ggc-min-expand=81 --param ggc-min-heapsize=95700
ignoring nonexistent directory "/usr/local/include/arm-linux-gnueabihf"
ignoring nonexistent directory "/usr/lib/gcc/arm-linux-gnueabihf/6/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/arm-linux-gnueabihf/6/include
 /usr/local/include
 /usr/lib/gcc/arm-linux-gnueabihf/6/include-fixed
 /usr/include/arm-linux-gnueabihf
 /usr/include
End of search list.
GNU C11 (Raspbian 6.3.0-18+rpi1+deb9u1) version 6.3.0 20170516 (arm-linux-gnueabihf)
    compiled by GNU C version 6.3.0 20170516, GMP version 6.1.2, MPFR version 3.1.5, MPC version 1.0.3, isl version 0.15
GGC heuristics: --param ggc-min-expand=81 --param ggc-min-heapsize=95700
Compiler executable checksum: 06e7ff733c4083cc9ddae86e30496f12
test2.c:2:18: fatal error: string: No such file or directory
 #include <string>
                  ^
compilation terminated.
```

### Reason for this error: 
**`#include <string>` is a C++ directive. So either rename your file to test.cpp and compile it with `g++ test.cpp -o test` or rather use `#include "string.h"` instead. You can find more info here:** https://askubuntu.com/a/554357

&nbsp; 

---

> GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
> Compiler executable checksum: f6f989d3b086e434dc27bd601d90af60
> cc1plus: error: no include path in which to search for [...]

**This is the real error and I'm working on to solve it.**

&nbsp; 

---

> I've ended up building mine, here's what its configure line looks like :
>
> root@d769aeb0970e:/usr/local/bin# ./arm-linux-gnueabihf-gcc-9 foo.c -v
> Using built-in specs. [...]

**@jcelerier Yes obviously building native binaries on a Raspberry Pi OS will work for you. But this won't work in my case,  as I said earlier all of my native GCC binaries are cross-compiled using the Cross-Compiler available with this repository. So when we compile these binaries we need important system header files and libs to make it work correctly. As the location of header files differs in ARM(in raspberry pi) and AMD/Intel system, thereby it causes conflict with the path of these header files and libs. I'm trying to isolate paths as much as possible to make it work properly with maintaining support for default headers located at `/usr/include`.**

abhiTronix, 2019-06-22:
@jcelerier Finally I managed to solve it, Thanks for helping me out indirectly :) Kindly test this using following steps:
1. Make important symlinks as follows:(do not miss any)
  ```shell
  sudo ln -sf /usr/include/arm-linux-gnueabihf/asm /usr/include/asm
  sudo ln -sf /usr/include/arm-linux-gnueabihf/gnu /usr/include/gnu
  sudo ln -sf /usr/include/arm-linux-gnueabihf/bits /usr/include/bits
  sudo ln -sf /usr/include/arm-linux-gnueabihf/sys /usr/include/sys
  sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl
  sudo ln -sf /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
  sudo ln -sf /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
  sudo ln -sf /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
  ```

2. Download [this new GCC Native 9.1.0 Binary]() and extract it: 
   ```shell
   tar xf native-gcc-9.1.0-pi_2-3.tar.gz
   ```
**:warning: DO NOT move extracted folder to `/opt` as we want to check if it works without it.**

3. Make the necessary paths:(do Both of them)
  ```shell
  PATH=/<extracted folder-name e.g native-pi-gcc-9.1.0-1>/bin:$PATH
   LD_LIBRARY_PATH=/<extracted folder-name e.g native-pi-gcc-9.1.0-1>/lib:$LD_LIBRARY_PATH
  ```
4. Test your code:
   ```c
    #include <limits.h>
    char c[PATH_MAX]; 
   ```
   * When you will run this foo.c with `g++-9.1.0  foo.c`. This will throw a undefined error as follows:
     ```shell
     /usr/lib/crt1.o: In function `_start':
     (.text+0x34): undefined reference to `main'
     collect2: error: ld returned 1 exit status
     ```
   * **Reason for this error:** This code is missing `main()` function.
   * **Temporary Solution:** add  `-nostartfiles` to your command i.e run your command as `g++-9.1.0 -nostartfiles foo.c`
   * **Permanent Solution:** Modify code to add `main()` function as follows:
        ```c
        #include <limits.h>
        #include <stdio.h>
        int main(void)
        {
           char c[PATH_MAX]; 
           return 0;
          }
        ```
5 Test if rest of your header files works:
   ```c
   #include <type_traits>
   #include <openssl/ssl.h>
   int main()
   {
     return 0;
   }
   ```
 and test if this ran successfully on your machine with `g++-9.1.0 -v foo.c`
    

6. Finally ran your tests and share your results. Goodluck

abhiTronix, 2019-07-08:
**Closed. Feel feel to reopen if something doesn't work for you.**

closed 2019-07-08 by abhiTronix