abhiTronix/raspberry-pi-cross-compilers #9: Raspbian Stretch

TheJediMaster, 2019-02-20:
I have followed the guide using the native version with no luck.
pi@raspberrypi:~ $ gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I am sure I am missing something.

abhiTronix, 2019-02-20:
@TheJediMaster 
Can you provide the output of following command(just copy paste them in your raspberry pi terminal one-by-one):
`which gcc`
`which g++`
`gcc-8 --version`
`$PATH`
`cd /opt && ls`
and paste the output of each command here.

TheJediMaster, 2019-02-20:
Can you provide the output of following command(just copy paste them in your raspberry pi terminal one-by-one):
which gcc
/usr/bin/gcc
which g++
/usr/bin/g++
gcc-8 –version
bash: gcc-8: command not found
$PATH
bash: /usr/local/gcc-8.2.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games: No such file or directory
cd /opt && ls
gcc-8.2.0  pigpio  vc

abhiTronix, 2019-02-20:
> $PATH
> bash: /usr/local/gcc-8.2.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games: No such file or directory

@TheJediMaster Here is the problem!
`/usr/local/gcc-8.2.0/bin` where it came from?  

Try the following commands again:
   ```
echo 'export PATH=/opt/<extracted folder-name e.g gcc-8.2.0>/bin:$PATH' >> .bashrc
echo 'export LD_LIBRARY_PATH=/opt/<extracted folder-name e.g gcc-8.2.0>/lib:$LD_LIBRARY_PATH' >> .bashrc
```
and paste the output of `$PATH` here.

ItsBits, 2019-02-20:
Same problem on a fresh install.
```
$PATH
/opt/gcc-8.2.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games: No such file or directory
```

TheJediMaster, 2019-02-20:
Exactly,
This is on a blackberry 3B+
If I am in the directory, the correct version shows...…..so as I leave it its gone.

abhiTronix, 2019-02-20:
@TheJediMaster @ItsBits Yes, It's a bug, I'm working on it. Wait till I push the related commit. Will be fixed soon.

TheJediMaster, 2019-02-20:
Awesome.....otherwise I was going to have to do a complete install of gcc :)

abhiTronix, 2019-02-20:
@TheJediMaster @ItsBits 

***Follow Readme section again Carefully and revert***. Fixed in commit https://github.com/abhiTronix/raspberry-pi-cross-compilers/commit/645352297e1074e5a0ab27c354ace6a1914b8467

abhiTronix, 2019-02-21:
Closed. Feel free to open it, If any issue occurs.

closed 2019-02-21 by abhiTronix