abhiTronix/raspberry-pi-cross-compilers #10: g++-arm-linux-gnueabihf

TheJediMaster, 2019-02-22:
I attempted to install the below package..........is this normal or do you have a work=around for it?

pi@raspberrypi:~/$ sudo apt-get install g++-arm-linux-gnueabihf curl
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version (7.52.1-5+deb9u9).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 g++-arm-linux-gnueabihf : Depends: g++-5-arm-linux-gnueabihf (>= 5.2.1-13cross1) but it is not installable
E: Unable to correct problems, you have held broken packages.

abhiTronix, 2019-02-22:
```
I attempted to install the below package..........is this normal or do you have a work=around for it?

pi@raspberrypi:~/$ sudo apt-get install g++-arm-linux-gnueabihf curl
Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version (7.52.1-5+deb9u9).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
g++-arm-linux-gnueabihf : Depends: g++-5-arm-linux-gnueabihf (>= 5.2.1-13cross1) but it is not installable
E: Unable to correct problems, you have held broken packages.
``` 
This package is not available for Raspberry Pi and has Nothing to do with this repo. **Kindly strictly follow the instructions given the ReadMe section only.**

TheJediMaster, 2019-02-22:
My apologies and thank-you for resolving the last issue so quickly.
It worked perfectly.

closed 2019-02-22 by TheJediMaster