abhiTronix/raspberry-pi-cross-compilers #13: Pi zero usage ?

jimcruse, 2019-03-01:
I just got a new pi zero, I'm willing to test these versions on it, should I use these as they are that worked pi3b, or is there some other version?
Are these 64 bit or 32, pi zero requires 32 bit I read somewhere

abhiTronix, 2019-03-01:
@jimcruse 

> Are these 64 bit or 32, pi zero requires 32 bit I read somewhere

Current Binaries are 32 bit. Also, these binaries aren't raspberry zero optimized yet.  ***Please wait till tomorrow(maybe today), I'm launching new `GCC 8.3.0 binaries`(Both native and cross-compiler) for every raspberry pi Models(including Rpi zero) seperately.*** Kuddos

jimcruse, 2019-03-01:
Ok I'll wait till this weekend, to download

jimcruse, 2019-03-02:
I know it's early, but I didn't see any update, can you send me a note,
once they are updated...,

abhiTronix, 2019-03-03:
I'm working on it. Also this issue will be closed once update is done.

jimcruse, 2019-03-06:
Please close

abhiTronix, 2019-03-06:
Closed in 2f87bc2899512822073d8228168c0f7da62154e2

closed 2019-03-06 by abhiTronix