abhiTronix/raspberry-pi-cross-compilers #18: Your readme.md links to the GitHub wiki, which is empty

jimcruse, 2019-03-24:
Inside the readme.md where it says to be sure to read carefully for native instructions

jimcruse, 2019-03-24:
https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Native-Compiler:-Linking-&-Testing-Instructions

abhiTronix, 2019-03-24:
> Your readme.md links to the GitHub wiki, which is empty
> Inside the readme.md where it says to be sure to read carefully for native instructions
> https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Native-Compiler:-Linking-&-Testing-Instructions
@jimcruse what do you mean by empty??? Links are all working fine.  **Share a screenshot.**

abhiTronix, 2019-03-24:
@jimcruse I'm closing this due to lack of information, Kindly provide proper & detailed information to elaborate your issue. Goodluck.

closed 2019-03-24 by abhiTronix