abhiTronix/raspberry-pi-cross-compilers #19: Installed on raspberry pi 3b+ and to make it work
Had to make a link to g++ because the files in the bin
Have a different name g++-8.3.0

jimcruse, 2019-03-24:
But after adding the g++ link in the bin directory it worked fine 
Your documentation says to check ld and ldd --version
But doesn't say what the answer should be...
And a little more on how to check gfortran is installed correctly
##

abhiTronix, 2019-03-24:
> But after adding the g++ link in the bin directory it worked fine
> Your documentation says to check ld and ldd --version
> But doesn't say what the answer should be...
> And a little more on how to check gfortran is installed correctly

**Do not open multiple issues. Kindly list all the related issue under one.** Also, instructions are already [here](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Native-Compiler:-Linking-&-Testing-Instructions) in detail. **NO LINKING whatsoever required. you can use binaries temporarily at the time of compilation by following step-6 of instruction without breaking your system.**

closed 2019-03-24 by abhiTronix

jimcruse, 2019-03-24:
Step 6 sets the environment variables for autotools, but not command line directly

abhiTronix, 2019-03-24:
> Step 6 sets the environment variables for autotools, but not command line directly

@jimcruse Yes, but **why you need them on the command line? What is the purpose?**. GCC tools are used for compiling programs only, isn't it. Anyways, you can already check its functioning with these commands as stated in the [wiki](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Native-Compiler:-Linking-&-Testing-Instructions#testing-post-linking) in case you missed as follows:

```
gcc-8.3.0 --version

g++-8.3.0 --version

gfortran-8.3.0 --version
```
**_If you experience any difficulty in executing these commands, provided you followed the wiki correctly, then there is a point of opening of issue there._**