abhiTronix/raspberry-pi-cross-compilers #21: Request for GDB

raitraak-rrk, 2019-04-17:
Hi,

first of all thanks for this project! It's very useful to quickly get started with Pi dev.

I noticed there is no GDB executable included in the cross compile binaries. Is it possible to add it there? For now I am using the one included in official arm-linux-gnueabihf and it seems to work.

Cheers,
Rait

abhiTronix, 2019-04-17:
@raitraak-rrk Didn't we cross-compile GDB afterward using cross-compiler toolchain? **GDB executable doesn't come inbuilt in my knowledge.**

## Solution:

You can easily use my Cross-compiler Toolchains to cross-compile GDB for your Raspberry Pi. If need step-by-step instructions, then kindly let me know.

raitraak-rrk, 2019-04-17:
It would be nice if GDB is a part of the binary distribution as in the package from ARM.
[image]

abhiTronix, 2019-04-17:
> It would be nice if GDB is a part of the binary distribution as in the package from ARM.

Yes, that would be good, But Adding GDB executables to each cross-compiler is quite a difficult and cumbersome task, might need recompiling and re-uploading those files from scratch which is not currently feasible.

**_Therefore I think, I'll add a separate folder for GCC Cross-compiler-8.3.0 built GDB executables for each Raspberry Pi Model. I'll commit the change and will notify you here once done. Stay tuned._**

raitraak-rrk, 2019-04-18:
No rush. I got gdb from ARM package. Simply a suggestion.

Maybe preserve the directory structure too, so unpacking puts gdb in the bin folder.

Thanks again for the project!

abhiTronix, 2019-04-18:
> Maybe preserve the directory structure too, so unpacking puts gdb in the bin folder.

@raitraak-rrk Yes good point, But that's will certainly need another update i.e building every file from scratch and the re-uploading whole project. **_I'll suggest you to keep using those alternate GDB executables for now. I'll add this feature to Milestone and will include this next major update that is coming soon. Also, leave this issue open, I'll close this once the update is done. Goodluck_**

abhiTronix, 2019-05-18:
@raitraak-rrk **New GDB 8.3.1 Debugger has been successfully Integrated into New Cross-Compiler Release. Kindly Test it and Revert.**

abhiTronix, 2019-05-18:
Closed in 7c40f92f9d0a003cd7504c37cc3734920223210a

closed 2019-05-18 by abhiTronix

raitraak-rrk, 2019-05-18:
Looks good! Thanks, mate.

abhiTronix, 2019-06-19:
**Update v2.8:** Released Native Compiler Toolchains with InBuilt GDB Binaries for Debugging.