abhiTronix/raspberry-pi-cross-compilers #22: Raspberry Pi Model 3 A+/B+ GCC 64bit?

TheDevMinerTV, 2019-05-04:
Is the gcc for Raspberry Pi Model 3 A+/B+ compiled against 64bit libraries?

ScallyBag, 2019-05-04:
Raspberry Pi’s are 32 bit compiles.

Al (I compile for Picochess)

TheDevMinerTV, 2019-05-04:
Nope, the 3 A+/B+ are 64bit capable, just not under raspbian.
Maybe @abhiTronix can compile the bins for the 3 A+/B+ next time on pi64.

abhiTronix, 2019-05-04:
@TheDevMinerTV, @ScallyBag you both are correct.

But, If you look at the ReadMe section [here](https://github.com/abhiTronix/raspberry-pi-cross-compilers#optimation-flags-involved), the compiler flags used for **Raspberry Pi - 3 Model A+/B+** are `-march=armv8-a -mfloat-abi=hard -mfpu=neon-fp-armv8` :

- First compiler flag, `-march=armv8-a` means ARMv8 (_a.k.a ARMv8-A  since there are also ARMv8-R and
ARMv8-M specifications as well_) is given architecture for our GCC binaries. The ARMv8 architecture defines two execution modes: AArch32 and AArch64. The AArch32 execution mode further has to states with separate ISAs: A32 and T32, more traditionally known as ARM and Thumb states.

- **But if you see the second flag, ` -mfpu=neon-fp-armv8` is specifically an arm (32-bit) compiler option .**  The aarch64 (64-bit) compiler, therefore, cannot accept this flag.

**Therefore the Raspberry Pi - 3 Model A+/B+ binaries in this repo. are currently compiled against 32-bit libraries only and NOT 64bit ones since these GCC binaries are built for especially for Raspberry Pi only which supports 32-bit compilers only as of the present.**

> Maybe @abhiTronix can compile the bins for the 3 A+/B+ next time on pi64.
@TheDevMinerTV Thanks for suggesting, I'll put this into next milestone.

TheDevMinerTV, 2019-05-04:
Okay, cool! I hope I could help with this recommendation.

abhiTronix, 2019-05-05:
@TheDevMinerTV Thankyou, I'll let you know here when this feature is added and will require your assistance in testing those binaries. good luck!

TheDevMinerTV, 2019-05-06:
Ummmm, forget the 64bit version of GCC when built under pi64.
The userland of pi64 is 32bit...
Well, thanks for the offer.

abhiTronix, 2019-05-06:
@TheDevMinerTV the [readme of Pi64](https://github.com/bamarni/pi64#can-i-still-run-32-bit-programs-with-pi64) states:
> Can I still run 32-bit programs with pi64?
> You should be able to run 32-bit programs out of the box as long as they're statically linked. You can check this with the file command :
> 
> $ file ./my-executable
> ./my-executable: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, not stripped
> If your 32-bit program is shown as dynamically linked instead, you should still be able to run it by enabling multiarch and installing program's required libraries :
> 
> sudo dpkg --add-architecture armhf
> sudo apt-get update
> sudo apt-get install libc6:armhf
> Here we're only installing the GNU C Library, but your program might need additional libraries.

**It may have work in the first place but may not optimized as on the 32-bit OS. That's why I insisted to build 64bit binaries. I also believe that the upcoming Raspberry 4 release will be in 64-bit. Anyways Kindly feel free to reopen this issue if anyone else wants this feature.**

abhiTronix, 2019-05-13:
Building this experimental feature!

TheDevMinerTV, 2019-05-13:
Okay, so, when you need to compile GCC on Barmani Full 64bit from CrazyHead90 otherwise the resulting GCC will be a 32bit aarch64 binary.

abhiTronix, 2019-05-15:
> Okay, so, you need to compile GCC on Barmani Full 64bit from CrazyHead90 otherwise the resulting GCC will be a 32bit aarch64 binary.
> I've experienced this in pain...

False, I didn't use any  `Barmani Full 64bit from CrazyHead90` and I've succeeded in compiling true Aarch64 GCC binaries for 64-bit ARM OS from scratch on my Raspberry Pi 3B+. Wait for them, updating repo. soon.

TheDevMinerTV, 2019-05-17:
Okay, lol

abhiTronix, 2019-05-18:
Closed in 7c40f92f9d0a003cd7504c37cc3734920223210a

closed 2019-05-18 by abhiTronix