abhiTronix/raspberry-pi-cross-compilers #26: Generic CrossCompiler

jano42, 2019-08-23:
Hello,

You're job is great.

I'm searching for raspberry cross toolchain based on a recent gcc (>8) which could run on old raspbian (jessie).
And a single toolchain for all RaspberryPI models.

I'm developing a [domotic software ](http://yadoms.com), and I would like to provide binaries which could run on any raspberrypi (from 1 to 4, which potentially old raspbian like jessie, without any specific optimizations).

And I need a recent gcc to support c++11/17 features (actually I'm using https://github.com/raspberrypi/tools but with gcc4.9.3)

So my question, do you provide such crosstoolchain ? if yes which one ?

Regards

abhiTronix, 2019-08-23:
Hello @jano42, 

> I'm searching for raspberry cross toolchain based on a recent gcc (>8) which could run on old raspbian (jessie). And a single toolchain for all RaspberryPI models.

This project provides separately optimized toolchains for all Raspberry pi models with any Raspbian OS installed, both 64bit/32bit are supported. _**But since you asked for single toolchain that works with all models then I'll advice to go for latest [Raspberry Pi 32-bit 9.1.0 toolchain for Raspberry Pi 3B+](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/GCC%209.1.0/Raspberry%20Pi%203A%2B%2C%203B%2B/) which can support any Rpi model from zero to 4. You can find further instructions to set this toolchain up [here](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler:-Installation-Instructions)**_  

**⚠️ But Remember NOT to specify any of [these optimization flags](https://github.com/abhiTronix/raspberry-pi-cross-compilers#optimization-flags-involved) in your build as they are system specific.**

Keep me Inform in case of any problems. Goodluck

abhiTronix, 2019-08-23:
Closed. Feel free to reopen, If something doesn't works.

closed 2019-08-23 by abhiTronix

jano42, 2019-08-26:
Hi abhiTronix,

Thanks for your answer. I've started to use the toolchain.

I'm facing a problem with python (never with any other crosscompilations tries)

/usr/local/include/python3.7m/pyport.h:699:2: error: #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
  699 | #error "LONG_BIT definition appears wrong for platform (bad gcc/glibc config?)."
      |  ^~~~~

Do you have any idea about that ?

abhiTronix, 2019-08-26:
@jano42 **Make sure you have the right bits-installed for python.** You will need the python 32-bit installed for the 32-bit Raspberry Pi Cross-compilers, otherwise, this issue will arise.

jano42, 2019-08-26:
I've tried to install the crosstoolchain on a Debian Strech 32bits (previous tries were made on 64bits env), but gcc/g++ executables are 64bits and not recognized by system.

[image]

jano42, 2019-08-26:
ok I'll check about installed python version

abhiTronix, 2019-08-26:
@jano42 You need to set links to be able to use these binaries correctly: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler:-Installation-Instructions#1-temporary-installation-use-these-binaries-directly-recommended 

> I've tried to install the crosstoolchain on a Debian Stretch 32bits (previous tries were made on 64bits env), but gcc/g++ executables are 64bits and not recognized by system.

That's not the case, these cross-toolchains are not dependent of OS architecture on which these are being used on. I've tested them on both 32-bit and 64-bit OSes without any issues.

abhiTronix, 2019-08-26:
@jano42 I've solved this problem, **you must also need cross-compile python using the 32-bit Raspberry Pi Cross-compiler to remove this error. Follow these instructions in the link to do it:** https://github.com/Yadoms/yadoms/wiki/Cross-compile-for-raspberry-PI#python . These instructions are for python 2.7 but you can easily change them to compile python 3 also. Goodluck.!

jano42, 2019-08-27:
@abhiTronix : I've already cross compile python for arm (I wrote the wiki link you provide me, I'm one of Yadoms developer). 
https://github.com/Yadoms/docker/blob/feature/upgradeContainers/build-raspberrypi/Dockerfile#L136

I'll look about cross-compiling python3 in 32bits mode

abhiTronix, 2019-08-27:
Good to know. It should work after compilation. Goodluck.

jano42, 2019-08-28:
OK finally I make it build !!!

Then when I take my executable on my Raspberry PI 1 with raspian Jessie installed I've got those errors:

```
./yadoms: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ./yadoms)
./yadoms: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by ./yadoms)
./yadoms: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by ./yadoms)

```

What should I do ? (provide arm-linux-gnueabihf argumzent to compile with old glibcxx ??)

jano42, 2019-08-28:
@abhiTronix The GCC version is 4.9.2

jano42, 2019-08-28:
@abhiTronix I don't know how do you generate toolchains, but is there a way to specify the minimum compatible glibcxx version ? (I've made a try with crosstoolng but without succes)

abhiTronix, 2019-08-28:
You need latest raspbian libstdc++6 binaries on your raspberry pi too. I meant you need to download the latest [native 32-bit 9.1.0 GCC binaries](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Native-Compiler%20Toolchains/GCC%209.1.0/) and do the [permanent installation step](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Native-Compiler:-Installation-Instructions#raspberry-pi-native-compiler-toolchain-installation-instruction-) on your raspberry pi i.e. you have to replace old GCC 6.3.0 with latest 9.1.0.

abhiTronix, 2019-08-28:
@jano42 Also, What is the default GCC version on raspbian Jessie?

abhiTronix, 2019-08-28:
@jano42 then it's bad news, as these toolchains are built on GCC 6.3.0 as the base GCC. If you could somehow update to stretch or buster than there will be no problem at all. But you try replacing old GCC 4.9.3 with GCC 9.1.0.

abhiTronix, 2019-08-28:
> but is there a way to specify the minimum compatible glibcxx version ?

 No, It is built along with compilers and cannot be altered later on.

abhiTronix, 2019-08-28:
@jano42 If you are trying to build Glibc 2.24 with a more modern GCC, like 7.x or 8.x, you are going to get a lot of errors and compatibility issues. The solution is to build Glibc with GCC 6.3.0 and later use it with the latest and greatest GCC, which at this time is 9.1. That's what I did while building these cross compilers.

jpires, 2019-09-06:
@jano42 One possible solution for the missing GLIBCXX_3.4.26 is to bundle the library and the executable together and use the LD_LIBRARY_PATH to change the order the libraries are loaded.

For example, On /home/bar/ you have a foo binary and the libstd++.
You can execute foo with this command
`LD_LIBRARY_PATH=/home/bar /home/bar/foo`

abhiTronix, 2019-09-06:
@jpires just for the knowledge, Building libstdc++ binaries require building the entire gcc package. You cannot just build libstdc++ alone.

jano42, 2020-10-12:
Hi @abhiTronix 

Last year I didn't manage with GLIBCXX link problems, then I didn't use your toolchain.

Since one month I've successfully fixed the glibcxx for another reason; simply by linking statically with glibc and glibcxx.

So I would like to try a new time your toolchains, but when I browse your files I can see that there is know a difference between Stretch and Buster distro.

Could you tell me what are those differences ?

I would like a single toolchain, which produce my software binaries compatible for all Raspberry hardwares, and also all distro (from jessie to last one).
Last year you told me to use any the gcc 9 for Pi3+; and without LTO it should work.
So my question is which toolchain between Strech or Buster should i take ?

Thanks

abhiTronix, 2020-10-14:
>  difference between Stretch and Buster distro.
> 
> Could you tell me what are those differences ?
> 

@jano42 Stretch and Buster Distros are later version of Official Raspbian Debian OS, next to Jessie Distro. Since, default Kernel, Glibc and Binutils(required for GCC to work) version is different on all distros therefore you cannot compile same GCC binary for all of them, and hence this repository provide different GCC binaries compatible with different distros. **Beware, If you've Glibc version mismatch between your target Pi OS and GCC Binary, then it will throw `GLIBCXX link problems` as seen your case, therefore you must only use binary compatible with your Raspberry Pi OS distro.**

> I would like a single toolchain, which produce my software binaries compatible for all Raspberry hardwares, and also all distro (from jessie to last one).

As explained above, you cannot have same GCC binary for all distros at same time. Also jessie is not supported, only later  version of Official Raspbian Debian OS ie. Stretch and Buster _(latest one)_ are supported with this repository.

> So my question is which toolchain between Strech or Buster should i take ?

Choose GCC toolchain w.r.t distro installed on your Target Raspberry Pi for which you're compiling software for.