abhiTronix/raspberry-pi-cross-compilers #27: Cannot find crt1.o

gfplab-scaffardi, 2019-09-04:
Hi there! Thank you for your work.
I am trying to use the armv6 gcc 8.3 toolchain to cross compile QTfor rpi zero (rpi0w) target. I followed the wiki and installed. I can build an example program but if i configure QT it gives me problems while trying to build their sample programs. The logs report this error:

`arm-linux-gnueabihf-g++ -mfloat-abi=hard --sysroot=/home/parallels/raspi/sysroot --sysroot=/home/parallels/raspi/sysroot -Wl,-O1 -Wl,-rpath-link,/home/parallels/raspi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf -o verifyspec verifyspec.o -v
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-g++
COLLECT_LTO_WRAPPER=/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/opt/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-multilib
Thread model: posix
gcc version 8.3.0 (GCC) 
COMPILER_PATH=/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/:/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/:/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/:/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/:/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/:/home/parallels/raspi/sysroot/lib/:/home/parallels/raspi/sysroot/usr/lib/
COLLECT_GCC_OPTIONS='-mfloat-abi=hard' '-o' 'verifyspec' '-v' '-shared-libgcc'  '-mfpu=vfp' '-mtls-dialect=gnu' '-marm' '-march=armv6+fp'
 /home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/collect2 -plugin /home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/liblto_plugin.so -plugin-opt=/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccPG7cKK.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --sysroot=/home/parallels/raspi/sysroot --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X -m armelf_linux_eabi -o verifyspec crt1.o crti.o /home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/crtbegin.o -L/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0 -L/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc -L/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib -L/home/parallels/raspi/sysroot/lib -L/home/parallels/raspi/sysroot/usr/lib -O1 -rpath-link /home/parallels/raspi/sysroot/opt/vc/lib -rpath-link /home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -rpath-link /home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf verifyspec.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/crtend.o crtn.o
/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: File o directory non esistente
/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: File o directory non esistente
/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status`

As you can see, QT cross compilation is using sysroot gcc option (/home/parallels/raspi/sysroot is my sysroot). I placed the toolchain exactly into the sysroot (/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0). Why the crt1,crti objects + m library are not found? #

abhiTronix, 2019-09-04:
@gfplab-scaffardi looks like you completely messed up up your system paths. 

Kindly answer the following queries for further debugging:

**1. Did you strictly followed the instructions given the [wiki section](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler:-Installation-Instructions#raspberry-pi-cross-compiler-toolchain-installation-instruction-)?**

**2. Kindly cd into bin directory(cross-gcc-8.3.0-pi_0-1/bin) and paste this command:** 
```sh
./arm-linux-gnueabihf-gcc -print-search-dirs | sed '/^lib/b 1;d;:1;s,/[^/.][^/]*/\.\./,/,;t 1;s,:[^=]*=,:;,;s,;,;  ,g' | tr \; \\012
```
**and share the output here.**

**3 Why are you setting sysroot? How do you came to conclusion to use sysroot while compiling QT? If you copied these instructions from a online source then kindly share that source here.**

gfplab-scaffardi, 2019-09-04:
> @gfplab-scaffardi looks like you've completely messed up your system paths.
> 
> ### Kindly answer the following queries for further debugging:
> **1. Did you strictly followed the instructions given the [Wiki Section](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler:-Installation-Instructions#raspberry-pi-cross-compiler-toolchain-installation-instruction-) before using this Cross-Compiler?**

Yes i did. Compiling without the sysroot argument works fine.

> **2. Kindly cd into bin directory(cross-gcc-8.3.0-pi_0-1/bin) and paste this command:**
> 
> ```shell
> ./arm-linux-gnueabihf-gcc -print-search-dirs | sed '/^lib/b 1;d;:1;s,/[^/.][^/]*/\.\./,/,;t 1;s,:[^=]*=,:;,;s,;,;  ,g' | tr \; \\012
> ```
> 
> **and share the output here.**

`libraries:
  /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/lib/gcc/arm-linux-gnueabihf/8.3.0/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/lib/gcc/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/lib/arm-linux-gnueabihf/8.3.0/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/lib/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc/lib/arm-linux-gnueabihf/8.3.0/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc/lib/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc/usr/lib/arm-linux-gnueabihf/8.3.0/:/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc/usr/lib/`

> **3. Why are you setting sysroot? How do you came to conclusion to use sysroot while cross-compiling QT? If you copied these instructions from a online source then kindly share that source here.**
> **4. Also, Kindly share the instructions you are using to cross-compile QT for your Raspberry Pi**

I followed two guides ([this one](https://mechatronicsblog.com/cross-compile-and-deploy-qt-5-12-for-raspberry-pi/) and [this second one](https://mechatronicsblog.com/cross-compile-and-deploy-qt-5-12-for-raspberry-pi/)) to cross compile QT for RPI. I could successfully build qt 5.12.3 with both linaro (gcc 7.2, armv7 + armv8 only) and RPI tools standard toolchains (gcc 4.6) but no way to build qtwebengine module, because it needs gcc > 5.0. That's why i tried your toolchain that should meet both requirements (armv6 target + gcc > 5.0).

abhiTronix, 2019-09-04:
Where your Cross-Compiler actually toolchains located. According to searchpath, it is here:

> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/

and according to you, You placed your toolchain here:

>  I placed the toolchain exactly into the sysroot (/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0).

Why these location not same?

gfplab-scaffardi, 2019-09-04:
> @gfplab-scaffardi Where your Cross-Compiler toolchains actually located? **According to system searchpath, it is here:**
> 
> > /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/
> 
> **and according to you, You placed your toolchain here:**
> 
> > I placed the toolchain exactly into the sysroot (/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0).
> 
> **Why these location not same? or Are you having two toolchains on your system?**

You are right. Sorry. I initially installed in /opt, then here:

`/home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/ `

And, just to perform a test, i copied here too:

'/home/parallels/raspi/sysroot/opt/cross-pi-gcc-8.3.0-0'

Now it's back in /opt and i also added permanent LD_LIBRARY_PATH and PATH variables into .bashrc (they are correctly set into the environments). These are the updated results.

The compiler output:

`arm-linux-gnueabihf-g++ -v -Wl,--verbose -mfloat-abi=hard --sysroot=/home/parallels/raspi/sysroot -Wl,-O1 -Wl,-rpath-link,/home/parallels/raspi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf -o verifyspec verifyspec.o      
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-g++
COLLECT_LTO_WRAPPER=/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/opt/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-multilib
Thread model: posix
gcc version 8.3.0 (GCC) 
COMPILER_PATH=/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/:/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/:/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/:/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/:/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/:/home/parallels/raspi/sysroot/lib/:/home/parallels/raspi/sysroot/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-mfloat-abi=hard' '-o' 'verifyspec' '-shared-libgcc'  '-mfpu=vfp' '-mtls-dialect=gnu' '-marm' '-march=armv6+fp'
 /opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/collect2 -plugin /opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/liblto_plugin.so -plugin-opt=/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper -plugin-opt=-fresolution=/tmp/cc82O1td.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --sysroot=/home/parallels/raspi/sysroot --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X -m armelf_linux_eabi -o verifyspec crt1.o crti.o /opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/crtbegin.o -L/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0 -L/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc -L/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib -L/home/parallels/raspi/sysroot/lib -L/home/parallels/raspi/sysroot/usr/lib --verbose -O1 -rpath-link /home/parallels/raspi/sysroot/opt/vc/lib -rpath-link /home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -rpath-link /home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf verifyspec.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/crtend.o crtn.o
GNU ld (GNU Binutils) 2.28
  Supported emulations:
   armelf_linux_eabi
   armelfb_linux_eabi
using internal linker script:
==================================================
/* Script for -z combreloc: combine and sort reloc sections */
/* Copyright (C) 2014-2017 Free Software Foundation, Inc.
   Copying and distribution of this script, with or without modification,
   are permitted in any medium without royalty provided the copyright
   notice and this notice are preserved.  */
OUTPUT_FORMAT("elf32-littlearm", "elf32-bigarm",
	      "elf32-littlearm")
OUTPUT_ARCH(arm)
ENTRY(_start)
SEARCH_DIR("/arm-linux-gnueabihf/lib");
SECTIONS
{
  /* Read-only sections, merged into text segment: */
  PROVIDE (__executable_start = SEGMENT_START("text-segment", 0x00010000)); . = SEGMENT_START("text-segment", 0x00010000) + SIZEOF_HEADERS;
  .interp         : { *(.interp) }
  .note.gnu.build-id : { *(.note.gnu.build-id) }
  .hash           : { *(.hash) }
  .gnu.hash       : { *(.gnu.hash) }
  .dynsym         : { *(.dynsym) }
  .dynstr         : { *(.dynstr) }
  .gnu.version    : { *(.gnu.version) }
  .gnu.version_d  : { *(.gnu.version_d) }
  .gnu.version_r  : { *(.gnu.version_r) }
  .rel.dyn        :
    {
      *(.rel.init)
      *(.rel.text .rel.text.* .rel.gnu.linkonce.t.*)
      *(.rel.fini)
      *(.rel.rodata .rel.rodata.* .rel.gnu.linkonce.r.*)
      *(.rel.data.rel.ro .rel.data.rel.ro.* .rel.gnu.linkonce.d.rel.ro.*)
[... 258 lines cut ...]

> **Also kindly paste the output of following command:**
> 
> ```shell
> echo $PATH | tr ":" "\n" | nl
> ```

```
      1	/opt/cross-pi-gcc-8.3.0-0/bin
     2	/usr/local/bin
     3	/usr/bin
     4	/bin
     5	/usr/local/games
     6	/usr/games
     7	/home/parallels/Programs/arm/4.1.1-920t/bin
     8	/home/parallels/Programs/arm/3.2.1-elf/bin
     9	/home/parallels/Programs/ba-elf-ba2/bin
    10	/home/parallels/Programs/avr/bin
    11	/home/parallels/Programs/android-sdks/tools
    12	/home/parallels/Programs/esp/xtensa-esp32-elf/bin
    13	/home/parallels/Programs/gradle/bin

```

It seems that the toolchain files are not found because out of the sysroot.

abhiTronix, 2019-09-04:
@gfplab-scaffardi Try again and remove `-shared-libgcc` flag this time and revert.

abhiTronix, 2019-09-04:
@gfplab-scaffardi Wait, I'm compiling QT at my end and if successful I'll post my configuration here. Till then keep trying at your end too.

abhiTronix, 2019-09-04:
@gfplab-scaffardi I found a hack, **Kindly Remove all paths from `.bashrc` and reboot and then replace its current location in the following command:**

```sh
PATH=/<extracted folder-name cross-pi-gcc-8.3.0-0>/bin:$PATH
LD_LIBRARY_PATH=/<extracted folder-name e.g cross-pi-gcc-8.3.0-0>/lib:$LD_LIBRARY_PATH
LD_LIBRARY_PATH=/<extracted folder-name e.g cross-pi-gcc-8.3.0-0>/arm-linux-gnueabihf/lib/:$LD_LIBRARY_PATH
LD_LIBRARY_PATH=/<extracted folder-name e.g cross-pi-gcc-8.3.0-0>/arm-linux-gnueabihf/libc/usr/lib:$LD_LIBRARY_PATH
LD_LIBRARY_PATH=/<extracted folder-name e.g cross-pi-gcc-8.3.0-0>/arm-linux-gnueabihf/libc/lib:$LD_LIBRARY_PATH
```
**Then rerun `./configure` command and revert your results goodluck.**

gfplab-scaffardi, 2019-09-05:
> @gfplab-scaffardi I found a hack, **Kindly Remove all paths from `.bashrc` and reboot and then replace its current location in the following commands and execute them in your terminal one-by-one:**
> 
> ```shell
> PATH=/<extracted folder-path cross-pi-gcc-8.3.0-0>/bin:$PATH
> LD_LIBRARY_PATH=/<extracted folder-path e.g cross-pi-gcc-8.3.0-0>/lib:$LD_LIBRARY_PATH
> LD_LIBRARY_PATH=/<extracted folder-path e.g cross-pi-gcc-8.3.0-0>/arm-linux-gnueabihf/lib/:$LD_LIBRARY_PATH
> LD_LIBRARY_PATH=/<extracted folder-path e.g cross-pi-gcc-8.3.0-0>/arm-linux-gnueabihf/libc/usr/lib:$LD_LIBRARY_PATH
> LD_LIBRARY_PATH=/<extracted folder-path e.g cross-pi-gcc-8.3.0-0>/arm-linux-gnueabihf/libc/lib:$LD_LIBRARY_PATH
> ```
> 
> **Then rerun `./configure` command and revert your results goodluck.**

I did it. This are the new environment variables:

```
echo $LD_LIBRARY_PATH
/opt/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc/lib:/opt/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc/usr/lib:/opt/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/lib:/opt/cross-pi-gcc-8.3.0-0/lib:

```

Tried it. Same results here (the same cmd works without sysroot option):

```
arm-linux-gnueabihf-g++ -v -Wl,--verbose -mfloat-abi=hard --sysroot=/home/parallels/raspi/sysroot -Wl,-O1 -Wl,-rpath-link,/home/parallels/raspi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf -o verifyspec verifyspec.o      
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-g++
COLLECT_LTO_WRAPPER=/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/opt/cross-pi-gcc-8.3.0-0/arm-linux-gnueabihf/libc --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-multilib
Thread model: posix
gcc version 8.3.0 (GCC) 
COMPILER_PATH=/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/:/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/:/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/:/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/:/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/:/home/parallels/raspi/sysroot/lib/:/home/parallels/raspi/sysroot/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-mfloat-abi=hard' '-o' 'verifyspec' '-shared-libgcc'  '-mfpu=vfp' '-mtls-dialect=gnu' '-marm' '-march=armv6+fp'
 /opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/collect2 -plugin /opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/liblto_plugin.so -plugin-opt=/opt/cross-pi-gcc-8.3.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/8.3.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccQOk0gS.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --sysroot=/home/parallels/raspi/sysroot --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X -m armelf_linux_eabi -o verifyspec crt1.o crti.o /opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/crtbegin.o -L/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0 -L/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc -L/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib -L/home/parallels/raspi/sysroot/lib -L/home/parallels/raspi/sysroot/usr/lib --verbose -O1 -rpath-link /home/parallels/raspi/sysroot/opt/vc/lib -rpath-link /home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -rpath-link /home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf verifyspec.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/crtend.o crtn.o
GNU ld (GNU Binutils) 2.28
  Supported emulations:
   armelf_linux_eabi
   armelfb_linux_eabi
using internal linker script:
==================================================
/* Script for -z combreloc: combine and sort reloc sections */
/* Copyright (C) 2014-2017 Free Software Foundation, Inc.
   Copying and distribution of this script, with or without modification,
   are permitted in any medium without royalty provided the copyright
   notice and this notice are preserved.  */
OUTPUT_FORMAT("elf32-littlearm", "elf32-bigarm",
	      "elf32-littlearm")
OUTPUT_ARCH(arm)
ENTRY(_start)
SEARCH_DIR("/arm-linux-gnueabihf/lib");
SECTIONS
{
  /* Read-only sections, merged into text segment: */
  PROVIDE (__executable_start = SEGMENT_START("text-segment", 0x00010000)); . = SEGMENT_START("text-segment", 0x00010000) + SIZEOF_HEADERS;
  .interp         : { *(.interp) }
  .note.gnu.build-id : { *(.note.gnu.build-id) }
  .hash           : { *(.hash) }
  .gnu.hash       : { *(.gnu.hash) }
  .dynsym         : { *(.dynsym) }
  .dynstr         : { *(.dynstr) }
  .gnu.version    : { *(.gnu.version) }
  .gnu.version_d  : { *(.gnu.version_d) }
  .gnu.version_r  : { *(.gnu.version_r) }
  .rel.dyn        :
    {
      *(.rel.init)
      *(.rel.text .rel.text.* .rel.gnu.linkonce.t.*)
      *(.rel.fini)
      *(.rel.rodata .rel.rodata.* .rel.gnu.linkonce.r.*)
[... 238 lines cut ...]
attempt to open /arm-linux-gnueabihf/lib/libm.a failed
[... cut ...]
/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: File o directory non esistente
/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: File o directory non esistente
/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status
Makefile:68: set di istruzioni per l'obiettivo "verifyspec" non riuscito
make: *** [verifyspec] Errore 1

```

Thank you again for your time. Any idea?

abhiTronix, 2019-09-05:
@gfplab-scaffardi This is caused by sysroot. The path in which it is searching for libs are incorrect:
`/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/` instead it should search in `/opt/cross-pi-gcc-8.3.0-0/bin/../arm-linux-gnueabihf/libc` which is the actual sysroot of these binaries. I'll try to replicate this on my machine, Kindly paste your all terminal commands used in this here so I can cross-check them one-by-one. Thank you.

gfplab-scaffardi, 2019-09-05:
> @gfplab-scaffardi This is maybe caused by custom sysroot. The path in which it is searching for libs are incorrect:
> `/opt/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/` instead it should search in `/opt/cross-pi-gcc-8.3.0-0/bin/../arm-linux-gnueabihf/libc` which is the actual sysroot of these binaries. I'll try to replicate this on my machine, Kindly paste your all terminal commands used in this compiling, here so I can cross-check them one-by-one. Thank you.

```
wget http://download.qt.io/official_releases/qt/5.12/5.12.3/single/ qt-everywhere-src-5.12.3.tar.xz
tar xvf  qt-everywhere-src-5.12.3.tar.xz
mkdir build
cd build
./configure.sh

```

where configure.sh is:

```
#!/bin/sh
CROSS_PI_PATH=/opt/cross-pi-gcc-8.3.0-0 
CROSS_COMPILE=$CROSS_PI_PATH/bin/arm-linux-gnueabihf- 
PATH=$CROSS_PI_PATH/bin:$PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/lib:$LD_LIBRARY_PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/arm-linux-gnueabihf/lib:$LD_LIBRARY_PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/arm-linux-gnueabihf/libc/usr/lib:$LD_LIBRARY_PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/arm-linux-gnueabihf/libc/lib:$LD_LIBRARY_PATH 

../qt-everywhere-src-5.12.3/configure \
    -release \
    -opengl es2 \
    -device linux-rasp-pi-g++ \
    -device-option CROSS_COMPILE=${CROSS_COMPILE} \
    -sysroot ~/raspi/sysroot \
    -opensource -confirm-license \
    -make libs \
    -prefix /usr/local/qt5pi \
    -extprefix ~/raspi/qt5pi-5.12.3 \
    -hostprefix ~/raspi/qt5-5.12.3 \
    -no-use-gold-linker
```

abhiTronix, 2019-09-06:
@gfplab-scaffardi Can you confirm what is your GCC version and OS installed on your Raspberry Pi Zero by pasting the output of the following command:
```sh
cat /etc/*-release
gcc -v
g++ -v
```

abhiTronix, 2019-09-06:
@gfplab-scaffardi Good news. I managed to solve this. **Just use this modified `configure.sh` instead of your previous one:**

```sh
#!/bin/sh
CROSS_PI_PATH=/opt/cross-pi-gcc-8.3.0-0 
CROSS_COMPILE=$CROSS_PI_PATH/bin/arm-linux-gnueabihf- 

LD_LIBRARY_PATH=$CROSS_PI_PATH/lib:$LD_LIBRARY_PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/arm-linux-gnueabihf/lib:$LD_LIBRARY_PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/arm-linux-gnueabihf/libc/usr/lib:$LD_LIBRARY_PATH 
LD_LIBRARY_PATH=$CROSS_PI_PATH/arm-linux-gnueabihf/libc/lib:$LD_LIBRARY_PATH

LDFLAGS="-L$CROSS_PI_PATH/lib/ -L$CROSS_PI_PATH/arm-linux-gnueabihf/lib/ -L$CROSS_PI_PATH/arm-linux-gnueabihf/libc/usr/lib/ -L$CROSS_PI_PATH/arm-linux-gnueabihf/libc/lib/"

../qt-everywhere-src-5.12.3/configure \
    -release \
    -opengl es2 \
    -device linux-rasp-pi-g++ \
    -device-option CROSS_COMPILE=${CROSS_COMPILE} \
    -sysroot ~/raspi/sysroot \
    -opensource -confirm-license \
    -make libs \
    -prefix /usr/local/qt5pi \
    -extprefix ~/raspi/qt5pi-5.12.3 \
    -hostprefix ~/raspi/qt5-5.12.3 \
    -no-use-gold-linker \
    -no-gcc-sysroot

```

 **and and voila error is gone. I totally positive that this should work, as I test it on machine. Once confirmed, I'll tell you why this is working. Goodluck.**

abhiTronix, 2019-09-06:
@gfplab-scaffardi Closing this issue & Marking Solved, since error is no longer reproducible with above solution. Free feel to reopen this issue if suggested solution doesn't work for you. Goodluck.

gfplab-scaffardi, 2019-09-13:
This is the failed OpenGL test:

```
parallels@debian-gnu-linux-vm:~/raspi/qt-build-5.12.3-rpi0/config.tests/opengl_es2$ make
/opt/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ -mfloat-abi=hard -Wl,-O1 -Wl,-rpath-link,/home/parallels/raspi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf -o opengl_es2 main.o   /home/parallels/raspi/sysroot/opt/vc/lib/libbrcmGLESv2.so /home/parallels/raspi/sysroot/opt/vc/lib/libbrcmEGL.so   
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__read_nocancel@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__libc_fcntl64@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__open64_nocancel@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__close_nocancel@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__mprotect@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__nanosleep_nocancel@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libdl.so.2: undefined reference to `_dl_catch_error@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__write_nocancel@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `_IO_enable_locks@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__sigtimedwait@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/librt.so.1: undefined reference to `fcntl@GLIBC_2.28'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libdl.so.2: undefined reference to `_dl_signal_error@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__pause_nocancel@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__munmap@GLIBC_PRIVATE'
/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf/libpthread.so.0: undefined reference to `__mmap@GLIBC_PRIVATE'
collect2: error: ld returned 1 exit status
Makefile:67: set di istruzioni per l'obiettivo "opengl_es2" non riuscito
make: *** [opengl_es2] Errore 1
```

My sysroot is taken from last raspbian buster and it works with other toolchains like linaro-raspbian.

abhiTronix, 2019-09-13:
> My sysroot is taken from last raspbian buster

@gfplab-scaffardi See that's the problem. These errors are due to shared libraries not finding correct glibc symbols.  The problem is that my toolchains uses Glibc 2.24 as it is targeting default GCC 6.3.0 on Raspbian stretch. Whereas on your Raspbian buster with default GCC 8.3.0, uses Glibc 2.27 for compiling programs which are incompatible with Glibc 2.24 that comes with my toolchains. Sorry but this is a dead-end for now, But I can surely compile a separate toolchain binary for your Raspbian Buster after few days as due to busy schedule now.

gfplab-scaffardi, 2019-09-13:
@abhiTronix thank you for your time and help. I could imagine it's not easy to provide a toolchain for any Glibc but i think it will help for sure any RPI user having it compatible with last raspbian version. Thank you again!

abhiTronix, 2019-09-13:
> but i think it will help for sure any RPI user having it compatible with last raspbian version.

Yes, I've to create separate branches for seperate Glibc, and while still it possible to release updates for Buster right now but Not everyone has yet switched to buster including me. As it still experimental and not compatible with major libraries like OpenCV. Also, it quoted by [Raspberry Pi Dev's on its official website](https://www.raspberrypi.org/blog/buster-the-new-version-of-raspbian/) that:

> We do not recommend upgrading an existing Stretch (or earlier) system to Buster – we can’t know what changes everyone has made to their system, and so have no idea what may break when you move to Buster. However, we have tested the following procedure for upgrading, and it works on a clean version of the last Stretch image we released. That does not guarantee it will work on your system, and we cannot provide support (or be held responsible) for any problems that arise if you try it. You have been warned – make a backup!

That's why I'm not getting huge demand for buster builds, But I will soon launch separate toolchains for Raspbian Buster family. Goodluck and thanks for understanding.

abhiTronix, 2019-09-13:
@gfplab-scaffardi I'll try to compile a single binary for you targeting buster and raspberry pi zero by tomorrow so stay tuned!

abhiTronix, 2019-09-13:
@gfplab-scaffardi **I've just compiled a Raspbian Buster exclusive Toolchains with following settings:**

**GCC** - 8.3.0
**Binutils** - 2.31.1
**Glibc** - 2.28
**GDB** - 8.3
**Target OS:** Raspbian Buster OS
**Target Machine:** Raspberry Pi Zero/W/WH and Raspberry Pi 1 Model A/B/A+/B+

**Kindly download this binary from here:**
https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Experimental%20Toolchains/Raspbian%20Buster%20Toolchains/GCC%208.3.0/Raspberry%20Pi%201%2C%20Zero/
**and test your QT build with and without `-no-gcc-sysroot` flag and revert your results here.**

gfplab-scaffardi, 2019-09-13:
Here is the result WITH -no-gcc-sysroot:

Here is the result WITHOUT -no-gcc-sysroot:
```
Running configuration tests...
Checking for machine tuple... yes
Checking for valid makespec... Note: Also available for Linux: linux-clang linux-icc

ERROR: Cannot compile a minimal program. The toolchain or QMakeSpec is broken.
```

Here is the error:
```
+ cd /home/parallels/raspi/qt-build-5.12.3/config.tests/verifyspec && MAKEFLAGS= /usr/bin/make clean && MAKEFLAGS= /usr/bin/make
> rm -f verifyspec.o
> rm -f *~ core *.core
> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ -c -pipe -marm -mfpu=vfp -mtune=arm1176jzf-s -march=armv6zk -mabi=aapcs-linux -mfloat-abi=hard --sysroot=/home/parallels/raspi/sysroot -O2 -w -fPIC  -I/home/parallels/raspi/qt-everywhere-src-5.12.3/qtbase/config.tests/verifyspec -I. -I/home/parallels/raspi/qt-everywhere-src-5.12.3/qtbase/mkspecs/devices/linux-rasp-pi-g++ -o verifyspec.o /home/parallels/raspi/qt-everywhere-src-5.12.3/qtbase/config.tests/verifyspec/verifyspec.cpp
> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ -mfloat-abi=hard --sysroot=/home/parallels/raspi/sysroot -Wl,-O1 -Wl,-rpath-link,/home/parallels/raspi/sysroot/opt/vc/lib -Wl,-rpath-link,/home/parallels/raspi/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/parallels/raspi/sysroot/lib/arm-linux-gnueabihf -o verifyspec verifyspec.o      
> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: File o directory non esistente
> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: File o directory non esistente
> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
> collect2: error: ld returned 1 exit status
> Makefile:68: set di istruzioni per l'obiettivo "verifyspec" non riuscito
> make: *** [verifyspec] Errore 1

```

abhiTronix, 2019-09-14:
> Here is the result WITH -no-gcc-sysroot:
> ERROR: Feature 'webengine-alsa' was enabled, but the pre-condition 'config.unix && tests.webengine-alsa' failed.
> 
> This is the failed test:
> 
> /home/parallels/raspi/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ -c -pipe -marm -mfpu=vfp -mtune=arm1176jzf-s -march=armv6zk -mabi=aapcs-linux -mfloat-abi=hard -O2 -w -fPIC  -I/home/parallels/raspi/qt-everywhere-src-5.12.3/qtwebengine/config.tests/alsa -I. -I/home/parallels/raspi/qt-everywhere-src-5.12.3/qtbase/mkspecs/devices/linux-rasp-pi-g++ -o alsatest.o /home/parallels/raspi/qt-everywhere-src-5.12.3/qtwebengine/config.tests/alsa/alsatest.cpp
> /home/parallels/raspi/qt-everywhere-src-5.12.3/qtwebengine/config.tests/alsa/alsatest.cpp:29:10: fatal error: alsa/asoundlib.h: File o directory non esistente
>  #include <alsa/asoundlib.h>
>           ^~~~~~~~~~~~~~~~~~

@gfplab-scaffardi You're so close to ending, don't give up yet. You just need to install `“libasound2-dev”` as a pre-requisite on your Raspberry Pi to get past the error posted above. Then resync after that and this error will be gone.

abhiTronix, 2019-09-14:
> Here is the result WITH -no-gcc-sysroot:> ERROR: Feature 'webengine-alsa' was enabled, but the pre-condition 'config.unix && tests.webengine-alsa' failed.> > This is the failed test:> > [...]

gfplab-scaffardi, 2019-09-16:
I confirm i have libasound2-dev installed and copied to sysroot (/usr/lib/arm-linux-gnueabihf/libasound.* and /usr/include/alsa/*). My configure script with the other toolchain still works correctly.

abhiTronix, 2019-09-16:
> I confirm i have libasound2-dev installed and copied to sysroot (/usr/lib/arm-linux-gnueabihf/libasound.* and /usr/include/alsa/*). My configure script with the other toolchain still works correctly.

I don't have a buster environment or a suitable time schedule to test this on buster environment. But when this happens, I'll ensure that these errors related to your end only or happening on my end too. But I can asure it compiling smoothly on my stretch environment with these toolchains. Until then, I'll agree with you to keep using "other" toolchain that works. Thank you for your feedback.

gfplab-scaffardi, 2019-09-18:
Here is my buster sysroot, if you want to test it in the future: 

[https://www.dropbox.com/s/6eejnu1smvnmqcr/sysroot.tgz?dl=0](url)

Please wait 5 minutes after the post untile the file is uploaded with Dropbox.

Anyway: thank you again for your time and help!

abhiTronix, 2019-09-29:
**@gfplab-scaffardi With your sysroot I got following error on running `configure` command:**

### Error 1:

> ERROR: Feature 'opengles2' was enabled, but the pre-condition 'config.win32 || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.
> 
> ERROR: The OpenGL functionality tests failed!
> You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
> QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.

**That confirms OpenGL isn't installed properly on your Raspberry Pi.**

> Checking for ALSA... no

> looking for library alsa
> Trying source 0 (type inline) of library alsa ...
> None of [libasound.so libasound.a] found in [] and global paths.
>   => source produced no result.
> test config.qtmultimedia_multimedia.libraries.alsa FAILED

**Means, ALSA libs also not installed on your Raspberry Pi. Pardon, but You need both `libasound2` and `libasound2-dev` installed and here is a similar issue as yours: https://forum.qt.io/topic/107174/qtmultimedia-cross-compile-can-t-find-gstream/8**

### Here is a complete log with **your** sysroot:
[config.log](https://github.com/abhiTronix/raspberry-pi-cross-compilers/files/3667359/config.log)

**_Which clearly means problem is at your end only as I just successfuly compiled QT for Raspbian buster._**

closed 2020-04-13 by abhiTronix

abhiTronix, 2020-12-02:
**Resolved with [Step-by-Step Instruction for Cross-Compiling Latest QT Framework for Raspberry Pi](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md) Document.**

wuzengfeng714, 2023-06-09:
i have met one similar issue, can you give me some useful suggestion?

arm-dspg-linux-gnueabi-ld: cannot find crtbeginS.o: No such file or directory

# **the compile command is below:**
/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/clang+llvm-12.0.1-x86_64-linux-gnu-ubuntu-/bin/clang++ \
-Wno-unused-command-line-argument \
--target=arm-dspg-linux-gnueabi \
-fuse-ld=/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/x86_64-dspg-linux/usr/bin/arm-dspg-linux-gnueabi/arm-dspg-linux-gnueabi-ld \
-L/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib \
-L/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/x86_64-dspg-linux/usr/bin/arm-dspg-linux-gnueabi/../../lib/arm-dspg-linux-gnueabi/gcc/arm-dspg-linux-gnueabi/10.2.0/ \
-L/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/x86_64-dspg-linux/usr/bin/arm-dspg-linux-gnueabi/../../lib/arm-dspg-linux-gnueabi/gcc/ \
-L/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/cortexa9-neon-dspg-linux-gnueabi/lib/ \
-L/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/cortexa9-neon-dspg-linux-gnueabi/usr/lib/arm-dspg-linux-gnueabi/10.2.0/ \
-L/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/cortexa9-neon-dspg-linux-gnueabi/usr/lib/ \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libabsl.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libssl.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libcrypto.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libbase.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libz.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libtinyxml2.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libflatbuffers.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libdbus-1.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libprotobuf-lite.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libmodp_b64.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libdouble-conversion.a \
/home/fluoride12/opensource/temp/bluetooth2_test/required/sysroot/lib/libevent.a \
-ldl \
-lpthread \
-lrt \
-lresolv \
-latomic \
--sysroot=/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/cortexa9-neon-dspg-linux-gnueabi \
--gcc-toolchain=/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/x86_64-dspg-linux \
-Wl,-z,relro \
-Wl,-z,noexecstack \
-Wl,-z,now \
-Wl,--as-needed \
-pie \
-Lrust \
-o \
bluetoothtbd_socketclient \
-Wl,--start-group \
obj/bt/system/service/client/bluetoothtbd_socketclient.main_socket.o \
obj/bt/system/service/client/bluetoothtbd_socketclient.main_menu.o \
-Wl,--end-group \
 \
-ldl \
-lpthread \
-lrt \
-lssl \
-lcrypto \
-lbase

**Then the failure arrived.**
/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/x86_64-dspg-linux/tools/sysroots/x86_64-dspg-linux/usr/bin/arm-dspg-linux-gnueabi/arm-dspg-linux-gnueabi-ld: cannot find crtbeginS.o: No such file or directory

I have already checked the below:
a,
fluoride12@wcd-sh-bt-2:~/opensource/temp/bluetooth2_test/required/compiler$ find /home/fluoride12/opensource/temp/bluetooth2_test/ -name crtbeginS.o
/home/fluoride12/opensource/temp/bluetooth2_test/required/compiler/gcc-arm-10.2-2020.11-x86_64-arm-none-linux-gnueabihf/lib/gcc/arm-none-linux-gnueabihf/10.2.1/crtbeginS.o
b,
my compile command has already specify the --sysroot, and -L, both of them include crtbeginS.o

If somebody can give me some suggestion, i will be very glad.
Thanks!!!