abhiTronix/raspberry-pi-cross-compilers #30: Raspbian Buster

takacsd, 2019-12-30:
Hi,

I was fighting for days to cross compile Qt for my new Pi Zero with your toolchais, and I almost gave up and reverted back to the gcc 4.6 what every tutorial suggests, when finally I stumble upon #27 (which starts with a problem I already resolved) and more specifically this comment: https://github.com/abhiTronix/raspberry-pi-cross-compilers/issues/27#issuecomment-531153222

First of all, thanks for the toolchain here (even though the link is not right): https://github.com/abhiTronix/raspberry-pi-cross-compilers/issues/27#issuecomment-531303919
It worked and solved my issues instantly.

But please, even if you don't want to make this toolchain "official", make it more clear that the current toolchains only support Stretch and not Buster, because that's not mentioned anywhere in the wiki or Readme.md.

Other than that, keep up the good work!

abhiTronix, 2019-12-31:
@takacsd Thanks for addressing this issue.

> I was fighting for days to cross compile Qt for my new Pi Zero with your toolchais, and I almost gave up and reverted back to the gcc 4.6 what every tutorial suggests, when finally I stumble upon #27

Thanks for confirming it. Glad it worked for you.

> when finally I stumble upon #27 (which starts with a problem I already resolved) and more specifically this comment: #27 (comment) First of all, thanks for the toolchain here (even though the link is not right): #27 (comment)
It worked and solved my issues instantly.

**Yup, I recently compiled separate exclusive binaries for Raspbian Buster OS which can be found here: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Exclusive%20%20Toolchains/Raspbian%20Buster%20Cross-Compiler%20Toolchains/** 

> But please, even if you don't want to make this toolchain "official", make it more clear that the current toolchains only support Stretch and not Buster

**This is a Unofficial non-profit Repository _(not associated with GCC or raspberry pi community)_ , which was earlier created for my research projects but I made it open-sourced for public benefits. The repository is divided into following parts:**

| Binaries | Supported Os | Status |
|----------|-----------------|--------|
| Cross-Compiler Binaries | Stretch OS only | Stable |
| Native Compiler Binaries | Stretch OS only | Stable |
| 64-bit Compiler Binaries | 64-bit OS only(like Pi64) | Stable |
| Exclusive Toolchains Binaries | Buster OS and other beta releases | Beta/Testing/Experimental |  

**Hence, Each binary has built targeting specific OS.** 

**But you may ask why it is not documented?** 
_**The answer is the Exclusive Toolchain binaries are still in beta-testing state and hasn't been tested well by the community. Also, Buster hasn't been replaced by many developers in their Raspberry Pi Hardware as it is still buggy.** So I compiled these buster toolchains just for testing, once it is matured enough, It will be moved to Cross/Native Compiler section and will be documented._

### Solution: 
**I'm tagging this issue at the top and once it get accepted by community and enough requests are made I'll add it to primary section.**

takacsd, 2020-01-01:
> **This is a Unofficial non-profit Repository _(not associated with GCC or raspberry pi community)_ , which was earlier created for my research projects but I made it open-sourced for public interest. The repository is divided into following parts:**
> 
> |Binaries|	Target OS|	Status|
> |-|-|-|
> |Cross-Compiler Binaries|	Stretch OS only|	Stable/Production|
> |Native Compiler Binaries|	Stretch OS only|	Stable/Production|
> |64-bit Compiler Binaries|	64-bit OS only(like Pi64)|	Stable/Production|
> |Exclusive Toolchains Binaries|	Buster OS and other beta releases|	Beta/Testing/Experimental|
> |Obsolete Compiler Binaries|	Not supported|	No longer supported|
>
> **Hence, Each binary has built targeting specific OS only.**

It is totally clear, I'm also just doing a hobby project. But if you would put this same table somewhere in the wiki, that would already help a lot.

> **But you may ask why it is not documented?**
> 
> _**The answer is the Exclusive Toolchain binaries are still in beta-testing state and hasn't been tested well by the community.

Hard to give testing feedback if we don't know it's existence.  :wink: 
If I were you, I would list the experimental toolchain right in the middle of the **features** on the main page.

Just my two cents. Anyway thanks for the work you put in this already.

abhiTronix, 2020-01-01:
> Hard to give testing feedback if we don't know it's existence.

@takacsd Very convincing argument. I'm adding this in docs right now. Thanks

abhiTronix, 2020-01-02:
**Successfully resolved in commit 5663480**

abhiTronix, 2020-02-01:
Realeasing support for Raspbian Buster(Debian 10) ARM/ARM64 Toolchains #35

closed 2020-02-05 by abhiTronix