abhiTronix/raspberry-pi-cross-compilers #34: Cross-Compiler and Buildroot

rcrespir, 2020-01-28:
Hi, 

I am trying to use your cross-compiler along with buildroot. My host is an Ubuntu 16.04 and the target is Raspberry PI 0 W.

I am using the cross-pi-gcc-8.3.0-0 and following next instructions: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler:-Installation-Instructions

```
ibercomp@ibercomp-MS-7994:~$ arm-linux-gnueabihf-gcc --version
arm-linux-gnueabihf-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

First I have tried with the temporary installation and after with permanent, but any one seems to work when try to build an image with buildroot.

```
ibercomp@ibercomp-MS-7994:~/rcr/projects/cp4/buildroot$ make
/usr/bin/make -j1 O=/home/ibercomp/rcr/projects/cp4/buildroot/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig

You seem to have the current working directory in your
LD_LIBRARY_PATH environment variable. This doesn't work.
support/dependencies/dependencies.mk:27: recipe for target 'dependencies' failed
make[1]: *** [dependencies] Error 1
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```

Any idea of what I am missing?

Kind regards, 

RAfa

abhiTronix, 2020-01-28:
@rcrespir Seems unwanted values in your `LD_LIBRARY_PATH`, Try unsetting it with: 
```
unset LD_LIBRARY_PATH
```
and try again.

rcrespir, 2020-01-28:
@abhiTronix 

Ok, I have unset LD_LIBRARY_PATH, and building fails with:

```
/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libdl.so.2, needed by ../libxtables/.libs/libxtables.so, not found (try using -rpath or -rpath-link)
../libxtables/.libs/libxtables.so: undefined reference to `dlerror@GLIBC_2.4'
../libxtables/.libs/libxtables.so: undefined reference to `dlopen@GLIBC_2.4'
collect2: error: ld returned 1 exit status
Makefile:636: recipe for target 'xtables-legacy-multi' failed
make[5]: *** [xtables-legacy-multi] Error 1
Makefile:536: recipe for target 'all' failed
make[4]: *** [all] Error 2
Makefile:507: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
Makefile:414: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:238: recipe for target '/home/ibercomp/rcr/projects/cp4/buildroot/output/build/iptables-1.8.2/.stamp_built' failed
make[1]: *** [/home/ibercomp/rcr/projects/cp4/buildroot/output/build/iptables-1.8.2/.stamp_built] Error 2
Makefile:84: recipe for target '_all' failed
make: *** [_all] Error 2
```
Maybe I am not defining correctly the paths?
 When installing the cross-compiler, I am running the "Permanent installation "commands from the ~ directory. 
`ibercomp@ibercomp-MS-7994:~$ echo 'export PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/bin:$PATH' >> .bashrc`

And then, I have to change to `~rcr/projects/cp4/buildroot` to run the `make `command from buildroot.

Where I have to run the `unset LD_LIBRARY_PATH`?  From ~ directory or from buildroot directory?

Kind regards

abhiTronix, 2020-01-28:
@rcrespir That's why it is not recommended to mix the Permanent installation method and Temporary method since you're unsetting `LD_LIBRARY_PATH` but you're not removing it from `.bashrc` and unsetting `LD_LIBRARY_PATH` also removes other important linker paths. **Try the following steps to reverse it:**

- re-edit `.bashrc` and remove `export PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/bin` and `echo 'export LD_LIBRARY_PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/lib:$LD_LIBRARY_PATH'` from it.

- If you've change `.profile` too, then re-edit `.profile` and remove `export PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/bin` and `echo 'export LD_LIBRARY_PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/lib:$LD_LIBRARY_PATH'` from it.

* restart your machine

* Run these commands
  ```sh
  PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/bin:$PATH
  LD_LIBRARY_PATH=/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0/lib:$LD_LIBRARY_PATH
  LIBS=-ldl
  ``` 

* That's all, Now Re-Run your build script and paste here if any error occurs.

rcrespir, 2020-01-29:
.Hi, 
.
I have followed the steps above and I get the same failure.

Maybe there is an incompatibility between the toolchain and buildroot (buildroot 2019.02.09)? I don't know if you are used to buildroot, but I am new on it. Please find attached my buildroot defconfig, maybe I have to add something to the toolchain config.

This config worked perfectly with the raspberry/tools toolchain [https://github.com/raspberrypi/tools/blob/master/arm-bcm2708/arm-linux-gnueabihf](url)

```
BR2_arm=y
BR2_arm1176jzf_s=y
BR2_TOOLCHAIN_EXTERNAL=y
BR2_TOOLCHAIN_EXTERNAL_PATH="/home/ibercomp/rcr/tools/toolchains/sourceforge/rpi0w/cross-pi-gcc-8.3.0-0"
BR2_TOOLCHAIN_EXTERNAL_CUSTOM_PREFIX="arm-linux-gnueabihf"
BR2_TOOLCHAIN_EXTERNAL_GCC_8=y
BR2_TOOLCHAIN_EXTERNAL_HEADERS_4_14=y
BR2_TOOLCHAIN_EXTERNAL_CUSTOM_GLIBC=y
# BR2_TOOLCHAIN_EXTERNAL_INET_RPC is not set
BR2_TOOLCHAIN_EXTERNAL_CXX=y
BR2_TARGET_GENERIC_HOSTNAME="ibLinux"
BR2_TARGET_GENERIC_ISSUE="Welcome to ibLinux"
BR2_TARGET_GENERIC_ROOT_PASSWD="0000"
# BR2_TARGET_GENERIC_GETTY is not set
BR2_TARGET_TZ_INFO=y
BR2_ROOTFS_OVERLAY="$(BR2_EXTERNAL_rscr_br2_extree_PATH)/board/ibercomp/rpi0w/rootfs_overlay"
BR2_ROOTFS_POST_BUILD_SCRIPT="$(BR2_EXTERNAL_rscr_br2_extree_PATH)/board/ibercomp/rpi0w/post-build.sh"
BR2_ROOTFS_POST_IMAGE_SCRIPT="support/scripts/genimage.sh"
BR2_ROOTFS_POST_SCRIPT_ARGS="-c $(BR2_EXTERNAL_rscr_br2_extree_PATH)/board/ibercomp/rpi0w/genimage_rpi0w.cfg"
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_GIT=y
BR2_LINUX_KERNEL_CUSTOM_REPO_URL="https://github.com/raspberrypi/linux.git"
BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="rpi-4.19.y"
BR2_LINUX_KERNEL_DEFCONFIG="bcmrpi"
BR2_LINUX_KERNEL_DTS_SUPPORT=y
BR2_LINUX_KERNEL_INTREE_DTS_NAME="bcm2708-rpi-zero-w"
BR2_LINUX_KERNEL_DTB_OVERLAY_SUPPORT=y
BR2_LINUX_KERNEL_NEEDS_HOST_OPENSSL=y
BR2_PACKAGE_BUSYBOX_SHOW_OTHERS=y
BR2_PACKAGE_GREP=y
BR2_PACKAGE_MAKE=y
BR2_PACKAGE_DEJAVU=y
BR2_PACKAGE_QT5=y
BR2_PACKAGE_QT5BASE_EXAMPLES=y
BR2_PACKAGE_QT5BASE_OPENGL_LIB=y
BR2_PACKAGE_QT5BASE_LINUXFB=y
BR2_PACKAGE_QT5BASE_EGLFS=y
BR2_PACKAGE_QT5BASE_DEFAULT_QPA="linuxfb"
BR2_PACKAGE_QT5BASE_FONTCONFIG=y
BR2_PACKAGE_QT5BASE_JPEG=y
BR2_PACKAGE_QT5BASE_PNG=y
BR2_PACKAGE_QT5BASE_ICU=y
BR2_PACKAGE_QT5BASE_TSLIB=y
BR2_PACKAGE_QT5GRAPHICALEFFECTS=y
BR2_PACKAGE_QT5QUICKCONTROLS=y
BR2_PACKAGE_QT5QUICKCONTROLS2=y
BR2_PACKAGE_QT5VIRTUALKEYBOARD=y
BR2_PACKAGE_QT5VIRTUALKEYBOARD_LANGUAGE_LAYOUTS="all"
BR2_PACKAGE_QT5VIRTUALKEYBOARD_HANDWRITING=y
BR2_PACKAGE_QT5WEBSOCKETS=y
BR2_PACKAGE_RPI_FIRMWARE=y
BR2_PACKAGE_RPI_WIFI_FIRMWARE=y
BR2_PACKAGE_I2C_TOOLS=y
BR2_PACKAGE_RPI_USERLAND=y
BR2_PACKAGE_LIBARCHIVE=y
BR2_PACKAGE_LEVELDB=y
BR2_PACKAGE_SQLITE=y
BR2_PACKAGE_BCM2835=y
BR2_PACKAGE_LIBCURL=y
BR2_PACKAGE_CURL=y
BR2_PACKAGE_PCRE=y
BR2_PACKAGE_PCRE_16=y
BR2_PACKAGE_PCRE_UTF=y
BR2_PACKAGE_CRDA=y
BR2_PACKAGE_DHCPCD=y
BR2_PACKAGE_IPROUTE2=y
BR2_PACKAGE_IPTABLES=y
BR2_PACKAGE_IW=y
BR2_PACKAGE_OPENSSH=y
BR2_PACKAGE_WIRELESS_TOOLS=y
BR2_PACKAGE_WIRELESS_TOOLS_LIB=y
BR2_PACKAGE_WPA_SUPPLICANT=y
BR2_PACKAGE_WPA_SUPPLICANT_EAP=y
BR2_PACKAGE_WPA_SUPPLICANT_WPS=y
BR2_PACKAGE_WPA_SUPPLICANT_CLI=y
BR2_PACKAGE_WPA_SUPPLICANT_WPA_CLIENT_SO=y
BR2_PACKAGE_WPA_SUPPLICANT_PASSPHRASE=y
BR2_PACKAGE_BASH=y
BR2_PACKAGE_BASH_COMPLETION=y
BR2_PACKAGE_SUDO=y
BR2_PACKAGE_HTOP=y
BR2_PACKAGE_KMOD=y
BR2_PACKAGE_UTIL_LINUX_LIBBLKID=y
BR2_PACKAGE_NANO=y
BR2_TARGET_ROOTFS_EXT2=y
BR2_TARGET_ROOTFS_EXT2_4=y
BR2_TARGET_ROOTFS_EXT2_SIZE="1G"
# BR2_TARGET_ROOTFS_TAR is not set
BR2_PACKAGE_HOST_DOSFSTOOLS=y
BR2_PACKAGE_HOST_GENIMAGE=y
BR2_PACKAGE_HOST_MTOOLS=y
BR2_PACKAGE_HOST_UBOOT_TOOLS=y
BR2_PACKAGE_HOST_UBOOT_TOOLS_FIT_SUPPORT=y
```

Any idea?

Kind Regards, 

Rafa

abhiTronix, 2020-01-29:
@rcrespir  What OS are you using on your Raspberry Pi 0 W? and any reason for using this `BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="rpi-4.19.y"` Kernel? Also, are you following any online tutorial to compile it?

rcrespir, 2020-01-29:
> What OS are you using on your Raspberry Pi 0 W? and any reason for using this BR2_LINUX_KERNEL_CUSTOM_REPO_VERSION="rpi-4.19.y" Kernel?

I am building my own image with Buildroot, based on the kernel https://github.com/raspberrypi/linux/tree/rpi-4.19.y. I am using  `rpi-4.19.y`  because it's the one by default on the repository.

Kind regards, 

Rafa

> Also, are you following any online tutorial to compile it?

Basically I have followed this two:
https://microchipdeveloper.com/32mpu:buildroot-custom-project
https://bootlin.com/doc/training/buildroot/buildroot-labs.pdf

abhiTronix, 2020-01-29:
@rcrespir There can be many reasons for why this is not working: Maybe some important libraries are missing from your machine., Maybe you're using the wrong configuration, or maybe this toolchain is incompatible. The tutorial links you provided are not meant exclusively for Raspberry Pi, therefore trying to cross-compile something from scratch will not work and not an ideal thing to do.

**Solution**: You've two choices:
1. Use the right online tutorial that is exclusively meant for Raspberry Pi. I found some by googling:
* https://ltekieli.com/buildroot-with-raspberry-pi-what-where-and-how/
* https://jumpnowtek.com/rpi/Raspberry-Pi-Systems-with-Buildroot.html
* https://blog.crysys.hu/2018/06/using-buildroot-to-create-custom-linux-system-images/

2. Use old raspberry/tools toolchain i.e. https://github.com/raspberrypi/tools/blob/master/arm-bcm2708/arm-linux-gnueabihf

Good luck, and if you managed to compile it with this project toolchain with this toolchain,  then do share here.

closed 2020-01-29 by abhiTronix

3ibusiness, 2020-11-20:
I had the same problem and was able to solve it with super user mode just type **sudo make**

nasser-miah, 2021-11-14:
> I had the same problem and was able to solve it with super user mode just type **sudo make**

Try:

unset LD_LIBRARY_PATH; make

Which worked without "sudo" for me.