abhiTronix/raspberry-pi-cross-compilers #44: I modify GLIBC_VERSION as 2.23, build native gcc failed.

jackyxinli, 2020-06-26:
Cause my development board supports glibc version is 2.23, so I modify GLIBC_VERSION in RTBuilder_32b as 2.23, build cross gcc succeed, but build native gcc failed. How can I do?

abhiTronix, 2020-06-26:
> Cause my development board supports glibc version is 2.23

@jackyxinli What board are you using?

> so I modify GLIBC_VERSION in RTBuilder_32b as 2.23

That's not enough, you need to compile supported KERNEL according to your board, and change ARCH, FPU, BINUTILS_VERSION too.

> build cross gcc succeed, but build native gcc failed.

Share the log, onl then I'll be able to help you.

abhiTronix, 2020-06-27:
**Closed. Kindly provide proper & detailed information to elaborate your issue.**

closed 2020-06-27 by abhiTronix