abhiTronix/raspberry-pi-cross-compilers #47: ASAN support is broken

BakaOsaka, 2020-07-01:
Trying to use the built compiler to enable ASAN support results in a number of errors.

```
user@machine:~/cross-pi-gcc-8.3.0-0$ ./bin/arm-linux-gnueabihf-gcc -fsanitize=address /home/user/projects/git/test.c -o/tmp/test
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libdl.so.2, needed by /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so, not found (try using -rpath or -rpath-link)
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libpthread.so.0, needed by /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so, not found (try using -rpath or -rpath-link)
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libstdc++.so.6, needed by /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so, not found (try using -rpath or -rpath-link)
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libm.so.6, needed by /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so, not found (try using -rpath or -rpath-link)
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `dlvsym@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `pthread_setspecific@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `shm_unlink'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `dlsym@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `pthread_attr_setstacksize@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `pthread_getattr_np@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `signgam@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `__signgam@GLIBC_2.23'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `pthread_getspecific@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `pthread_key_create@GLIBC_2.4'
/home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: /home/user/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/lib/libasan.so: undefined reference to `shm_open'
```

Using the standard version of GCC shipped with Debian works without issue:
```
user@machine:~/cross-pi-gcc-8.3.0-0$ arm-linux-gnueabi-gcc -fsanitize=address /home/user/projects/git/test.c -o/tmp/test
user@machine:~/cross-pi-gcc-8.3.0-0$ 
```

An strace log shows that it is not searching the standard compiler search paths for the required libraries:
```
bgamble@gemeinschaft:~/cross-pi-gcc-8.3.0-0$ strace -ff ./bin/arm-linux-gnueabihf-gcc -fsanitize=address /home/bgamble/projects/git/test.c -o/tmp/test 2>&1  | grep 'libdl\|libpthread\|libstdc++\|libm'
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
[pid 24819] openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
[pid 24819] openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
[pid 25137] openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
[pid 25205] openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
[pid 25258] openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libdl.so.2", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libdl.so.2", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] write(2, "libdl.so.2", 10)  = 10
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libpthread.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libpthread.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] write(2, "libpthread.so.0", 15) = 15
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libstdc++.so.6", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libstdc++.so.6", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] write(2, "libstdc++.so.6", 14) = 14
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libm.so.6", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] openat(AT_FDCWD, "/arm-linux-gnueabihf/lib/libm.so.6", O_RDONLY) = -1 ENOENT (No such file or directory)
[pid 25258] write(2, "libm.so.6", 9)    = 9
[pid 25205] write(2, "libdl.so.2", 10libdl.so.2)  = 10
[pid 25205] write(2, "libpthread.so.0", 15libpthread.so.0) = 15
[pid 25205] write(2, "libm.so.6", 9libm.so.6)    = 9
```

abhiTronix, 2020-07-01:
@BakaOsaka Did you set the environment variables as follows?

```sh
PATH=/<extracted folder-name e.g cross-pi-gcc-9.2.0-1>/bin:$PATH
LD_LIBRARY_PATH=/<extracted folder-name e.g cross-pi-gcc-9.2.0-1>/lib:$LD_LIBRARY_PATH
```

BakaOsaka, 2020-07-01:
The issue still remains with the above variables. I used the following command line:
```
LD_LIBRARY_PATH=/home/user/cross-pi-gcc-8.3.0-0/lib:$LD_LIBRARY_PATH PATH=/home/user/cross-pi-gcc-8.3.0-0/bin/:$PATH arm-linux-gnueabihf-gcc -fsanitize=address /home/user/projects/git/test.c -o /tmp/test
```

This gives the same error about being unable to find the required libraries.

abhiTronix, 2020-07-01:
@BakaOsaka Ok, I'll look into it tomorrow. In the mean time, Can you try other GCC versions and cross-compiler binaries too, available in this project? Thanks.

BakaOsaka, 2020-07-01:
Thanks for looking in to this. I've tried with GCC 7.3.0, GCC 8.3.0 and GCC 10.1.0 as well as all variants of Pi and both variants of Raspbian. All builds show the same issue, so I'm assuming something is wrong as part of the build process rather than specific to a version of GCC.

For reference, a compiler built with crosstool-ng does not have this issue - Building with the "-fsanitize=..." flag works without any errors.

abhiTronix, 2020-07-28:
@BakaOsaka Thanks for your patience, I've run some tests and found that -fsanitize=address alone does not works. 

### Solution:

For GCC(works with clang too), build with the following options to enable address/leak sanitizer.

- For `-march=armv6` architecture CPUs compilers:
```sh
arm-linux-gnueabi-gcc -ggdb -fsanitize=address -fno-omit-frame-pointer -static-libstdc++ -static-libasan -lrt
// or
arm-linux-gnueabi-gcc  -ggdb -fsanitize=leak -fno-omit-frame-pointer -static-libstdc++ -static-liblsan -lrt
```
- For any other variant:
```sh
arm-linux-gnueabihf-gcc -ggdb -fsanitize=address -fno-omit-frame-pointer -static-libstdc++ -static-libasan -lrt
// or
arm-linux-gnueabihf-gcc  -ggdb -fsanitize=leak -fno-omit-frame-pointer -static-libstdc++ -static-liblsan -lrt
```

### Options Explanation:

-   `-ggdb and -fno-omit-frame-pointer` - enable the call stack and line number for better report format to locate bug in user code
-    `-fsanitize=${type}` - sanitizer type:
        address, address level detection, heap and stack
        leak, heap leak and heap overflow detection, no stack
-    `-lrt`- needed by linux shared memory operation: shm_open and shm_unlink
-    `-static-libasan/-static-liblsan` - choose corresponding to option -fsanitize, asan for address, lsan for leak
-    Finally `-static-libstdc++` and `-static-libgcc`  as the options’ literal meaning

**Note:** Do not use any memory manager, say tcmalloc/jemalloc, when you try to use the sanitizer of gcc

closed 2020-07-28 by abhiTronix