abhiTronix/raspberry-pi-cross-compilers #49: Version for Raspberry Pi Desktop (for PC and Mac)

jimcruse, 2020-08-07:
I want to get the exact same compiler for raspberry pi PC version
Here is the version info from the PC:
  Distributor ID: Debian
  Description:    Debian GNU/Linux 10 (buster)
  Release:        10
  Codename:       buster

abhiTronix, 2020-08-07:
@jimcruse What's the kernel running on raspberry pi PC version? Kindly paste the value of `uname -a` command from its terminal.

jimcruse, 2020-08-08:
Linux raspberry 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64 GNU/Linux

abhiTronix, 2020-08-08:
> Linux raspberry 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64 GNU/Linux

@jimcruse First of all, it is not Raspbian/ Raspberry Pi OS. Its just Debian Buster with a modified LXDE Desktop, it is similar to Lubuntu with a x86-64 architecture, similar to what you find in PCs, laptops etc. So think of it as just a Ubuntu running on your PC in a emulator. So if you like to install latest GCC like GCC-10, just do:
```sh
sudo apt-get update && sudo apt-get install software-properties-common
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt install gcc-10
```

jimcruse, 2020-08-08:
Your right it is a PC running the raspberry Ubuntu customization, I wanted to run the same gcc version , so can verify on Ubuntu before testing on pi4, I tried to run as above but I got invalid no release

abhiTronix, 2020-08-08:
> I tried to run as above but I got invalid no release

@jimcruse Yeah, Debian got gcc-10 update for its unstable distribution [`sid` only](https://packages.debian.org/sid/gcc-10), and you've to wait for GCC 10 until next distribution.

abhiTronix, 2020-08-08:
It won't be easy task to compile it for x86-64 kernels but I'll give it a try anytime soon.

abhiTronix, 2020-08-08:
> I wanted to run the same gcc version , so can verify on Ubuntu before testing on pi4, I tried to run as above but I got invalid no release

It won't help, since  x86-64 kernel is completely different from arm kernel, infact ARM is a RISC (Reduced Instruction Set Computing) architecture while x86 is a CISC (Complex Instruction Set Computing) architecture. So your GCC executables may run fine on x86-64 kernels but it might be faulty on arm devices, or vice-versa.

jimcruse, 2020-08-10:
I understand the actual compiled file won't work between I'm more thinking compilation issues, i.e. program debugging...

closed 2020-09-12 by abhiTronix

abhiTronix, 2020-09-22:
@jimcruse **Successfully resolved and merged in commit 6883115**

commit aca67c39 (2020-08-31): Initial support for Raspberry Pi Desktop toolchains (Fixes #49)

- Added `x86_64TB` CI build script.
- Added support x86 & x86_64 Raspberry Pi Desktop OS types.
- Added Github Action workflow for these toolchains.
- Updated SF_deployer and SF_docs_deployer to include these new changes.
- Updated README.md and added new docs.
- Updated `curl` to latest v7.72.0