abhiTronix/raspberry-pi-cross-compilers #59: QT_build_instructions.md does not work for me

gylead, 2020-12-20:
I tried to follow your recipe literally with one exception: I didn't use "sudo" for all the commands on the host. I use buster on Raspberry Pi 3 B+ and your toolchain cross-gcc-8.3.0-pi_3+. After rsync the file structure looks very good.

When I start configure qmake is successfully build. But the very first build of a tool with the cross compiler toolchain fails because crt1.o is not found:

I tried with different configure command lines (e.g. replacing the * in CROSS_COMPILER_LOCATION with the real folder name and replace the $(echo ...) command by $CROSS_COMPILER_LOCATION. It doesn't help.

There are two files "crt1.o" in the file tree:

./sysroot/usr/lib/arm-linux-gnueabihf/crt1.o
./tools/cross-pi-gcc-8.3.0-2/arm-linux-gnueabihf/libc/usr/lib/crt1.o

Here is the output from configure:

Running configuration tests...
Checking for machine tuple... 
+ /home/jochen/rpi-qt/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-g++ -dumpmachine
> arm-linux-gnueabihf
test config.qtbase.tests.machineTuple succeeded
Checking for valid makespec... 
+ cd /home/jochen/rpi-qt/build/config.tests/verifyspec && /home/jochen/rpi-qt/build/qtbase/bin/qmake "CONFIG -= qt debug_and_release app_bundle lib_bundle" "CONFIG += shared warn_off console single_arch" 'QMAKE_LIBDIR += /home/jochen/rpi-qt/sysroot/usr/lib/arm-linux-gnueabihf' 'INCLUDEPATH += /home/jochen/rpi-qt/sysroot/usr/include/arm-linux-gnueabihf' "QMAKE_CFLAGS += --sysroot=/home/jochen/rpi-qt/sysroot/" "QMAKE_CXXFLAGS += --sysroot=/home/jochen/rpi-qt/sysroot/" "QMAKE_LFLAGS += --sysroot=/home/jochen/rpi-qt/sysroot/" -early "CONFIG += cross_compile" /home/jochen/rpi-qt/qt-everywhere-src-5.15.2/qtbase/config.tests/verifyspec
+ cd /home/jochen/rpi-qt/build/config.tests/verifyspec && MAKEFLAGS= /usr/bin/make
> /home/jochen/rpi-qt/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-g++ -c -march=armv8-a -mtune=cortex-a72 -mfpu=crypto-neon-fp-armv8 -mfloat-abi=hard --sysroot=/home/jochen/rpi-qt/sysroot/ -O2 -w -fPIC  -I/home/jochen/rpi-qt/qt-everywhere-src-5.15.2/qtbase/config.tests/verifyspec -I. -I/home/jochen/rpi-qt/sysroot/usr/include/arm-linux-gnueabihf -I/home/jochen/rpi-qt/qt-everywhere-src-5.15.2/qtbase/mkspecs/devices/linux-rasp-pi4-v3d-g++ -o verifyspec.o /home/jochen/rpi-qt/qt-everywhere-src-5.15.2/qtbase/config.tests/verifyspec/verifyspec.cpp
> /home/jochen/rpi-qt/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-g++ -mfloat-abi=hard --sysroot=/home/jochen/rpi-qt/sysroot/ -Wl,-O1 -Wl,-rpath-link,/home/jochen/rpi-qt/sysroot/usr/lib/arm-linux-gnueabihf -Wl,-rpath-link,/home/jochen/rpi-qt/sysroot/lib/arm-linux-gnueabihf -o verifyspec verifyspec.o   -L/home/jochen/rpi-qt/sysroot/usr/lib/arm-linux-gnueabihf   
> /home/jochen/rpi-qt/tools/cross-pi-gcc-8.3.0-2/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: crt1.o kann nicht gefunden werden: Datei oder Verzeichnis nicht gefunden
> /home/jochen/rpi-qt/tools/cross-pi-gcc-8.3.0-2/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: crti.o kann nicht gefunden werden: Datei oder Verzeichnis nicht gefunden
> collect2: Fehler: ld gab 1 als Ende-Status zurück
> make: *** [Makefile:68: verifyspec] Fehler 1

gylead, 2020-12-20:
Found closed issue #27 which seems to attack the same issue. Used the same script as suggested in that issue (replacing the configure options for my needs). configure is now running without fatal error.

Unfortunately the config.summary has not even eglfs enabled. From config.log I see that even the tests are not compiled, there seems to be a major issue with the system configuration.

Here is what I get:

```
ERROR: Feature 'opengles2' was enabled, but the pre-condition '(config.win32 && !features.opengl-dynamic) || (!config.watchos && !features.opengl-desktop && libs.opengl_es2)' failed.

ERROR: Feature 'eglfs' was enabled, but the pre-condition '!config.android && !config.darwin && !config.win32 && !config.wasm && features.egl' failed.

ERROR: The OpenGL functionality tests failed!
You might need to modify the include and library search paths by editing QMAKE_INCDIR_OPENGL[_ES2],
QMAKE_LIBDIR_OPENGL[_ES2] and QMAKE_LIBS_OPENGL[_ES2] in the mkspec for your platform.
```

abhiTronix, 2020-12-21:
> I tried to follow your recipe literally with one exception: I didn't use "sudo" for all the commands on the host.

@gylead **Why? If you're not willing to follow carefully crafted instructions, then you must not use it. Follow them properly, an if then it refused to work, then let me know.**

closed 2020-12-21 by abhiTronix

fuzun, 2021-01-23:
@abhiTronix This issue is not invalid. I followed your guide without any changes but I get the same error.

@gylead Using -no-gcc-sysroot as a workaround solves the problem. But it is not a real solution.

abhiTronix, 2021-01-24:
> @abhiTronix This issue is not invalid. I followed your guide without any changes but I get the same error.

@fuzun First of all, have sense about how to interact with other developers. Secondly this is not a argument "I get the same error.", I need the logs and terminal output of exact error to debug anything as I can't telepathically reach your system to find out myself.

> @gylead Using -no-gcc-sysroot as a workaround solves the problem. But it is not a real solution.

Also, who  said that `-no-gcc-sysroot` is a solution and why and where you need this? There's no  single command in the guide that tells you to do so. Quit with this nonsense.

fuzun, 2021-01-24:
I have no idea what you are talking about how I interact. I noticed a problem and posted it here. It is obvious that you got triggered for some reason.

I get the same error is clearly an argument given that I followed the same steps you provided and the log is the same if else I would not have said "same error"

In the issue he mentioned, you brought up "-no-gcc-sysroot". And the new guide does not work at all (crt1.o, crti.o not found) for me and some other people.

abhiTronix, 2021-01-24:
> I have no idea what you are talking about how I interact. I noticed a problem and posted it here. It is obvious that you got triggered for some reason.

@fuzun You cannot just chim in wherever and say I have this problem and fix it. You bring facts and proof to the table then ask for the help. Qt is highly susceptible to dirty build environments, and it is surely a missing step or configuration that is why it is not working but I cannot say without logs from your system.

>  else I would not have said "same error"

"Same error" doesn't mean your system image is identical. Every system is different and QT is system configuration dependent. Stop saying nonsense thing to support your bullshit arguments. 

> In the issue he mentioned, you brought up "-no-gcc-sysroot"

That is because sysroot was previously hardcoded in these binaries earlier, and was solved recently. Noone asked you to follow old comments in favour of new guide, that's on you.

> some other people.

That's where you draw the line for yourself. You cannot say it don't worked because whole toolchain is broken. This repository have over 10k+ downloads in peak months from all around the world and everything is open-source here, so quit with this already.

fuzun, 2021-01-24:
> You cannot just chim in wherever and say I have this problem and fix it.

I stated a problem, it is your responsibility if you want to confirm that the problem is there and to fix it. You could have ignored it that'd be your choice or further asked questions but you should not talk like a bigot knowing everything because that just makes you funny.

> You bring facts and proof to the table then ask for the help. 

I did not ask for help. I looked it up on the internet and concluded that its your toolchain that is configured improperly. Given the lack of knowledge of yours I doubt anyone would ask any help from you.
And of course one has to bring facts and proofs. This is a developer portal there is not place for chit-chat like this. Time is important so I always try to be clear and simple unless I have to answer some crap.

> That is because sysroot was previously hardcoded in these binaries earlier, and was solved recently. Noone asked you to follow old comments in favour of new guide, that's on you.

Owner of this issue said that he fixed the problem by using the methods you provided in `that` issue. Considering he had to use no sys root parameter that you provided why did not you tell him that the recent builds don't need that but instead you provoked him with your bullshit. And of course no one asked me to follow that but it is not working without using no-gcc-sysroot, which indicates an improperly configured / broken toolchain. Sysroot being hardcoded is a problem on its own at the first place.

> You cannot say it don't worked because whole toolchain is broken.

I never said that. I had a suspicion that there is something wrong with this toolchain but instead of talking properly and proving that it is not, you chose to provoke so I think yeah there must be something wrong with the toolchain.

That being said, I will not be further involved with this conversation.

abhiTronix, 2021-01-24:
@fuzun Stop spamming like mongers and noone asked specifically you to stick to this repository, there are lots of option on the internet _(considering you a legitimate spammer)_ , you can move onto them. Thanks for wasting our time.