abhiTronix/raspberry-pi-cross-compilers #60: Hello World cmake with toolchain example?

oosavu, 2021-01-18:
Hello!!!
My enviroment is:
- ubuntu 20.04 as "host machine"
- "raspberry pi zero w" with  Raspbian GNU/Linux 10 (buster)  as a target machine
- cross-tool extracted from: cross-gcc-8.3.0-pi_0-1.tar.gz ("Raspberry Pi GCC Cross-Compiler Toolchains(Buster)")
- synced rootfs from raspberry to the host OS
- simple hello_world c++ cmake project with simple toolchain file (actually i have a big project, but it is minimal example for reproducing):
toolchain_s.cmake:

```
set(CMAKE_VERBOSE_MAKEFILE ON)
SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_VERSION 1)
SET(CMAKE_SYSTEM_PROCESSOR armhf)

set(tools /home/oosavu/cv/cross-pi-gcc-8.3.0-0)
set(rootfs_dir /home/oosavu/cv/rootfs)

set(CMAKE_C_COMPILER ${tools}/bin/arm-linux-gnueabihf-gcc)
set(CMAKE_CXX_COMPILER ${tools}/bin/arm-linux-gnueabihf-g++)

SET(CMAKE_FIND_ROOT_PATH ${rootfs_dir})
SET(CMAKE_SYSROOT ${rootfs_dir})

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

SET (CMAKE_C_COMPILER_WORKS 1)
SET (CMAKE_CXX_COMPILER_WORKS 1)
```

CMakeList.txt:
```
cmake_minimum_required(VERSION 3.10)
project(rpi_main)
set(CMAKE_VERBOSE_MAKEFILE ON)
set(CMAKE_CXX_STANDARD 17)
add_executable(rpi_main main.cpp ${rpi_src})
include_directories(${rootfs_dir}/usr/include/arm-linux-gnueabihf )
```

main.cpp:
```
#include <iostream>
int main(int argc, char *argv[])
{
    std::cout << "qweqwe" << std::endl;
	return 0;
}
```

**Error occurs on the linkage step: 
ld: cannot find crt1.o: No such file or directory
ld: cannot find crti.o: No such file or directory
ld: cannot find -lm**

What i'm doing wrong?! i spent many time for migrating to this tool and it is a last obstacle. anybody please help!

output of make and cmake:
[image]

abhiTronix, 2021-01-18:
@oosavu Just add `link_directories(${rootfs_dir}/usr/lib/arm-linux-gnueabihf)` at end of your `CMakeList.txt` as follows:

```cmake
cmake_minimum_required(VERSION 3.10)
project(rpi_main)
set(CMAKE_VERBOSE_MAKEFILE ON)
set(CMAKE_CXX_STANDARD 17)
add_executable(rpi_main main.cpp ${rpi_src})
include_directories(${rootfs_dir}/usr/include/arm-linux-gnueabihf )
link_directories(${rootfs_dir}/usr/lib/arm-linux-gnueabihf)
```

and problem will be solved.

**Warning ⚠️ : Also make sure you ran `rsync` command with `sudo` privileges and fixed symbolic links, as mentioned in docs at [step-6](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md#6-sync-raspberry-pi-sysroot-most-important) and [step-7](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md#7-fix-symbolic-links)**

abhiTronix, 2021-01-18:
@oosavu Test solution in previous comment and comment your results here.

oosavu, 2021-01-18:
Thanks for quick reply!
- My previous rootfs  was synced without you guide for qt build (but with similar rsync command). I synced it again according to you instructions, including symlinks fixing
- Unfortunely ` link_directories(${rootfs_dir}/usr/lib/arm-linux-gnueabihf)` does not help. I have the same errors.
- **I found the mistake myself! It is the `SET(CMAKE_SYSROOT ${rootfs_dir})`!**

Similar bug has already present: https://github.com/abhiTronix/raspberry-pi-cross-compilers/issues/3
But unfortunately solutions from this old bug does not help for me :(

So for now commenting this line in the toolchain file is only that i want. It is not good because it broke many benefits of cmake, like "find_library". 

If you have another ideas what can i do - it will be great!

p.s.: you can merge four rsync commands into one: `rsync -vR --progress -rl --delete-after --safe-links pi@192.168.0.100:/{lib,usr,opt/vc/lib} $HOME/cv/rootfs`

p.s.2: it will be great to add this hello world application to you repository since it is hard to find good example for cross-compilation project!

abhiTronix, 2021-01-19:
> Similar bug has already present: #3
> But unfortunately solutions from this old bug does not help for me :( (unset(CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES))

@oosavu Apologies but I'm unable to understand if you still having bugs after `SET(CMAKE_SYSROOT ${rootfs_dir})` fix or you already resolved them? If there are bugs, are they similar to #3?

> So for now commenting this line in the toolchain file is only that i can do. It is not good because it brokes many benefits of cmake, like "find_library".

> If you have another ideas what can i do - it will be great!

Also, by "this line" you meant "unset(CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES)" ?

> p.s.2: it will be great to add this hello world application to you repository since it is hard to find good example for cross-compilation project!

Sure, I'm going paste the results of hello world application in our wiki docs.

oosavu, 2021-01-19:
@abhiTronix

uuups. My mistake. #3  is NOT a similar bug. Forget about it.

 "this line"  - is "SET(CMAKE_SYSROOT ${rootfs_dir})".
 
 This works:
 ```
set(CMAKE_VERBOSE_MAKEFILE ON)
SET(CMAKE_SYSTEM_NAME Linux)
SET(CMAKE_SYSTEM_VERSION 1)
SET(CMAKE_SYSTEM_PROCESSOR armhf)

set(tools /home/oosavu/cv/cross-pi-gcc-8.3.0-0)
set(rootfs_dir /home/oosavu/cv/rootfs)

set(CMAKE_C_COMPILER ${tools}/bin/arm-linux-gnueabihf-gcc)
set(CMAKE_CXX_COMPILER ${tools}/bin/arm-linux-gnueabihf-g++)

SET(CMAKE_FIND_ROOT_PATH ${rootfs_dir})

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

SET (CMAKE_C_COMPILER_WORKS 1)
SET (CMAKE_CXX_COMPILER_WORKS 1) 
```
 
[image]

abhiTronix, 2021-01-19:
@oosavu Nice. I'll close this once wiki docs are updated. Thank  you.

fuzun, 2021-01-23:
Omitting setting CMAKE_SYSROOT is not a solution. It is rather a workaround and does not work for projects where sysroot must be defined.

I think this issue is related with the toolchain. However, I don't know how it works for @abhiTronix given that I followed his guide.

abhiTronix, 2021-01-24:
@fuzun Stop Hijacking other issues, and open your own or rather just quit with your bullshit. First of all, This is not the problem with Toolchains but with CMake _(if you being ignorant)_, since CMake prefixes all paths with the `CMAKE_SYSROOT` value if defined, so `find_library` (used to find required libs for compiling` main.cpp`) will be looking in the wrong places. You can easily test this by copying your entire `$WORKSPACE` to `$CMAKE_SYSROOT/$WORKSPACE`. Then, the CMake scripts worked fine, and `find_library` is found correctly and it compiles without any problems. Therefore as @oosavu correctly concluded, only `CMAKE_FIND_ROOT_PATH` is the correct way to handle the sysroot.

fuzun, 2021-01-24:
It is not related to CMake at all. CMake just passes sysroot params and/or translates paths accordingly. It is probably the broken toolchain that causes this. I don't get this error with any other toolchain.

This issue can be reproduced when CMake is not used. In fact, your Qt guide gives the same problem if "-no-gcc-sysroot" is not used.

The `solution` that you gave in the other issue does not fix the problem. And for this case, oosavu did not resolve the problem and he is going to have trouble with libraries later on. I'm saying this because I already tried doing what he did.

Lastly, good luck with your broken toolchain and shitty attitude.

abhiTronix, 2021-01-24:
@fuzun Bring a right attitude on the table to comment here or just begone.

abhiTronix, 2021-02-05:
Closed in docs: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Raspberry-Pi-GCC-Cross-Compiler-ARM-Toolchains-CMake-Usage-Guide/17f85af96eafabab415351a21d8d653bd3feca5c

closed 2021-02-05 by abhiTronix