abhiTronix/raspberry-pi-cross-compilers #61: Crosscompile for 64 Bit Debian 10

visglz, 2021-01-20:
Hi Abhishek,

i am trying to crosscompile a small example libusb example for Raspberry PI OS 64 Bit (Debian 10 / Buster).

The example is basically [testlibusb.c][https://github.com/libusb/libusb/blob/v1.0.22/examples/testlibusb.c] from libusb version 1.0.22 (which is also installed in Raspberry PI OS Buster).

To crosscompile I downloaded "Raspberry Pi GCC 64-Bit Cross-Compiler Toolchains" for gcc 8.3.0, [link](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Bonus%20Raspberry%20Pi%20GCC%2064-Bit%20Toolchains/Raspberry%20Pi%20GCC%2064-Bit%20Cross-Compiler%20Toolchains/GCC%208.3.0/)

Besides other libs I need to fetch the aarch64 libusb from raspberry PI to a local folder, here "rootfs_part".

The problem I run into is that the libusb from Debian Buster is linked to glibc 2.28, the version Buster is build with, but the toolchain I downloaded is built with 2.24:

```
/opt/sdk/cross-pi-gcc-8.3.0-64/bin/aarch64-linux-gnu-gcc   -rdynamic CMakeFiles/testlibusb.dir/testlibusb.c.o  -o testlibusb  -L/home/visglz/src/vistrrdedock10/raspi/cmake/../build/raspi_libs/rootfs_part/usr/lib/aarch64-linux-gnu -lusb-1.0 -L/home/visglz/src/vistrrdedock10/raspi/cmake/../build/raspi_libs/rootfs_part/lib/aarch64-linux-gnu -ludev 
/home/visglz/src/vistrrdedock10/raspi/cmake/../build/raspi_libs/rootfs_part/lib/aarch64-linux-gnu/libudev.so: undefined reference to `getrandom@GLIBC_2.25'
/home/visglz/src/vistrrdedock10/raspi/cmake/../build/raspi_libs/rootfs_part/lib/aarch64-linux-gnu/libudev.so: undefined reference to `fcntl64@GLIBC_2.28'
/home/visglz/src/vistrrdedock10/raspi/cmake/../build/raspi_libs/rootfs_part/lib/aarch64-linux-gnu/libudev.so: undefined reference to `reallocarray@GLIBC_2.26'
```

```
$ find /opt/sdk/cross-pi-gcc-8.3.0-64 -name "libc-2*so"
/opt/sdk/cross-pi-gcc-8.3.0-64/aarch64-linux-gnu/libc/lib64/libc-2.24.so
```

Wouldn't it make sense to also provide a 64 Bit version with the Compiler and glibc matching for Debian Buster? I can't find such a combination in `builder_64.yml` currently.

Locally I built a toolchain with the following settings:

	./RTBuilder_64b -g 8.3.0 -t 2
	
With the resulting crosscompiler I was able to crossbuild `testlibusb.c` and let run the resulting binary on the pi.

BTW: Great work you are doing with this project. 👍 

Regards
   Georg

abhiTronix, 2021-01-20:
> To crosscompile I downloaded "Raspberry Pi GCC 64-Bit Cross-Compiler Toolchains" for gcc 8.3.0, link
>
> Besides other libs I need to fetch the aarch64 libusb from raspberry PI to a local folder, here [...]

@visglz Thanks. Yeah that's right. I  wasn't aware buster 64bit OS is already out. I've been out of the loop lately 😄 . I'll add this to TODO list and will work on it few days. 

> Wouldn't it make sense to also provide a 64 Bit version with the Compiler and glibc matching for Debian Buster? I can't find such a combination in builder_64.yml currently.
>
> Locally I built a [...]

Kudos for finding this 🎉 . I've already predicted that Raspbian will soon release 64 bit version so I kept this hidden option for it. Also, Thanks again for testing it.

abhiTronix, 2021-01-21:
> /opt/sdk/cross-pi-gcc-8.3.0-64/bin/aarch64-linux-gnu-gcc -Llib -Iinclude -lpng -lz pngtest.c
> 
> -> Works, a.out is generated and can be run on raspberry pi.

@visglz What's the `strace` on this command _(with downloaded project's binary)_? I'm guessing your locally build binaries is incomplete, but not sure.

abhiTronix, 2021-01-21:
@visglz Also can you run `nm` command on both `a.out` _(i.e. from project's GCC binary and your locally build)_ and paste the output here.

visglz, 2021-01-21:
> > /opt/sdk/cross-pi-gcc-8.3.0-64/bin/aarch64-linux-gnu-gcc -Llib -Iinclude -lpng -lz pngtest.c
> > -> Works, a.out is generated and can be run on raspberry pi.
> 
> @visglz What's the `strace` on this command _(i.e. with downloaded project's binary)_? I'm guessing your locally build binaries are incomplete, but not sure.

Please see here for the relevant strace section:

```
openat(AT_FDCWD, "/opt/sdk/cross-pi-gcc-8.3.0-64/bin/../aarch64-linux-gnu/libc/etc/ld.so.conf", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/opt/sdk/cross-pi-gcc-8.3.0-64/bin/../aarch64-linux-gnu/libc/etc/ld.so.conf", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/opt/sdk/cross-pi-gcc-8.3.0-64/bin/../aarch64-linux-gnu/libc/aarch64-linux-gnu/lib64/libm.so.6", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/opt/sdk/cross-pi-gcc-8.3.0-64/bin/../aarch64-linux-gnu/libc/usr/local/lib64/libm.so.6", O_RDONLY) = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/opt/sdk/cross-pi-gcc-8.3.0-64/bin/../aarch64-linux-gnu/libc/lib64/libm.so.6", O_RDONLY) = 20
```

My guess is also that my local build gcc is somewhat different :-) , so I re-run `./RTBuilder_64b` two times - same result.

Interestingly the compiler finds the filename without problems when asked:

```
$ /opt/sdk/cross-pi-gcc-8.3.0-64_2/bin/aarch64-linux-gnu-gcc -print-file-name=libm.so.6
/opt/sdk/cross-pi-gcc-8.3.0-64_2/bin/../aarch64-linux-gnu/libc/lib/../lib64/libm.so.6
```

Please find attached the output of nm:

* [nm_prebuild_gcc.txt](https://github.com/abhiTronix/raspberry-pi-cross-compilers/files/5850468/nm_prebuild_gcc.txt)
* [nm_ubuntu_gcc.txt](https://github.com/abhiTronix/raspberry-pi-cross-compilers/files/5850469/nm_ubuntu_gcc.txt)

I also dumped the specs (gcc --dumpspecs and printed a few variables like compiled in sysroots, see the following attachments:

* [gcc_dump_specs.txt](https://github.com/abhiTronix/raspberry-pi-cross-compilers/files/5850433/gcc_dump_specs.txt)
* [gcc_print_variables.txt](https://github.com/abhiTronix/raspberry-pi-cross-compilers/files/5850434/gcc_print_variables.txt)

abhiTronix, 2021-01-21:
@visglz Thanks. There were some important sysroot patches missing from build-scripts, and is fixed in commit https://github.com/abhiTronix/raspberry-pi-cross-compilers/commit/fa538612843abd1f82aac1ee211abcbf7e3a2359.  Can you try that branch by re-cloning as follows:

```sh
git clone https://github.com/abhiTronix/raspberry-pi-cross-compilers.git && cd raspberry-pi-cross-compilers
git checkout patch@buildscript
```

And then rebuild your binaries by running `./RTBuilder_64b` again.

abhiTronix, 2021-01-22:
@visglz Please try the given solution.

visglz, 2021-01-22:
Thanks for the patch! I rebuild locally and can confirm that the compiler works now like it should in my example:

	/opt/sdk/cross-pi-gcc-8.3.0-64_3/bin/aarch64-linux-gnu-gcc -Llib -Iinclude -lpng -lz pngtest.c

I found only a minor issue:

The resulting tar file is named `cross-gcc-8.3.0-pi_.tar.gz`. I am not sure about the naming, but I guess it should be a `64` at the end? (RPI_TYPE evalutates to empty string in the script).

Also I had a issue where one wget command failed because of flaky network and extracted an corrupted folder. It took me some time to find out the reason why the build wasn't working . Do you mind to put a "set -eo pipefail" at the top of the bash script? That way any failed command will stop the script and exit with return code != 0. Currently if an error occurs in any of the steps it is silently ignored and the script proceeds with the next instruction.

abhiTronix, 2021-01-22:
> Thanks for the patch! I rebuild locally and can confirm that the compiler works now like it should in my example:
> 
> /opt/sdk/cross-pi-gcc-8.3.0-64_3/bin/aarch64-linux-gnu-gcc -Llib -Iinclude -lpng -lz pngtest.c

@visglz Thanks for the update. I somehow forgot to patch the build scripts. 🤔 

> I found only a minor issue:
> 
> The resulting tar file is named cross-gcc-8.3.0-pi_.tar.gz. I am not sure about the naming, but I guess it should be a 64 at the end? (RPI_TYPE evalutates to empty string in the script).

@visglz Wow, that is also a bug, It should be `FOLDER_VERSION` instead of `RPI_TYPE` . I wondered how it remain hidden for long. 

> Also I had a issue where one wget command failed because of flaky network and extracted an corrupted folder. It took me some time to find out the reason why the build wasn't working . Do you mind to [...]

Sure thing, this will be a great addition to build-scripts. I will close this issue when all these bugs and enhancements are added.

abhiTronix, 2021-01-22:
@visglz I think `set -euxo pipefail` will be more appropriate as _`-x` is used for debugging/xtrace and nounset (a.k.a. set `-u` ) used to exit when script tries to use undeclared variables._

visglz, 2021-01-22:
@abhiTronix Yes, this would be useful. In fact I run nearly all my bash scripts with something like the above. :-)

"set -x" is a bit noisy but might be worthwile to set for debug and error reporting.

If you enable "-u" you would need to check the availbility of variables in another way, this is what I do:

```
if [ -z "${fooxyz:-}" ]; then
    echo "Fooxyz not set";
fi
```

abhiTronix, 2021-01-22:
> @abhiTronix Yes, this would be useful. In fact I run nearly all my bash scripts with something like the above. :-)
> 
> "set -x" is a bit noisy but might be worthwile to set for debug and error reporting.

@visglz I think we can add `-v` verbose flag in script for debugged output. Otherwise `set -eo pipefail  is good.

> If you enable "-u" you would need to check the availbility of variables in another way, this is what I do:
> 
> if [ -z "${fooxyz:-}" ]; then
>     echo "Fooxyz not set";
> fi

 Wow, that will be a extra burden. Let's leave it for now, and If it is necessary will be add this the future.

abhiTronix, 2021-01-22:
@visglz Also, `-v` flag works better than `-x`, as it produce less noisy output.

abhiTronix, 2021-01-23:
@visglz Please review PR #62, and suggest any changes if there is a scope of improvement. Thank you.

visglz, 2021-01-26:
There is still one more minor glitch :) , its more a question than an issue, but I would be interested in your option.

When I compiled the pngtest example I was wondering why I had to provide the transitive dependency to libz (-lz):

	/opt/sdk/cross-pi-gcc-8.3.0-64/bin/aarch64-linux-gnu-gcc -Llib -Iinclude pngtest.c -lpng -lz

With with the native compiler I didn't need to provide the transitive dependency:

	gcc pngtest.c -lpng

Also with other crosscompile toolchains like yocto (where compiler and target libraries and delivered together as a package) I didn't need to provide transitive dependencies.

When I ommit "-lz" like the following the link step fails:

	/opt/sdk/raspi64-202101221000/compiler/bin/aarch64-linux-gnu-gcc -Llib -Iinclude pngtest.c -lpng

Here is the output (for reasons I'll explain later I moved cross-pi-gcc-8.3.0-64 to `/opt/sdk/raspi64-202101221000/compiler`):

```
pngtest.c: In function 'test_one_file':
pngtest.c:1607: warning: "pass_height" redefined
 #        define pass_height height
 
pngtest.c:1544: note: this is the location of the previous definition
 #              define pass_height frame_height
 
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: warning: libz.so.1, needed by lib/libpng.so, not found (try using -rpath or -rpath-link)
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `inflate'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `crc32'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `deflateReset'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `deflate'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `deflateInit2_'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `inflateEnd'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `adler32'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `deflateEnd'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `inflateInit2_'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `inflateReset'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `inflateValidate@ZLIB_1.2.9'
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: lib/libpng.so: undefined reference to `inflateReset2@ZLIB_1.2.3.4'
collect2: error: ld returned 1 exit status
```

It knows that libz is a dependency but can't resolve it.

Here is the strace output:

```
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/aarch64-linux-gnu/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/usr/local/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/usr/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/aarch64-linux-gnu/lib/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/usr/local/lib/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/lib/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/usr/lib/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/aarch64-linux-gnu/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/usr/local/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/opt/sdk/raspi64-202101221000/compiler/bin/../aarch64-linux-gnu/libc/usr/lib64/libz.so.1", O_RDONLY) = -1 ENOENT (No such file or directory)
[... cut ...]
The only additional parameter is for sysroot (because of external libraries for sysroot). But this can be set e.g. in the CMake toolchain file, because its specific to the toolchain/target.

What do you think? Is this approach legit? I can remember that here has been discussion in some other ticket about setting sysroot or not.

Background is that I would like to avoid toolchain specifics in the projects CMake files (like explicitely defining transitive dependencies).

Hm. The raspberry PI folder structure is using "aarch64-linux-gnu" in its folder names while cross-pi-gcc is using `lib64`. I can live with the symlink workaround, but I wonder a bit about the difference.

abhiTronix, 2021-01-26:
> With with the native compiler I didn't need to provide the transitive dependency:
> 
> gcc pngtest.c -lpng

@visglz Behavior of indirect linking for shared libraries varies from linker to linker, so no actual comparison here. According to this [article](https://wiki.ubuntu.com/NattyNarwhal/ToolchainTransition), responsible `ld` option is `--no-copy-dt-needed-entries` (sometimes called `--no-add-needed`, which in latest gcc releases (>4.5) is enabled by default.

Also is `LD_LIBRARY_PATH=/<extracted folder-name e.g cross-pi-gcc-9.2.0-1>/lib64:$LD_LIBRARY_PATH` set? Try setting it first and see it resolves the problem.

visglz, 2021-01-27:
Thanks for the link, didn't knew about that option. Looks like `--copy-dt-needed-entries` is the  default for all my gcc based compilers. Otherwise it wouldn't search libz in first place.

The folder structure is now the following:

```
/opt/sdk/raspi64-202101221000
	compiler
	rootfs
```

`compiler` is where I installed the locally build cross-pi-gcc, `rootfs` is where I copied the rootfs of the raspberry Pi. The links which I made above was in the rootfs folder, not in the compiler folder. I did not touch the contents of cross-pi-gcc or changed anything within.

The path `compiler/lib64` is not existing.

So I set `LD_LIBRARY_PATH` to `compiler/lib` following the [instructions](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/64-Bit-Cross-Compiler:-Installation-Instructions#c1-temporary-installation-use-these-binaries-directly-recommended):

```
export RASPI64_BASE=/opt/sdk/raspi64-202101221000
export PATH=$RASPI64_BASE/compiler/bin:$PATH
export LD_LIBRARY_PATH=$RASPI64_BASE/compiler/lib
$RASPI64_BASE/compiler/bin/aarch64-linux-gnu-gcc -L${RASPI64_BASE}/rootfs/usr/lib/aarch64-linux-gnu -I${RASPI64_BASE}/rootfs/usr/include pngtest.c -lpng
```

Hm. This doesn't work as the linker is now searching under `/lib`:

```
pngtest.c: In function 'test_one_file':
pngtest.c:1607: warning: "pass_height" redefined
 #        define pass_height height
 
pngtest.c:1544: note: this is the location of the previous definition
 #              define pass_height frame_height
 
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: cannot find /lib/aarch64-linux-gnu/libc.so.6
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: cannot find /usr/lib/aarch64-linux-gnu/libc_nonshared.a
/opt/sdk/raspi64-202101221000/compiler/bin/../lib/gcc/aarch64-linux-gnu/8.3.0/../../../../aarch64-linux-gnu/bin/ld: cannot find /lib/aarch64-linux-gnu/ld-linux-aarch64.so.1
collect2: error: ld returned 1 exit status
```
What works is setting the sysroot (with the symlinked / patched rootfs folder):

```
export RASPI64_BASE=/opt/sdk/raspi64-202101221000
$RASPI64_BASE/compiler/bin/aarch64-linux-gnu-gcc --sysroot=$RASPI64_BASE/rootfs pngtest.c -lpng
```

Personally I try to avoid `LD_LIBRARY_PATH` where possible, because of potential side effects:

* https://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html#AEN80
* http://xahlee.info/UnixResource_dir/_/ldpath.html

For me the cross-pi compiler is working perfectly without LD_LIBRARY_PATH as long as I don't link to external libraries. When I link to those (like libz) setting the sysroot like it is described in the Qt wiki entry works - except for the minor glitch with the slightly patched rootfs.

abhiTronix, 2021-01-27:
@visglz `-I${RASPI64_BASE}/rootfs/usr/include` must be `-I${RASPI64_BASE}/rootfs/usr/include/aarch64-linux-gnu`

abhiTronix, 2021-01-27:
> Hm. This doesn't work as the linker is now searching under /lib:

This is weird behaviour. I'll look into it. But for now, stick with what works. Also, see if there's a better settings available for handling sysroot, IDK but it works erratic occasionally.

abhiTronix, 2021-02-05:
**Patches merged in commit https://github.com/abhiTronix/raspberry-pi-cross-compilers/commit/7ead221f4f744a9dcd10f3ef07005422db68ec9e**

commit fa538612 (2021-01-21): Patched `RTBuilder_64b` build-script sysroot (Fix #61)
commit b8aa4b6e (2021-01-22): Build-Script: Critical Bug Fixes (Fixes #61)

- Extended build-script sysroot patch to `RTBuilder_32b`.
- Replaced incorrect `RPI_TYPE` with `FOLDER_VERSION`.
- Replaced `OS_TYPE` with `RPIOS_TYPE` in `RTBuilder_64b.
- Added Buster and Stretch native support for `RTBuilder_64b` build script.
commit ba04de84 (2021-01-22): Build Scripts: Added Verbose Flag for debugged output (Fixes #61)
commit d2f39723 (2021-01-23): CI: Added Buster/Stretch OS support for 64-bit envs (Fixes #61)

- Updated CI Bash scripts with new `-o` flag to support OS selection.
- Updated YAMLs to include buster & stretch Pi OS support.
- Updated and added new docs.
closed 2021-02-05 by abhiTronix