abhiTronix/raspberry-pi-cross-compilers #64: /usr/include/features.h:424:12: fatal error: sys/cdefs.h

vrichomme, 2021-02-03:
Hi,

I have a raspberry pi 4 8Gb running raspbian 64 bits (Buster):
pi@jarvis2:~/Dev $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
pi@jarvis2:~/Dev $ uname -a
Linux jarvis2 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux

(I have downloaded native-gcc-10.2.0-pi_64.tar.gz)

sudo tar xvf ~/Downloads/native-gcc-10.2.0-pi_64.tar.gz -C /opt/
sudo mv /opt/native-pi-gcc-10.2.0-64 /opt/gcc-10.2.0
echo 'export PATH=/opt/gcc-10.2.0/bin:$PATH' >> ~/.bashrc
echo 'export LD_LIBRARY_PATH=/opt/gcc-10.1.0/lib:$LD_LIBRARY_PATH' >> ~/.bashrc
. ~/.bashrc

then I try to compile the following file :

`#include <iostream>
 
int main() {
     // if block with init-statement:
     if(int a = 5; a < 8) {
         std::cout << "Local variable a is < 8\n";
      } else {
          std::cout << "Local variable a is >= 8\n";
      }
     return 0;
}`
pi@jarvis2:~/Dev $ g++-10.2.0 -std=c++17 -Wall -pedantic if_test.cpp -o if_test
In file included from /opt/gcc-10.2.0/include/c++/10.2.0/aarch64-linux-gnu/bits/os_defines.h:39,
                 from /opt/gcc-10.2.0/include/c++/10.2.0/aarch64-linux-gnu/bits/c++config.h:518,
                 from /opt/gcc-10.2.0/include/c++/10.2.0/iostream:38,
                 from if_test.cpp:1:
/usr/include/features.h:424:12: fatal error: sys/cdefs.h: Aucun fichier ou dossier de ce type
  424 | #  include <sys/cdefs.h>

abhiTronix, 2021-02-03:
> I have a raspberry pi 4 8Gb running raspbian 64 bits (Buster)

@vrichomme Support for Raspberry Pi 4 (Buster) is still in development. Wait for PR  #62 to be merged or compile it manually as follows:

```sh
#clone patch
git clone https://github.com/abhiTronix/raspberry-pi-cross-compilers.git && cd raspberry-pi-cross-compilers
git checkout patch@buildscript

#and compile binary:
./RTBuilder_64b -g 10.2.0 -t 2
```

vrichomme, 2021-02-03:
Thanks.

When I try I have this:

libtool: link: ranlib .libs/libgmp.a
libtool: link: rm -fr .libs/libgmp.lax
libtool: link: ( cd ".libs" && rm -f "libgmp.la" && ln -s "../libgmp.la" "libgmp.la" )
make[1]: *** [Makefile:27341: stage2-bubble] Error 2
make: *** [Makefile:952: all] Error 2

vrichomme, 2021-02-03:
Do I need to run it on the raspbian or on a x86_64 host ?
Because on the raspi I don't understand why it recompiles gcc-8.3 when it's already available on the system ?
And the first time I run the script I had an error and the second times I thought it would display the same error but now it's starting again from zero I think...

abhiTronix, 2021-02-04:
Yeah obviously you'll need any x86_64 linux machine/system to cross-compile for raspberry pi. You can't cross-compile for raspberry pi itself.

abhiTronix, 2021-02-04:
Script will generate two files: `native-gcc-{GCC_VERSION}-pi_64.tar.gz` and `cross-gcc-{GCC_VERSION}-pi_64.tar.gz` at your $HOME director of x86_64 linux machine. Use `native-gcc-{GCC_VERSION}-pi_64.tar.gz` on your pi4.

vrichomme, 2021-02-04:
Ok I compiled the native and copied it  but  still the same error...

abhiTronix, 2021-02-05:
@vrichomme Did you setup the important symlink as mentioned in wiki docs:

```sh
 sudo ln -sf /usr/include/aarch64-linux-gnu/asm/* /usr/include/asm
 sudo ln -sf /usr/include/aarch64-linux-gnu/gnu/* /usr/include/gnu
 sudo ln -sf /usr/include/aarch64-linux-gnu/bits/* /usr/include/bits
 sudo ln -sf /usr/include/aarch64-linux-gnu/sys/* /usr/include/sys
 sudo ln -sf /usr/include/aarch64-linux-gnu/openssl/* /usr/include/openssl
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crtn.o /usr/lib/crtn.o
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crt1.o /usr/lib/crt1.o
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crti.o /usr/lib/crti.o
```

vrichomme, 2021-02-05:
My bad. now it works. Sorry for the noise

abhiTronix, 2021-02-05:
@vrichomme Idk sometimes we need `/*` and other times we don't. I'll look into this.

abhiTronix, 2021-02-05:
@vrichomme Ok these are new commands:

```sh
 sudo ln -sf /usr/include/aarch64-linux-gnu/asm/* /usr/include/asm/
 sudo ln -sf /usr/include/aarch64-linux-gnu/gnu/* /usr/include/gnu/
 sudo ln -sf /usr/include/aarch64-linux-gnu/bits/* /usr/include/bits/
 sudo ln -sf /usr/include/aarch64-linux-gnu/sys/* /usr/include/sys/
 sudo ln -sf /usr/include/aarch64-linux-gnu/openssl/* /usr/include/openssl/
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crtn.o /usr/lib/crtn.o
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crt1.o /usr/lib/crt1.o
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crti.o /usr/lib/crti.o
```

closed 2021-02-05 by abhiTronix