abhiTronix/raspberry-pi-cross-compilers #65: G++ does not recognize -fmodules-ts

MartinMajewski, 2021-02-05:
Hi there,

I am trying to compile the latest and greatest C++20 features but fail to get even the minimal module example to work on my Raspberry Pi 4 with Buster (64bit). I've installed the native GCC-10.2.0 toolchain.

However, in order to build modules, I'll have to use the "-fmodules-ts" compiler argument, which is rejected by telling me that it is an unrecognized option.

What am I missing? The GNU GCC tools should support that by that time.

Best wishes,
Martin

abhiTronix, 2021-02-06:
@MartinMajewski The repository is in unstable state due to yesterday's update. Please wait few hours before stable binaries are build or you can manually build them on a x86_64 linux machine as follows:
```sh
# update your system
sudo apt update && sudo apt dist-upgrade

# install prerequisites
sudo apt-get -y install gcc g++ gperf flex texinfo gawk gfortran texinfo bison \
    build-essential openssl unzip wget git pigz \
    libncurses-dev autoconf automake tar figlet

# clone patch
git clone https://github.com/abhiTronix/raspberry-pi-cross-compilers.git && cd raspberry-pi-cross-compilers

# and compile binary:
cd build-scripts
./RTBuilder_64b -g "10.2.0" -o "buster"
```

abhiTronix, 2021-02-06:
@MartinMajewski Please wait for v3.0.6 release (happening in few hours) and then you can download it from sourceforge as usual.

MartinMajewski, 2021-02-06:
@abhiTronix Thank's for the reply.

However, I had the binaries downloaded from Sourceforge on Thursday and I'm relating to the native arm64 compilers, which I want to run directly on the RPi 4 with Rasperry OS 64.

Can I assume to build the native compilers with your instruction above as well by changing into the correct subdirectory of your repo?

Best wishes,
Martin

abhiTronix, 2021-02-06:
@MartinMajewski No need. It is avaliable here: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Bonus%20Raspberry%20Pi%20GCC%2064-Bit%20Toolchains/Raspberry%20Pi%20GCC%2064-Bit%20Native-Compiler%20Toolchains/Buster/GCC%2010.2.0/

abhiTronix, 2021-02-06:
@MartinMajewski **Also, use these exact wiki docs to setup these toolchains which are given here:** https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/64-Bit-Native-Compiler:-Installation-Instructions#c-extracting-and-linking-binary

Goodluck, and let me know if something doesn't worked.

closed 2021-02-06 by abhiTronix

MartinMajewski, 2021-02-06:
@abhiTronix 

1. Just downloaded the latest release that you pointed me to:

Tested quickly with the `-fmodules-ts` option by calling g++-10.2.0 (gcc, c++, etc.) with it but got

`g++-10.2.0: error: unrecognized command-line option ‘-fmodules-ts’`

What do I not understand here?

2. I followed the wiki step by step multiple times on different fresh RPi setups but ended up pointing towards the toolchain explicitly in my IDE. I cannot recall what was wrong in detail but I will redo it as soon as I get the C++20 confusion sorted out. Then I will report if I have found the cause,

abhiTronix, 2021-02-06:
@MartinMajewski Try specifying `-std=c++20 - -fmodules-ts` instead.

abhiTronix, 2021-02-06:
@MartinMajewski Yes you need addition `-std=c++20` command-line parameter to enable C++20 support. More information here: https://gcc.gnu.org/projects/cxx-status.html

MartinMajewski, 2021-02-06:
@abhiTronix Same:

```
pi@devpi:~/Downloads/native-pi-gcc-10.2.0-64/bin $ ./g++-10.2.0 -std=c++20 -fmodules-ts
g++-10.2.0: error: unrecognized command-line option ‘-fmodules-ts’
```
I tried with -std=c++20 before and this option is recognized but -fmodules-ts is not.

abhiTronix, 2021-02-06:
Are symbolic links present:

```sh
 sudo ln -sf /usr/include/aarch64-linux-gnu/asm/* /usr/include/asm
 sudo ln -sf /usr/include/aarch64-linux-gnu/gnu/* /usr/include/gnu
 sudo ln -sf /usr/include/aarch64-linux-gnu/bits/* /usr/include/bits
 sudo ln -sf /usr/include/aarch64-linux-gnu/sys/* /usr/include/sys
 sudo ln -sf /usr/include/aarch64-linux-gnu/openssl/* /usr/include/openssl
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crtn.o /usr/lib/crtn.o
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crt1.o /usr/lib/crt1.o
 sudo ln -sf /usr/lib/aarch64-linux-gnu/crti.o /usr/lib/crti.o
```

abhiTronix, 2021-02-06:
@MartinMajewski Can you also try `-std=c++2a` instead of `-std=c++20`?

abhiTronix, 2021-02-06:
Or even `-std=gnu++20`? I don't have pi4 or even Raspbian 64-bit installed to test it  directly so this just speculations.

MartinMajewski, 2021-02-06:
@abhiTronix 

-std=c++2a produces the same result

Question about the symbolic links. The target is the first parameter, right? So this would mean to have the downloaded package copied into /usr/lib/... and /usr/include/... respectively?! But we do not do anything like this in the **Temporary Installation** section. So where do the files come from?

MartinMajewski, 2021-02-06:
```
pi@devpi:~/Downloads/native-pi-gcc-10.2.0-64/bin $ ./g++-10.2.0 -std=gnu++20 -fmodules-ts
g++-10.2.0: error: unrecognized command-line option ‘-fmodules-ts’
```

abhiTronix, 2021-02-06:
What's the output of command `./g++-10.2.0 -v --help 2> /dev/null | sed -n '/^ *-std=\([^<][^ ]\+\).*/ {s//\1/p}'
`

MartinMajewski, 2021-02-06:
```
pi@devpi:~/Downloads/native-pi-gcc-10.2.0-64/bin $ ./g++-10.2.0 -v --help 2> /dev/null | sed -n '/^ *-std=\([^<][^ ]\+\).*/ {s//\1/p}'
f2003
f2008
f2008ts
f2018
f95
gnu
legacy
c++03
c++98.
c++0x
c++11.
c++11
c++14
c++17
c++1y
c++14.
c++1z
c++17.
c++20
c++2a.
c++2a
c++98
c11
c17
c18
c1x
c11.
c2x
c89
c90.
c90
c99
c9x
c99.
gnu++03
gnu++0x
gnu++11.
gnu++11
gnu++14
gnu++17
gnu++1y
gnu++14.
gnu++1z
gnu++17.
gnu++20
gnu++2a
gnu++98
gnu11
gnu17
gnu18
gnu1x
gnu11.
gnu2x
gnu89
gnu90
gnu99
gnu9x
gnu99.
iso9899:1990
c90.
iso9899:199409
iso9899:1999
c99.
iso9899:199x
iso9899:2011
c11.
iso9899:2017
iso9899:2018
```

abhiTronix, 2021-02-06:
@MartinMajewski means `c++20` is present.

abhiTronix, 2021-02-06:
@MartinMajewski can you test this on 32bit Raspberry Pi os? Can it problem with arm64/aarch64 only?

MartinMajewski, 2021-02-06:
I have to be afk for a while now. Later, I will setup a second image with a 32bit Raspbian OS, retry, and report.

Thank you so much for now! Have a great time!

abhiTronix, 2021-02-06:
@MartinMajewski Also try other toolchains other than repository too, and let me know if you find something weird. Thank you and goodluck.

MartinMajewski, 2021-02-06:
> other toolchains other than repository too

Which other toolchains do you refer to?

abhiTronix, 2021-02-06:
@MartinMajewski https://wiki.musl-libc.org/getting-started.html uses musl instead of glibc.

MartinMajewski, 2021-02-06:
Okay, I'm back but with bad news:

```
pi@devpi32:~/development/toolchains/gcc/gcc_10_2_0_2/bin $ g++ -std=c++2a -fmodules-ts
g++: error: unrecognized command line option ‘-fmodules-ts’; did you mean ‘-fmoduleinfo’?
```

Freshly installed Rasperry OS Buster 32Bit, same issue.

Btw.:

```
sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm
 sudo ln -sf /usr/include/arm-linux-gnueabihf/gnu/* /usr/include/gnu
 sudo ln -sf /usr/include/arm-linux-gnueabihf/bits/* /usr/include/bits
 sudo ln -sf /usr/include/arm-linux-gnueabihf/sys/* /usr/include/sys
 sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl
```

This does not work! One has to remove `/*` to symlink a folder!

MartinMajewski, 2021-02-06:
```
pi@devpi32:~/development/toolchains/gcc/gcc_10_2_0_2/bin $ ./g++-10.2.0 -v --help 2> /dev/null | sed -n '/^ *-std=\([^<][^ ]\+\).*/ {s//\1/p}'
f2003
f2008
f2008ts
f2018
f95
gnu
legacy
c++03
c++0x
c++11
c++14
c++17
c++1y
c++1z
c++20
c++2a
c++98
c11
c17
c18
c1x
c2x
c89
c90
c99
c9x
gnu++03
gnu++0x
gnu++11
gnu++14
gnu++17
gnu++1y
gnu++1z
gnu++20
gnu++2a
gnu++98
gnu11
gnu17
gnu18
gnu1x
gnu2x
gnu89
gnu90
gnu99
gnu9x
iso9899:1990
iso9899:199409
iso9899:1999
iso9899:199x
iso9899:2011
iso9899:2017
iso9899:2018
```

MartinMajewski, 2021-02-06:
Building the toolchain for 32bit, RPi 3+, Buster now myself.

Have you tried these steps yourself for 10.2.0? I mean, even without any symlinks set, just pure out of the downloaded GCC bin directory g++ should be able to consume this option!

MartinMajewski, 2021-02-06:
Seems to be a widely spread issue:

```
pi@devpi32:~/development/toolchains/arm-linux-musleabihf-native/bin $ ./g++ -std=c++2a -fmodules-ts
g++: error: unrecognized command-line option ‘-fmodules-ts’
```
Something is absolutely not in sync with what GNU GCC is claiming... :-/

abhiTronix, 2021-02-06:
> Btw.:
>
> sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/gnu/* /usr/include/gnu
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/bits/* [...]

@MartinMajewski Ok, I'll check it. Thanks.

abhiTronix, 2021-02-06:
> Seems to be a widely spread issue:
> 
> pi@devpi32:~/development/toolchains/arm-linux-musleabihf-native/bin $ ./g++ --version
> g++ (GCC) 10.2.1 20210116
> Copyright (C) 2020 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> 
> pi@devpi32:~/development/toolchains/arm-linux-musleabihf-native/bin $ ./g++ -std=c++2a -fmodules-ts
> g++: error: unrecognized command-line option ‘-fmodules-ts’
> Something is absolutely not in sync with what GNU GCC is claiming... :-/

C++20 Support in GCC support is still experimental and I'm guessing it hasn't been ported to ARM architecture yet. I'm not sure but you can raise a issue and ask there developers directly: https://gcc.gnu.org/bugs/

abhiTronix, 2021-02-06:
@MartinMajewski I was right, here's a similar issue: https://stackoverflow.com/questions/58960695/c20-modules-in-arm-none-eabi-g

MartinMajewski, 2021-02-06:
But this is from 2019... with release 10.2 most of the features are implemented. 🤔 I don't see a reason for a language feature being platform dependent. But maybe I'm wrong.

I'm compiling now directly from the original sources. We'll have to wait and see.

Moreover, I will try to get my hands on cross compilers that work on Mac but compile for Pi. Haven't tested this yet. I'll report as soon as new insights become available.

Best wishes,
Martin