abhiTronix/raspberry-pi-cross-compilers #67: Passing --sysroot option to compiler does not seem to work ?

jcelerier, 2021-03-01:
Hello,
from a x64 linux host, cross-compiling for a 32-bit Pi 4:
(/opt/ossia-sdk-rpi/pi/sysroot/  contains the rsync'd /usr/lib and /usr/include from my pi)

```bash
echo 'int main() { return 0; }' > foo.cpp
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/arm-linux-gnueabihf-g++ \
    --sysroot=/opt/ossia-sdk-rpi/pi/sysroot/ \
    foo.cpp
```

gives me 
```bash
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: Aucun fichier ou dossier de ce type
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: Aucun fichier ou dossier de ce type
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status
```

what am I missing ? 

I tried adding a ton of link paths but that did not help: 

```bash
-L/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/lib/ 
-L/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/arm-linux-gnueabihf/lib/ 
-L/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/arm-linux-gnueabihf/libc/usr/lib/ 
-L/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/arm-linux-gnueabihf/libc/lib/ 
-L/opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf 
```

abhiTronix, 2021-03-01:
> echo 'int main() { return 0; }' > foo.cpp
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/arm-linux-gnueabihf-g++ \
    --sysroot=/opt/ossia-sdk-rpi/pi/sysroot/ \
    foo.cpp

@jcelerier Just add `-L/opt/ossia-sdk-rpi/pi/sysroot/ usr/lib/arm-linux-gnueabihf -I/opt/ossia-sdk-rpi/pi/sysroot/ usr/include/arm-linux-gnueabihf` flags to it and should work.

jcelerier, 2021-03-01:
It finds libm but not crt1 / crti.o:

```
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/arm-linux-gnueabihf-g++  --sysroot=/opt/ossia-sdk-rpi/pi/sysroot -L/opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf -I/opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf foo.cpp                                                                                                                 
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: Aucun fichier ou dossier de ce type
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: Aucun fichier ou dossier de ce type
collect2: error: ld returned 1 exit status
```

(the files are there: 

```
 ls -lha /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/*.o 
-rw-r--r-- 1 root root 1,5K 14 mai    2019 /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/crt1.o
-rw-r--r-- 1 root root 1,2K 14 mai    2019 /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/crti.o
-rw-r--r-- 1 root root  788 14 mai    2019 /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/crtn.o
-rw-r--r-- 1 root root 2,0K 14 mai    2019 /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/gcrt1.o
-rw-r--r-- 1 root root  444 14 mai    2019 /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/Mcrt1.o
-rw-r--r-- 1 root root 1,6K 14 mai    2019 /opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf/Scrt1.o
```
)

abhiTronix, 2021-03-01:
@jcelerier after adding those flags?

jcelerier, 2021-03-01:
yes, see the first lines.

verbose compiler output:
```
$  /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/arm-linux-gnueabihf-g++  --sysroot=/opt/ossia-sdk-rpi/pi/sysroot -L/opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf -I/opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf foo.cpp  -v
Using built-in specs.
COLLECT_GCC=/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/arm-linux-gnueabihf-g++
COLLECT_LTO_WRAPPER=/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/tmp/cross-pi-gcc-10.2.0-2/arm-linux-gnueabihf/libc --with-arch=armv8-a --with-fpu=neon-fp-armv8 --with-float=hard --disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-L/opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf' '-I' '/opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-fp-armv8' '-mtls-dialect=gnu' '-marm' '-march=armv8-a+simd'
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/cc1plus -quiet -v -I /opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf -iprefix /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/ -isysroot /opt/ossia-sdk-rpi/pi/sysroot -D_GNU_SOURCE foo.cpp -quiet -dumpbase foo.cpp -mfloat-abi=hard -mfpu=neon-fp-armv8 -mtls-dialect=gnu -marm -march=armv8-a+simd -auxbase foo -version -o /tmp/ccgYYqUI.s
GNU C++14 (GCC) version 10.2.0 (arm-linux-gnueabihf)
	compiled by GNU C version 7.5.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include"
ignoring nonexistent directory "/opt/ossia-sdk-rpi/pi/sysroot/usr/local/include"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/include
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include
 /opt/ossia-sdk-rpi/pi/sysroot/usr/include
End of search list.
GNU C++14 (GCC) version 10.2.0 (arm-linux-gnueabihf)
	compiled by GNU C version 7.5.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: fe3d46b46971cd3b57cc2f549fffafbf
COLLECT_GCC_OPTIONS='-L/opt/ossia-sdk-rpi/pi/sysroot/usr/lib/arm-linux-gnueabihf' '-I' '/opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-fp-armv8' '-mtls-dialect=gnu' '-marm' '-march=armv8-a+simd'
 /opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/as -v -I /opt/ossia-sdk-rpi/pi/sysroot/usr/include/arm-linux-gnueabihf -march=armv8-a -mfloat-abi=hard -mfpu=neon-fp-armv8 -meabi=5 -o /tmp/ccYLNQbI.o /tmp/ccgYYqUI.s
GNU assembler version 2.31 (arm-linux-gnueabihf) using BFD version (GNU Binutils) 2.31
COMPILER_PATH=/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/:/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../libexec/gcc/:/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/:/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/:/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/lib/:/opt/ossia-sdk-rpi/pi/sysroot/li [... cut ...] t/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/crtend.o crtn.o
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: Aucun fichier ou dossier de ce type
/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: Aucun fichier ou dossier de ce type
collect2: error: ld returned 1 exit status
```

abhiTronix, 2021-03-01:
@jcelerier What are these duplicate paths:

```sh
gnoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include"
ignoring nonexistent directory "/opt/ossia-sdk-rpi/pi/sysroot/usr/local/include"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed"
ignoring duplicate directory "/opt/ossia-sdk-rpi/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include"
```

abhiTronix, 2021-03-01:
@jcelerier Also this line `ignoring nonexistent directory "/opt/ossia-sdk-rpi/pi/sysroot/usr/local/include"` check if sysroot is synced properly.

abhiTronix, 2021-03-01:
@jcelerier Can you try cleaning path first with command `source /etc/environment ` and then try.

closed 2021-03-01 by abhiTronix