abhiTronix/raspberry-pi-cross-compilers #70: Cross Compiling with Cmake

rafal1137, 2021-04-11:
I have followed this GUIDE:
https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Raspberry-Pi-GCC-Cross-Compiler-ARM-Toolchains-CMake-Usage-Guide/17f85af96eafabab415351a21d8d653bd3feca5c#build-dummy-hello-world-project

I have added those to my cmake file.
```cmake
            include_directories(${SYSROOT_PATH}/usr/include/arm-linux-gnueabihf)
            link_directories(${SYSROOT_PATH}/usr/lib/arm-linux-gnueabihf)
```
Have rasbian rootfs with correct linking.

My Toolchain Cmake file. Is a bit different from the guide posted there. But that should not be an issue.

```cmake
if("$ENV{RASPBIAN_ROOTFS}" STREQUAL "")
	message(FATAL_ERROR "Define the RASPBIAN_ROOTFS environment variable to point to the raspbian rootfs.")
else()
	set(SYSROOT_PATH "$ENV{RASPBIAN_ROOTFS}")
endif()
set(TOOLCHAIN_HOST "arm-linux-gnueabihf")

message(STATUS "Using sysroot path: ${SYSROOT_PATH}")

set(TOOLCHAIN_CC "${TOOLCHAIN_HOST}-gcc")
set(TOOLCHAIN_CXX "${TOOLCHAIN_HOST}-g++")
set(TOOLCHAIN_LD "${TOOLCHAIN_HOST}-ld")
set(TOOLCHAIN_AR "${TOOLCHAIN_HOST}-ar")
set(TOOLCHAIN_RANLIB "${TOOLCHAIN_HOST}-ranlib")
set(TOOLCHAIN_STRIP "${TOOLCHAIN_HOST}-strip")
set(TOOLCHAIN_NM "${TOOLCHAIN_HOST}-nm")

set(CMAKE_CROSSCOMPILING TRUE)
set(CMAKE_SYSROOT "${SYSROOT_PATH}")

# Define name of the target system
set(CMAKE_SYSTEM_NAME "Linux")
set(CMAKE_SYSTEM_PROCESSOR "armv7l")

# Define the compiler
set(CMAKE_C_COMPILER ${TOOLCHAIN_CC})
set(CMAKE_CXX_COMPILER ${TOOLCHAIN_CXX})
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)

set(CMAKE_PREFIX_PATH "${CMAKE_PREFIX_PATH};${SYSROOT_PATH}/usr/lib/${TOOLCHAIN_HOST}")
set(CMAKE_FIND_ROOT_PATH "${CMAKE_INSTALL_PREFIX};${CMAKE_PREFIX_PATH};${CMAKE_SYSROOT}")

# search for programs in the build host directories
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
# for libraries and headers in the target directories
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
```

But when I want to compile it. It fails.

Those are the errors/issues I am getting during complilation

```
/lib/ld-linux-armhf.so.3: No such file or directory
```

This one comes at linking procedure
```
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: No such file or directory
collect2: error: ld returned 1 exit status
```

abhiTronix, 2021-04-12:
@rafal1137 Please remove `set(CMAKE_SYSROOT "${SYSROOT_PATH}")` from your Toolchain Cmake file and try again. Make sure you clean up build directory before trying again.

abhiTronix, 2021-04-12:
@rafal1137 Please test it and report back here, check if this fixes your issue. Good luck.

abhiTronix, 2021-04-12:
@rafal1137 For more information on why `CMAKE_FIND_ROOT_PATH` is preferred over `CMAKE_SYSROOT ` for cross compiling,  see here: https://stackoverflow.com/a/34077078

rafal1137, 2021-04-12:
@abhiTronix 
So I did followed your suggestion there to remove those 2 lines but my project in short might depend on ```CMAKE_SYSROOT``` being set.
Because after testing those changes.

```Cmake
CMake Error at /usr/share/cmake-3.13/Modules/FindX11.cmake:429 (message):
  Could not find X11
....
-- Configuring incomplete, errors occurred!
```

abhiTronix, 2021-04-12:
@rafal1137 Did you made following symlinks before syncing sysroot with RPi?
```sh
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/asm /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/gnu /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/bits /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/sys /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/openssl /usr/include
sudo ln -sf /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
sudo ln -sf /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
sudo ln -sf /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
```

Also, check this comment: https://github.com/abhiTronix/raspberry-pi-cross-compilers/discussions/69#discussioncomment-419083

abhiTronix, 2021-04-12:
Also, can you test without `CMAKE_FIND_ROOT_PATH` and `CMAKE_PREFIX_PATH ` as following toolchain cmake configuration:

```cmake
if("$ENV{RASPBIAN_ROOTFS}" STREQUAL "")
	message(FATAL_ERROR "Define the RASPBIAN_ROOTFS environment variable to point to the raspbian rootfs.")
else()
	set(SYSROOT_PATH "$ENV{RASPBIAN_ROOTFS}")
endif()
set(TOOLCHAIN_HOST "arm-linux-gnueabihf")

message(STATUS "Using sysroot path: ${SYSROOT_PATH}")

set(TOOLCHAIN_CC "${TOOLCHAIN_HOST}-gcc")
set(TOOLCHAIN_CXX "${TOOLCHAIN_HOST}-g++")
set(TOOLCHAIN_LD "${TOOLCHAIN_HOST}-ld")
set(TOOLCHAIN_AR "${TOOLCHAIN_HOST}-ar")
set(TOOLCHAIN_RANLIB "${TOOLCHAIN_HOST}-ranlib")
set(TOOLCHAIN_STRIP "${TOOLCHAIN_HOST}-strip")
set(TOOLCHAIN_NM "${TOOLCHAIN_HOST}-nm")

set(CMAKE_CROSSCOMPILING TRUE)
set(CMAKE_SYSROOT "${SYSROOT_PATH}")

# Define name of the target system
set(CMAKE_SYSTEM_NAME "Linux")
set(CMAKE_SYSTEM_PROCESSOR "armv7l")

# Define the compiler
set(CMAKE_C_COMPILER ${TOOLCHAIN_CC})
set(CMAKE_CXX_COMPILER ${TOOLCHAIN_CXX})
set(CMAKE_C_COMPILER_WORKS 1)
set(CMAKE_CXX_COMPILER_WORKS 1)

# search for programs in the build host directories
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
# for libraries and headers in the target directories
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
```

but keep `CMAKE_SYSROOT` and `CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER` as previous.

rafal1137, 2021-04-12:
@abhiTronix
> @rafal1137 Did you made following symlinks before syncing sysroot with RPi?
> 
> ```shell
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/asm /usr/include
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/gnu /usr/include
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/bits /usr/include
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/sys /usr/include
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/openssl /usr/include
> sudo ln -sf /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
> sudo ln -sf /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
> sudo ln -sf /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
> ```
> 
> Also, check this comment: [#69 (reply in thread)](https://github.com/abhiTronix/raspberry-pi-cross-compilers/discussions/69#discussioncomment-419083)

This may be tricky because I do not own RPI hardware. I did used 3rd Party sh script to download raspbian.iso and rsync. I can do only run rapsbian on docker. I have tried to ssh into it but I had no experience with docker before back then.

> Also, can you test without `CMAKE_FIND_ROOT_PATH` and `CMAKE_PREFIX_PATH ` as following toolchain cmake configuration:
>
> ```cmake
> if("$ENV{RASPBIAN_ROOTFS}" STREQUAL "")
> 	message(FATAL_ERROR "Define [...]

Tried that one with no luck.

abhiTronix, 2021-04-12:
> This may be tricky because I do not own RPI hardware. I did used 3rd Party sh script to download raspbian.iso and rsync. I can do only run rapsbian on docker. I have tried to ssh into it but I had no experience with docker before back then.
> 
> PS: Or will it work when doing those symlinks in rootfs ?

@rafal1137 Ok this was the actual problem. Yes, it won't work without these symlinks, see this comment https://github.com/abhiTronix/raspberry-pi-cross-compilers/discussions/69#discussioncomment-419083

abhiTronix, 2021-04-12:
@rafal1137 Can you try these commands after setting correct `RASPBIAN_ROOTFS` value:
```sh
sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/asm "$RASPBIAN_ROOTFS"/usr/include
sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/gnu "$RASPBIAN_ROOTFS"/usr/include
sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/bits "$RASPBIAN_ROOTFS"/usr/include
sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/sys "$RASPBIAN_ROOTFS"/usr/include
sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/openssl "$RASPBIAN_ROOTFS"/usr/include
sudo ln -sf "$RASPBIAN_ROOTFS"/usr/lib/arm-linux-gnueabihf/crtn.o "$RASPBIAN_ROOTFS"/usr/lib/crtn.o
sudo ln -sf "$RASPBIAN_ROOTFS"/usr/lib/arm-linux-gnueabihf/crt1.o "$RASPBIAN_ROOTFS"/usr/lib/crt1.o
sudo ln -sf "$RASPBIAN_ROOTFS"/usr/lib/arm-linux-gnueabihf/crti.o "$RASPBIAN_ROOTFS"/usr/lib/crti.o
```
and see if it works?

rafal1137, 2021-04-12:
> @rafal1137 Can you try these commands after setting correct `RASPBIAN_ROOTFS` environment variable value:
> 
> ```shell
> sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/asm "$RASPBIAN_ROOTFS"/usr/include
> sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/gnu "$RASPBIAN_ROOTFS"/usr/include
> sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/bits "$RASPBIAN_ROOTFS"/usr/include
> sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/sys "$RASPBIAN_ROOTFS"/usr/include
> sudo ln -sf -r "$RASPBIAN_ROOTFS"/usr/include/arm-linux-gnueabihf/openssl "$RASPBIAN_ROOTFS"/usr/include
> sudo ln -sf "$RASPBIAN_ROOTFS"/usr/lib/arm-linux-gnueabihf/crtn.o "$RASPBIAN_ROOTFS"/usr/lib/crtn.o
> sudo ln -sf "$RASPBIAN_ROOTFS"/usr/lib/arm-linux-gnueabihf/crt1.o "$RASPBIAN_ROOTFS"/usr/lib/crt1.o
> sudo ln -sf "$RASPBIAN_ROOTFS"/usr/lib/arm-linux-gnueabihf/crti.o "$RASPBIAN_ROOTFS"/usr/lib/crti.o
> ```
> 
> and see if it works?

Well I did went another way to make those symlinks. Now it started to compile again.
But its still fails because it still  complains about

```cmake
/lib/ld-linux-armhf.so.3: No such file or directory
Exiting!
```

rafal1137, 2021-04-12:
@abhiTronix ^
Just to ping you.

abhiTronix, 2021-04-12:
@rafal1137 Seems like your sysroot missing some important Rpi packages. 

Answer these queries:
- What toolchain from this project are you using? Paste the downloading page link or screenshot or something else?
- Which Raspberry Pi model are you targeting in your project.
- What exact version of Raspbian sysroot are you using in your project?  (Stretch or Buster)
as they all are correlated.

Also, Did you tried any other toolchain beside this project?

rafal1137, 2021-04-12:
@abhiTronix
> @rafal1137 Seems like your sysroot missing some important Rpi packages.
>
> Answer these queries:
>
>     * What toolchain from this project are you using? Paste the downloading page link or screenshot or [...]

* I am using precompilined toolchain downloaded from links you provide ```cross-gcc-10.2.0-pi_2-3.tar.gz```
  Also link:
  https://deac-fra.dl.sourceforge.net/project/raspberry-pi-cross-compilers/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%2010.2.0/Raspberry%20Pi%202%2C%203/cross-gcc-10.2.0-pi_2-3.tar.gz
  
* I am targeting RPI 2/3 armv7 arch compatible.
* Sysroot is from Buster
* Yes I did tried some old 8.3 Toolchain but that one had issue with ```limits.h``` header file.

rafal1137, 2021-04-12:
@abhiTronix Also I did just ls that file. And it shows there with symlinking.

```bash
ls -l  /opt/rootfs/lib/ld*
lrwxrwxrwx 1 root root 30 May 14  2019 /opt/rootfs/lib/ld-linux-armhf.so.3 -> arm-linux-gnueabihf/ld-2.28.so
lrwxrwxrwx 1 root root 24 Mar  4 22:48 /opt/rootfs/lib/ld-linux.so.3 -> /lib/ld-linux-armhf.so.3
```

abhiTronix, 2021-04-13:
@rafal1137 Can you try compiling bare-minimum C++ Program with `--sysroot` flag as follows:

```c
echo 'int main() { return 0; }' > foo.cpp
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++ \
    --sysroot=/opt/rootfs/ \
    foo.cpp
```

to ensure sysroot is working fine.

rafal1137, 2021-04-13:
@abhiTronix Just did it and this what happened
```
root@720bc2f03c8a:/# arm-linux-gnueabihf-g++ --sysroot=/opt/rootfs/ foo.cpp
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status
```

abhiTronix, 2021-04-13:
@rafal1137 Real problem is still hidden. Please, try with verbose parameter and link directories directly as follows:

```sh
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++  --sysroot=/opt/rootfs -L/opt/rootfs/usr/lib/arm-linux-gnueabihf -I/opt/rootfst/usr/include/arm-linux-gnueabihf foo.cpp  -v
```

and paste the complete output here.

rafal1137, 2021-04-13:
@abhiTronix This is entire output:
```
root@720bc2f03c8a:/# /opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++  --sysroot=/opt/rootfs -L/opt/rootfs/usr/lib/arm-linux-gnueabihf -I/opt/rootfs/usr/include/arm-linux-gnueabihf foo.cpp  -v
Using built-in specs.
COLLECT_GCC=/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++
COLLECT_LTO_WRAPPER=/opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/tmp/cross-pi-gcc-10.2.0-1/arm-linux-gnueabihf/libc --with-arch=armv7-a --with-fpu=neon-vfpv4 --with-float=hard --disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-L/opt/rootfs/usr/lib/arm-linux-gnueabihf' '-I' '/opt/rootfs/usr/include/arm-linux-gnueabihf' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/cc1plus -quiet -v -I /opt/rootfs/usr/include/arm-linux-gnueabihf -iprefix /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/ -isysroot /opt/rootfs -D_GNU_SOURCE foo.cpp -quiet -dumpbase foo.cpp -mfloat-abi=hard -mfpu=neon-vfpv4 -mtls-dialect=gnu -marm -march=armv7-a+neon-vfpv4 -auxbase foo -version -o /tmp/ccUK1rL0.s
GNU C++14 (GCC) version 10.2.0 (arm-linux-gnueabihf)
        compiled by GNU C version 7.5.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/rootfs/usr/include/arm-linux-gnueabihf
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/include
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include
 /opt/rootfs/usr/local/include
 /opt/rootfs/usr/include
End of search list.
GNU C++14 (GCC) version 10.2.0 (arm-linux-gnueabihf)
        compiled by GNU C version 7.5.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: fe3d46b46971cd3b57cc2f549fffafbf
COLLECT_GCC_OPTIONS='-L/opt/rootfs/usr/lib/arm-linux-gnueabihf' '-I' '/opt/rootfs/usr/include/arm-linux-gnueabihf' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/as -v -I /opt/rootfs/usr/include/arm-linux-gnueabihf -march=armv7-a -mfloat-abi=hard -mfpu=neon-vfpv4 -meabi=5 -o /tmp/ccP84BHf.o /tmp/ccUK1rL0.s
GNU assembler version 2.31 (arm-linux-gnueabihf) using BFD version (GNU Binutils) 2.31
COMPILER_PATH=/opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/:/opt/cross-pi-gcc/bin/../libexec/gcc/:/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/:/opt/cross-pi-gcc/bin/../lib/gcc/:/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/lib/:/opt/rootfs/lib/:/opt/rootfs/usr/lib/
COLLECT_GCC_OPTIONS='-L/opt/rootfs/usr/lib/arm-linux-gnueabihf' '-I' '/opt/rootfs/usr/include/arm-linux-gnueabihf' '-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/collect2 -plugin /opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/liblto_plugin.so -plugin-opt=/opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccqe0NEu.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --sysroot=/opt/rootfs --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X -m armelf_linux_eabi /opt [... cut ...]
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/lib/libstdc++.so: undefined reference to `exp@GLIBC_2.4'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/lib/libstdc++.so: undefined reference to `cosh@GLIBC_2.4'
collect2: error: ld returned 1 exit status
```
Wait what it can't find GLIBC_2.4 ?

abhiTronix, 2021-04-13:
@rafal1137 IMO we are close to solution as `-lm` is set correctly this time. So `--sysroot=/opt/rootfs` will set correct sysroot but `--sysroot=/opt/rootfs/` doesn't. Can you iterate above command by eliminating all linked directories:

```sh
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++  --sysroot=/opt/rootfs  -v
```

Thank you for helping in debugging this issue.

rafal1137, 2021-04-13:
@abhiTronix
```bash
root@720bc2f03c8a:/# /opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++  --sysroot=/opt/rootfs foo.cpp -v
Using built-in specs.
COLLECT_GCC=/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++
COLLECT_LTO_WRAPPER=/opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/tmp/cross-pi-gcc-10.2.0-1/arm-linux-gnueabihf/libc --with-arch=armv7-a --with-fpu=neon-vfpv4 --with-float=hard --disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) 
COLLECT_GCC_OPTIONS='-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/cc1plus -quiet -v -iprefix /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/ -isysroot /opt/rootfs -D_GNU_SOURCE foo.cpp -quiet -dumpbase foo.cpp -mfloat-abi=hard -mfpu=neon-vfpv4 -mtls-dialect=gnu -marm -march=armv7-a+neon-vfpv4 -auxbase foo -version -o /tmp/ccxFnr5P.s
GNU C++14 (GCC) version 10.2.0 (arm-linux-gnueabihf)
        compiled by GNU C version 7.5.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed"
ignoring duplicate directory "/opt/cross-pi-gcc/bin/../lib/gcc/../../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/arm-linux-gnueabihf
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include/c++/10.2.0/backward
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/include
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/include-fixed
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/include
 /opt/rootfs/usr/local/include
 /opt/rootfs/usr/include
End of search list.
GNU C++14 (GCC) version 10.2.0 (arm-linux-gnueabihf)
        compiled by GNU C version 7.5.0, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: fe3d46b46971cd3b57cc2f549fffafbf
COLLECT_GCC_OPTIONS='-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/as -v -march=armv7-a -mfloat-abi=hard -mfpu=neon-vfpv4 -meabi=5 -o /tmp/cclKUC93.o /tmp/ccxFnr5P.s
GNU assembler version 2.31 (arm-linux-gnueabihf) using BFD version (GNU Binutils) 2.31
COMPILER_PATH=/opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/:/opt/cross-pi-gcc/bin/../libexec/gcc/:/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/
LIBRARY_PATH=/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/:/opt/cross-pi-gcc/bin/../lib/gcc/:/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/lib/:/opt/rootfs/lib/:/opt/rootfs/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-shared-libgcc'  '-mfloat-abi=hard' '-mfpu=neon-vfpv4' '-mtls-dialect=gnu' '-marm' '-march=armv7-a+neon-vfpv4'
 /opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/collect2 -plugin /opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/liblto_plugin.so -plugin-opt=/opt/cross-pi-gcc/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/lto-wrapper -plugin-opt=-fresolution=/tmp/cchwMofi.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --sysroot=/opt/rootfs --eh-frame-hdr -dynamic-linker /lib/ld-linux-armhf.so.3 -X -m armelf_linux_eabi /opt/rootfs/lib/crt1.o /opt/rootfs/lib/crti.o /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/crtbegin.o -L/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0 -L/opt/cross-pi-gcc/bin/../lib/gcc -L/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/lib -L/opt/rootfs/lib -L/opt/rootfs/usr/lib /tmp/cclKUC93.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/crtend.o /opt/rootfs/lib/crtn.o
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lm
collect2: error: ld returned 1 exit status
```

abhiTronix, 2021-04-13:
@rafal1137 again. Sorry for the inconvenience. Can you share how you setup sysroot, I'll try everything locally and share the final result here by tomorrow.

rafal1137, 2021-04-13:
> @rafal1137 again. Sorry for the inconvenience. Can you share how you setup sysroot, I'll try everything locally and share the final result here by tomorrow. OR if possible upload and share sysroot through mega or gdrive.

I will share the sysroot setup here since uploading entire rootfs could take too long since my upload is not good for such things.
I am using this repo https://github.com/kclyu/rpi_rootfs with some changes on my side.

Steps I did inside a rpi_rootfs repo:
* inside scripts dir there is ```update_upgrade_install_package.sh``` which I have eddited
```bash
#!/bin/bash
apt update
apt full-upgrade -y
apt -y install build-essential libfreeimage-dev libopenal-dev libpango1.0-dev libsndfile-dev libudev-dev \
libasound2-dev libjpeg8-dev libwebp-dev automake libgl1-mesa-glx libjpeg62-turbo libogg0 libopenal1 libvorbis0a \
libvorbisfile3 zlib1g libraspberrypi0 libraspberrypi-bin libraspberrypi-dev libx11-dev libglew-dev libegl1-mesa-dev \
nasm autoconf git cmake zip gcc g++ libtool libxrandr-dev x11proto-randr-dev libgles-dev
```
* I did runned command:
```bash
./build_rootfs.sh create ./2021-03-04-raspios-buster-armhf-lite.img
```
Note: I did downloaded earlier raspbian os lite version, then I did extracted it.
* After when it was done there was rootfs dir with all files. Copied it over to place where after I did created docker image.

```docker
FROM debian:10

# Install some dependencies
RUN apt update && apt -y install nano git cmake autoconf libtool nasm zip

# Add RasbianOS RootFS and Cross Compilers directories to the image
ADD ./rootfs /opt/rootfs
ADD ./cross-pi-gcc /opt/cross-pi-gcc

# Add extra Env Variables
ENV PATH=$PATH:/opt/cross-pi-gcc/bin \
    RASPBIAN_ROOTFS=/opt/rootfs \
    CC=arm-linux-gnueabihf-gcc \
    CXX=arm-linux-gnueabihf-g++ \
    NM=arm-linux-gnueabihf-nm \
    LD=arm-linux-gnueabihf-ld \
    RANLIB=arm-linux-gnueabihf-ranlib \
    AR=arm-linux-gnueabihf-ar
```
* I am not counting here the symlinks that you requested here.

rafal1137, 2021-04-13:
@abhiTronix ^

abhiTronix, 2021-04-14:
@rafal1137 I made some changes to https://github.com/kclyu/rpi_rootfs sysroot and bare minimum example is seems to be working ok, but the real problem with sysroot within cmake is still as it is. I'll run more test and to find the root cause, will update you when it is ready. IMO, the toolchain itself might need some rework. In the mean time you can try [linaro ARM toolchains](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-a/downloads) instead, give them a try.

closed 2021-04-14 by abhiTronix