abhiTronix/raspberry-pi-cross-compilers #72: libgomp.so.1, needed by ../src/libdoublefann.so.2.2.0, not found

distlibs, 2021-05-02:
Hi. I trying to build [libfann/fann](https://github.com/libfann/fann) on Ubuntu machine for Raspberry Pi 4. I get error:
 `libgomp.so.1, needed by ../src/libdoublefann.so.2.2.0, not found`

Information:
Host OS: Ubuntu 20.04
Target OS: Buster 32-bit OS
Toolchain: Raspberry Pi GCC Cross-Compiler Toolchains/Buster/GCC 8.3.0/Raspberry Pi 3A+, 3B+, 4/cross-gcc-8.3.0-pi_3+.tar.gz

```
Scanning dependencies of target floatfann
Scanning dependencies of target doublefann
Scanning dependencies of target floatfann_static
Scanning dependencies of target fixedfann_static
Scanning dependencies of target doublefann_static
Scanning dependencies of target fixedfann
Scanning dependencies of target fann
Scanning dependencies of target fann_static
[  3%] Building C object src/CMakeFiles/doublefann.dir/doublefann.c.o
[  6%] Building C object src/CMakeFiles/fixedfann_static.dir/fixedfann.c.o
[ 17%] Building C object src/CMakeFiles/fann.dir/floatfann.c.o
[ 17%] Building C object src/CMakeFiles/floatfann_static.dir/floatfann.c.o
[ 17%] Building C object src/CMakeFiles/doublefann_static.dir/doublefann.c.o
[ 20%] Building C object src/CMakeFiles/fixedfann.dir/fixedfann.c.o
[ 24%] Building C object src/CMakeFiles/floatfann.dir/floatfann.c.o
[ 27%] Building C object src/CMakeFiles/fann_static.dir/floatfann.c.o
[ 31%] Linking C shared library libfixedfann.so
[ 31%] Built target fixedfann
Scanning dependencies of target gtest
[ 34%] Building CXX object lib/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 37%] Linking C static library libfixedfann.a
[ 37%] Built target fixedfann_static
[ 41%] Linking C shared library libfloatfann.so
[ 44%] Linking C shared library libdoublefann.so
[ 44%] Built target floatfann
[ 48%] Linking C shared library libfann.so
[ 48%] Built target doublefann
[ 48%] Built target fann
[ 51%] Linking C static library libfloatfann.a
[ 55%] Linking C static library libfann.a
[ 58%] Linking C static library libdoublefann.a
[ 58%] Built target floatfann_static
[ 58%] Built target fann_static
[ 58%] Built target doublefann_static
[ 62%] Linking CXX shared library ../../libgtest.so
[ 62%] Built target gtest
Scanning dependencies of target gmock
Scanning dependencies of target gtest_main
Scanning dependencies of target fann_tests
[ 65%] Building CXX object lib/googletest/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 68%] Building CXX object lib/googletest/googlemock/CMakeFiles/gmock.dir/src/gmock-all.cc.o
[ 72%] Building CXX object tests/CMakeFiles/fann_tests.dir/fann_test.cpp.o
[ 79%] Building CXX object tests/CMakeFiles/fann_tests.dir/main.cpp.o
[ 79%] Building CXX object tests/CMakeFiles/fann_tests.dir/fann_test_data.cpp.o
[ 82%] Building CXX object tests/CMakeFiles/fann_tests.dir/fann_test_train.cpp.o
[ 86%] Linking CXX shared library ../../libgtest_main.so
[ 86%] Built target gtest_main
[ 89%] Linking CXX executable fann_tests
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: warning: libgomp.so.1, needed by ../src/libdoublefann.so.2.2.0, not found (try using -rpath or -rpath-link)
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: ../src/libdoublefann.so.2.2.0: undefined reference to `GOMP_barrier@GOMP_1.0'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: ../src/libdoublefann.so.2.2.0: undefined reference to `GOMP_parallel@GOMP_4.0'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: ../src/libdoublefann.so.2.2.0: undefined reference to `omp_set_num_threads@OMP_1.0'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: ../src/libdoublefann.so.2.2.0: undefined reference to `omp_get_thread_num@OMP_1.0'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: ../src/libdoublefann.so.2.2.0: undefined reference to `omp_set_dynamic@OMP_1.0'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: ../src/libdoublefann.so.2.2.0: undefined reference to `omp_get_num_threads@OMP_1.0'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/fann_tests.dir/build.make:131: tests/fann_tests] Error 1
make[1]: *** [CMakeFiles/Makefile2:538: tests/CMakeFiles/fann_tests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Linking CXX shared library ../../libgmock.so
[ 93%] Built target gmock
make: *** [Makefile:152: all] Error 2
```

abhiTronix, 2021-05-02:
@distlibs Kindly paste your `cmake` file and also `CMakeError.log` file from build folder. Also what instructions are you using?

abhiTronix, 2021-05-03:
@distlibs Please provide requested information in order to debug this issue.

closed 2021-05-03 by abhiTronix

distlibs, 2021-05-03:
Hi, @abhiTronix

pi.cmake:
```
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)
set(CMAKE_C_COMPILER /opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc)
set(CMAKE_CXX_COMPILER /opt/cross-pi-gcc/bin/arm-linux-gnueabihf-g++)
```
CMakeError.log:

```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/test/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_d5b7c/fast && /usr/bin/make -f CMakeFiles/cmTC_d5b7c.dir/build.make CMakeFiles/cmTC_d5b7c.dir/build
make[1]: Entering directory '/home/test/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_d5b7c.dir/src.c.o
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_d5b7c.dir/src.c.o   -c /home/test/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_d5b7c
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d5b7c.dir/link.txt --verbose=1
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_d5b7c.dir/src.c.o  -o cmTC_d5b7c
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: CMakeFiles/cmTC_d5b7c.dir/src.c.o: in function `main':
src.c:(.text+0x44): undefined reference to `pthread_create'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: src.c:(.text+0x50): undefined reference to `pthread_detach'
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: src.c:(.text+0x60): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_d5b7c.dir/build.make:87: cmTC_d5b7c] Error 1
make[1]: Leaving directory '/home/test/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_d5b7c/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/test/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_bc1a1/fast && /usr/bin/make -f CMakeFiles/cmTC_bc1a1.dir/build.make CMakeFiles/cmTC_bc1a1.dir/build
make[1]: Entering directory '/home/test/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_bc1a1.dir/CheckFunctionExists.c.o
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_bc1a1.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_bc1a1
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_bc1a1.dir/link.txt --verbose=1
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_bc1a1.dir/CheckFunctionExists.c.o  -o cmTC_bc1a1  -lpthreads
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_bc1a1.dir/build.make:87: cmTC_bc1a1] Error 1
make[1]: Leaving directory '/home/test/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_bc1a1/fast] Error 2
```

Instructions:
```
mkdir build
cd build
cmake -D CMAKE_TOOLCHAIN_FILE=pi.cmake ../fann
make
```

abhiTronix, 2021-05-03:
> pi.cmake:
>
> set(CMAKE_SYSTEM_NAME Linux)
> set(CMAKE_SYSTEM_PROCESSOR arm)
> set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
> set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
> set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE [...]

@distlibs Errors indicate you need to provide sysroot files to cross-compile it properly. Please follow according to instructions given in this wiki page: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-Raspbian-32-bit-Image#cross-compiler-cmake-usage-guide-with-raspbian-32-bit-image

distlibs, 2021-05-03:
I followed instructions to build Sysroot. I copied your PI.cmake. I changed tools path to:
`set(tools /opt/cross-pi-gcc)`

Now I got this error:
```
Determining if the C compiler works failed with the following output:
Change Dir: /home/test/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_1c408/fast && /usr/bin/make -f CMakeFiles/cmTC_1c408.dir/build.make CMakeFiles/cmTC_1c408.dir/build
make[1]: Entering directory '/home/test/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1c408.dir/testCCompiler.c.o
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc --sysroot="~/cmake-test/rpi_rootfs/rootfs"   -fPIC -Wl,-rpath-link,~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -fPIC -Wl,-rpath-link,~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf    -o CMakeFiles/cmTC_1c408.dir/testCCompiler.c.o   -c /home/test/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_1c408
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1c408.dir/link.txt --verbose=1
/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-gcc --sysroot="~/cmake-test/rpi_rootfs/rootfs"  -fPIC -Wl,-rpath-link,~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -fPIC -Wl,-rpath-link,~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf    -fPIC -Wl,-rpath-link,~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf -L~/cmake-test/rpi_rootfs/rootfs/usr/lib/arm-linux-gnueabihf  -rdynamic CMakeFiles/cmTC_1c408.dir/testCCompiler.c.o  -o cmTC_1c408
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: No such file or directory
/opt/cross-pi-gcc/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: No such file or directory
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_1c408.dir/build.make:87: cmTC_1c408] Error 1
make[1]: Leaving directory '/home/test/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_1c408/fast] Error 2
```

abhiTronix, 2021-05-03:
Please wait till I test this locally.

abhiTronix, 2021-05-03:
@distlibs Not reproduceable:

Output of `cmake`:
```sh
absys@Test-Notebook:~/Documents/rpi_rootfs$ git clone https://github.com/libfann/fann.git
Cloning into 'fann'...
remote: Enumerating objects: 2746, done.
remote: Counting objects: 100% (32/32), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 2746 (delta 6), reused 8 (delta 1), pack-reused 2714
Receiving objects: 100% (2746/2746), 10.24 MiB | 12.09 MiB/s, done.
Resolving deltas: 100% (1602/1602), done.
absys@Test-Notebook:~/Documents/rpi_rootfs$ cd ./fann
absys@Test-Notebook:~/Documents/rpi_rootfs/fann$ mkdir build
absys@Test-Notebook:~/Documents/rpi_rootfs/fann$ cd build
absys@Test-Notebook:~/Documents/rpi_rootfs/fann/build$ cmake -D CMAKE_TOOLCHAIN_FILE=/home/absys/Documents/rpi_rootfs/PI.cmake ..
CMake Deprecation Warning at CMakeLists.txt:41 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/absys/rpi_rootfs/tools/cross-pi-gcc-10.2.0-1/bin/arm-linux-gnueabihf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/absys/rpi_rootfs/tools/cross-pi-gcc-10.2.0-1/bin/arm-linux-gnueabihf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- FANN is used as APPLICATION_NAME
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
CMake Deprecation Warning at lib/googletest/CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at lib/googletest/googlemock/CMakeLists.txt:45 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at lib/googletest/googletest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
[... 34 lines cut ...]
[ 48%] Building CXX object lib/googletest/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 51%] Linking C static library libdoublefann.a
[ 51%] Built target doublefann_static
[ 55%] Linking C shared library libfann.so
[ 55%] Built target fann
[ 58%] Linking C static library libfann.a
[ 58%] Built target fann_static
[ 62%] Linking CXX shared library ../../libgtest.so
[ 62%] Built target gtest
[ 65%] Building CXX object lib/googletest/googletest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 68%] Building CXX object tests/CMakeFiles/fann_tests.dir/fann_test.cpp.o
[ 75%] Building CXX object lib/googletest/googlemock/CMakeFiles/gmock.dir/src/gmock-all.cc.o
[ 75%] Building CXX object tests/CMakeFiles/fann_tests.dir/main.cpp.o
[ 79%] Linking CXX shared library ../../libgtest_main.so
[ 82%] Building CXX object tests/CMakeFiles/fann_tests.dir/fann_test_data.cpp.o
[ 82%] Built target gtest_main
[ 86%] Building CXX object tests/CMakeFiles/fann_tests.dir/fann_test_train.cpp.o
[ 89%] Linking CXX executable fann_tests
[ 89%] Built target fann_tests
[ 93%] Linking CXX shared library ../../libgmock.so
[ 93%] Built target gmock
[ 96%] Building CXX object lib/googletest/googlemock/CMakeFiles/gmock_main.dir/src/gmock_main.cc.o
[100%] Linking CXX shared library ../../libgmock_main.so
[100%] Built target gmock_main
```

abhiTronix, 2021-05-03:
@distlibs I just added new commit is made to rpi_rootfs that fixed some critical bugs: https://github.com/abhiTronix/rpi_rootfs/commit/df5f3b7d663128fd028ad4e5a3865ef79ea45d69 ,thereby I'll advise to re-clone it and rebuild sysroot. And your problem will be solved(seem output above), also clean build folder before trying again.

abhiTronix, 2021-05-03:
@distlibs Also I used GCC `10.2.0` but I don't think that will affect build.