abhiTronix/raspberry-pi-cross-compilers #74: "-sysroot" problem

sbela, 2021-07-21:
Following these instructions I have compile problem with missing libraries like libm etc. which is in the sysroot for sure!
These links:
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/asm /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/gnu /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/bits /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/sys /usr/include
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/openssl /usr/include

maybe, should be like :
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/asm /usr/include/**asm**
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/gnu /usr/include/**gnu**
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/bits /usr/include/**bits**
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/sys /usr/include/**sys**
sudo ln -sf -r /usr/include/arm-linux-gnueabihf/openssl /usr/include/**openssl**

Am I wrong?!
Compiling still fails with missing libraries anyway...
Thank you!

abhiTronix, 2021-07-22:
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/asm /usr/include
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/gnu /usr/include
> sudo ln -sf -r /usr/include/arm-linux-gnueabihf/bits /usr/include [...]

@sbela Actually sometimes first one works and other times second one. I've not able to find a reliable way to create symlinks. I'm thinking of switching to python script in future.

> Compiling still fails with missing libraries anyway...

What's the error, kindly share the log? Did you checked each symlink manually? Also what're you trying to compile?

abhiTronix, 2021-07-22:
@sbela Try these symlinks instead:

```sh
 sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/gnu/* /usr/include/gnu/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/bits/* /usr/include/bits/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/sys/* /usr/include/sys/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl/
 sudo ln -sf /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
 sudo ln -sf /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
 sudo ln -sf /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
```

sbela, 2021-07-22:
> @sbela Try these symlinks instead:
> 
> ```shell
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm/
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/gnu/* /usr/include/gnu/
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/bits/* /usr/include/bits/
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/sys/* /usr/include/sys/
>  sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl/
>  sudo ln -sf /usr/lib/arm-linux-gnueabihf/crtn.o /usr/lib/crtn.o
>  sudo ln -sf /usr/lib/arm-linux-gnueabihf/crt1.o /usr/lib/crt1.o
>  sudo ln -sf /usr/lib/arm-linux-gnueabihf/crti.o /usr/lib/crti.o
> ```

Thanks that worked! And after that I had to run the rsync commands... and I misspelled the -I$HOME.. and -L$HOME.. path too so I made triple mistake :) ... my bad!

closed 2021-07-22 by sbela

abhiTronix, 2021-07-22:
@sbela Glad it worked 👍🏽