abhiTronix/raspberry-pi-cross-compilers #75: Compiler cross-pi-gcc-8.3.0-0 is broken

ValeriiKim, 2021-08-10:
Good day, sir
I have followed your instruction "Cross-Compiler CMake Usage Guide with rsynced Raspberry Pi 32-bit OS" (link: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os) for simple CMake project setup.
Everything was fine until I run this command for configuring my project:
`cmake -DCMAKE_TOOLCHAIN_FILE=~/cmake-test/PI.cmake  -DCMAKE_BUILD_TYPE=Debug ..`
Unfortunately I get this output from my host Ubuntu 18.04 system:
```
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0//bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0//bin/arm-linux-gnueabihf-gcc - broken
CMake Error at /opt/cmake/share/cmake-3.17/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0//bin/arm-linux-gnueabihf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/xbron/Projects/picamera/build/CMakeFiles/CMakeTmp
```
Also there is an additional info:
```
/home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: No such file or directory
    /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/../lib/gcc/arm-linux-gnueabihf/8.3.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: No such file or directory
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_0687a.dir/build.make:106: recipe for target 'cmTC_0687a' failed
    make[1]: *** [cmTC_0687a] Error 1
    make[1]: exit catalog «/home/xbron/Projects/picamera/build/CMakeFiles/CMakeTmp»
    Makefile:141: recipe for target 'cmTC_0687a/fast' failed
    make: *** [cmTC_0687a/fast] Error 2
```
I downloaded cross-compiler from here: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%208.3.0/Raspberry%20Pi%201%2C%20Zero/cross-gcc-8.3.0-pi_0-1.tar.gz/download

ValeriiKim, 2021-08-10:
For this command
`sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl/`
I get following:
`ln: target '/usr/include/openssl/' is not a directory: No such file or directory`
Also these symlinks are different compared to that posted in tutorial. But I think this time I get different output.

abhiTronix, 2021-08-10:
> For this command
> sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl/
> I get following:
> ln: target '/usr/include/openssl/' is not a directory: No such file or directory

@ValeriiKim This is not that important is you're not using openssl.

> But I think this time I get different output.

Can you paste the log? Also, what are you trying to compile, please do share if its open-sourced?

ValeriiKim, 2021-08-10:
> @ValeriiKim This is not that important is you're not using openssl.
>
>
> Can you paste the log? Also, what are you trying to compile, please do share if its open-sourced?

For example I get this log after `sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm/`

abhiTronix, 2021-08-11:
> For example I get this log after sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm/:
> 
> ln: '/usr/include/arm-linux-gnueabihf/asm/auxvec.h' and '/usr/include/asm/auxvec.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/bitsperlong.h' and '/usr/include/asm/bitsperlong.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/bpf_perf_event.h' and '/usr/include/asm/bpf_perf_event.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/byteorder.h' and '/usr/include/asm/byteorder.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/errno.h' and '/usr/include/asm/errno.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/fcntl.h' and '/usr/include/asm/fcntl.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/hwcap.h' and '/usr/include/asm/hwcap.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/ioctl.h' and '/usr/include/asm/ioctl.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/ioctls.h' and '/usr/include/asm/ioctls.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/ipcbuf.h' and '/usr/include/asm/ipcbuf.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/kvm_para.h' and '/usr/include/asm/kvm_para.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/mman.h' and '/usr/include/asm/mman.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/msgbuf.h' and '/usr/include/asm/msgbuf.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/param.h' and '/usr/include/asm/param.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/perf_regs.h' and '/usr/include/asm/perf_regs.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/poll.h' and '/usr/include/asm/poll.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/posix_types.h' and '/usr/include/asm/posix_types.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/ptrace.h' and '/usr/include/asm/ptrace.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/resource.h' and '/usr/include/asm/resource.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/sembuf.h' and '/usr/include/asm/sembuf.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/setup.h' and '/usr/include/asm/setup.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/shmbuf.h' and '/usr/include/asm/shmbuf.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/sigcontext.h' and '/usr/include/asm/sigcontext.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/siginfo.h' and '/usr/include/asm/siginfo.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/signal.h' and '/usr/include/asm/signal.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/socket.h' and '/usr/include/asm/socket.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/sockios.h' and '/usr/include/asm/sockios.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/statfs.h' and '/usr/include/asm/statfs.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/stat.h' and '/usr/include/asm/stat.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/swab.h' and '/usr/include/asm/swab.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/termbits.h' and '/usr/include/asm/termbits.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/termios.h' and '/usr/include/asm/termios.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/types.h' and '/usr/include/asm/types.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/unistd-common.h' and '/usr/include/asm/unistd-common.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/unistd-eabi.h' and '/usr/include/asm/unistd-eabi.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/unistd.h' and '/usr/include/asm/unistd.h' are the same file
> ln: '/usr/include/arm-linux-gnueabihf/asm/unistd-oabi.h' and '/usr/include/asm/unistd-oabi.h' are the same file

@ValeriiKim Did it affected the cross-compiling or you getting same error? **Can you generate `output.txt` file using following command and then share it here:**
```sh
tree /home/xbron/Projects/picamera/ > output.txt
```

> I want to get photos from a raspberry pi zero using another device, recently I worked directly on the raspberry, but now I want to switch to cross-compiling. 

Photos? Is it a software? Please link the project.

> Another problem is to include raspicam API in this setup.

Are you getting same error with cross-compiling `raspicam`?

ValeriiKim, 2021-08-11:
Currently I don't have repository with my project because there is much work to do (at first I want to continue my work in the user-friendly environment...)
I think now I got it to work: after the command
`cmake -DCMAKE_TOOLCHAIN_FILE=~/Projects/picamera/PI.cmake -DCMAKE_BUILD_TYPE=Debug ..`
I get:
```
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xbron/Projects/picamera/build
```
And after `make`:
```
/opt/cmake/bin/cmake -S/home/xbron/Projects/picamera -B/home/xbron/Projects/picamera/build --check-build-system CMakeFiles/Makefile.cmake 0
/opt/cmake/bin/cmake -E cmake_progress_start /home/xbron/Projects/picamera/build/CMakeFiles /home/xbron/Projects/picamera/build/CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: вход в каталог «/home/xbron/Projects/picamera/build»
make  -f CMakeFiles/picamera.dir/build.make CMakeFiles/picamera.dir/depend
make[2]: вход в каталог «/home/xbron/Projects/picamera/build»
cd /home/xbron/Projects/picamera/build && /opt/cmake/bin/cmake -E cmake_depends "Unix Makefiles" /home/xbron/Projects/picamera /home/xbron/Projects/picamera /home/xbron/Projects/picamera/build /home/xbron/Projects/picamera/build /home/xbron/Projects/picamera/build/CMakeFiles/picamera.dir/DependInfo.cmake --color=
Scanning dependencies of target picamera
make[2]: выход из каталога «/home/xbron/Projects/picamera/build»
make  -f CMakeFiles/picamera.dir/build.make CMakeFiles/picamera.dir/build
make[2]: вход в каталог «/home/xbron/Projects/picamera/build»
[ 50%] Building CXX object CMakeFiles/picamera.dir/main.cpp.o
/home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ --sysroot=/home/xbron/Projects/picamera/rootfs    -fPIC -Wl,-rpath-link,/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -L/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -fPIC -Wl,-rpath-link,/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -L/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -g   -std=gnu++17 -o CMakeFiles/picamera.dir/main.cpp.o -c /home/xbron/Projects/picamera/main.cpp
[100%] Linking CXX executable picamera
/opt/cmake/bin/cmake -E cmake_link_script CMakeFiles/picamera.dir/link.txt --verbose=1
/home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-g++ --sysroot=/home/xbron/Projects/picamera/rootfs   -fPIC -Wl,-rpath-link,/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -L/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -fPIC -Wl,-rpath-link,/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -L/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -g   -fPIC -Wl,-rpath-link,/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -L/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -fPIC -Wl,-rpath-link,/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf -L/home/xbron/Projects/picamera/rootfs/usr/lib/arm-linux-gnueabihf CMakeFiles/picamera.dir/main.cpp.o  -o picamera 
make[2]: выход из каталога «/home/xbron/Projects/picamera/build»
[100%] Built target picamera
make[1]: выход из каталога «/home/xbron/Projects/picamera/build»
/opt/cmake/bin/cmake -E cmake_progress_start /home/xbron/Projects/picamera/build/CMakeFiles 0
```
But the strange thing is that in `main.cpp` VScode shows me that `iostream` is located in my host Ubuntu `usr` path (but as I know there must be path from cross-compiler). Also I get this error:
`no operator "<<" matches these operands -- operand types are: std::ostream << const char [25]`

abhiTronix, 2021-08-12:
> Also I get this error:
> no operator "<<" matches these operands -- operand types are: std::ostream << const char [25]

@ValeriiKim This is because your code have problems. https://stackoverflow.com/questions/56497296/no-operator-matches-these-operands-operand-types-are-stdistream-const-cha

closed 2021-08-12 by abhiTronix

ValeriiKim, 2021-08-12:
> > Also I get this error:
> > no operator "<<" matches these operands -- operand types are: std::ostream << const char [25]
> 
> @ValeriiKim This is because your code have problems. https://stackoverflow.com/questions/56497296/no-operator-matches-these-operands-operand-types-are-stdistream-const-cha

I am sorry but my code is just the **copy of yours** (https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os). This is my `main.cpp`:
```cpp
#include <iostream>

int main() {
  // if block with init-statement:
  if (int a = 5; a < 8) {
    std::cout << "Local variable a is < 8\n";
  } else {
    std::cout << "Local variable a is >= 8\n";
  }
  return 0;
}
```
And here is my `CMakeLists.txt` file:
`
cmake_minimum_required(VERSION 3.10)
project(cmake_test)
set(CMAKE_VERBOSE_MAKEFILE ON)
set(CMAKE_CXX_STANDARD 17)
add_executable(cmake_test foo.cpp)
```

abhiTronix, 2021-08-12:
> I am sorry but my code is just the copy of yours (https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os). This is my main.cpp:
> 
> #include <iostream>
> 
> int main() {
>   // if block with init-statement:
>   if (int a = 5; a < 8) {
>     std::cout << "Local variable a is < 8\n";
>   } else {
>     std::cout << "Local variable a is >= 8\n";
>   }

@ValeriiKim What's your PI.cmake?

ValeriiKim, 2021-08-12:
Here it is (like an example):
```
set(CMAKE_VERBOSE_MAKEFILE ON)
set(CMAKE_SYSTEM_NAME Linux)
set(CMAKE_SYSTEM_VERSION 1)
set(CMAKE_SYSTEM_PROCESSOR arm)

set(tools /home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0) # warning change toolchain path here. Alternative: /opt/cross-pi-gcc
set(rootfs_dir /home/xbron/Projects/picamera/rootfs)

set(CMAKE_FIND_ROOT_PATH ${rootfs_dir})
set(CMAKE_SYSROOT ${rootfs_dir})

set(CMAKE_LIBRARY_ARCHITECTURE arm-linux-gnueabihf)
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -fPIC -Wl,-rpath-link,${CMAKE_SYSROOT}/usr/lib/${CMAKE_LIBRARY_ARCHITECTURE} -L${CMAKE_SYSROOT}/usr/lib/${CMAKE_LIBRARY_ARCHITECTURE}")
set(CMAKE_C_FLAGS "${CMAKE_CXX_FLAGS} -fPIC -Wl,-rpath-link,${CMAKE_SYSROOT}/usr/lib/${CMAKE_LIBRARY_ARCHITECTURE} -L${CMAKE_SYSROOT}/usr/lib/${CMAKE_LIBRARY_ARCHITECTURE}")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fPIC -Wl,-rpath-link,${CMAKE_SYSROOT}/usr/lib/${CMAKE_LIBRARY_ARCHITECTURE} -L${CMAKE_SYSROOT}/usr/lib/${CMAKE_LIBRARY_ARCHITECTURE}")

## Compiler Binary 
SET(BIN_PREFIX ${tools}/bin/arm-linux-gnueabihf)

SET (CMAKE_C_COMPILER ${BIN_PREFIX}-gcc)
SET (CMAKE_CXX_COMPILER ${BIN_PREFIX}-g++ )
SET (CMAKE_LINKER ${BIN_PREFIX}-ld 
            CACHE STRING "Set the cross-compiler tool LD" FORCE)
SET (CMAKE_AR ${BIN_PREFIX}-ar 
            CACHE STRING "Set the cross-compiler tool AR" FORCE)
SET (CMAKE_NM {BIN_PREFIX}-nm 
            CACHE STRING "Set the cross-compiler tool NM" FORCE)
SET (CMAKE_OBJCOPY ${BIN_PREFIX}-objcopy 
            CACHE STRING "Set the cross-compiler tool OBJCOPY" FORCE)
SET (CMAKE_OBJDUMP ${BIN_PREFIX}-objdump 
            CACHE STRING "Set the cross-compiler tool OBJDUMP" FORCE)
SET (CMAKE_RANLIB ${BIN_PREFIX}-ranlib 
            CACHE STRING "Set the cross-compiler tool RANLIB" FORCE)
SET (CMAKE_STRIP {BIN_PREFIX}-strip 
            CACHE STRING "Set the cross-compiler tool RANLIB" FORCE)

set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
```

abhiTronix, 2021-08-12:
@ValeriiKim Try compiling this code:

```cpp
#include <iostream>
#include <string>
using namespace std;

int main() {
  // if block with init-statement:
  if (int a = 5; a < 8) {
    std::cout << "Local variable a is < 8\n";
  } else {
    std::cout << "Local variable a is >= 8\n";
  }
  return 0;
}
```

abhiTronix, 2021-08-12:
Also, what's output of this command:

```sh
/home/xbron/Projects/picamera/tools/cross-pi-gcc-8.3.0-0/bin/arm-linux-gnueabihf-gcc -v --help 2> /dev/null | sed -n '/^ *-std=\([^<][^ ]\+\).*/ {s//\1/p}'
```

ValeriiKim, 2021-08-12:
Here is the output:
```
f2003
f2008
f2008ts
f2018
f95
gnu
legacy
c++03
c++0x
c++11
c++14
c++17
c++1y
c++1z
c++2a
c++98
c11
c17
c18
c1x
c89
c90
c99
c9x
gnu++03
gnu++0x
gnu++11
gnu++14
gnu++17
gnu++1y
gnu++1z
gnu++2a
gnu++98
gnu11
gnu17
gnu18
gnu1x
gnu89
gnu90
gnu99
gnu9x
iso9899:1990
iso9899:199409
iso9899:1999
iso9899:199x
iso9899:2011
iso9899:2017
iso9899:2018
```
Now it's complaining about this: 
`no operator "<<" matches these operands -- operand types are: std::ostream << const char [26]`

abhiTronix, 2021-08-13:
@ValeriiKim That's weird. I'll retest this example in my free time. Can you instead try: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-Raspbian-32-bit-Image#cross-compiler-cmake-usage-guide-with-raspbian-32-bit-image

ValeriiKim, 2021-08-13:
Ok, thank you! I should say that I tried **one more thing**. I followed your instructions (with an adjustment from your first answer to my question) to install cross-compilation environment, but in the end I tried _another_ cross-compiler and toolchain file (you can find it here: https://github.com/Pro/raspi-toolchain). After some fuss (deleting the `build` directory, cleaning, etc.) my project started to work (CMake build produces no problem). And if for example I click "Go to definition" for `#include <iostream>` VScode redirects me to my cross-compiler directory with this include file. 
But it wouldn't have worked correctly without your detailed instructions. Rather, it worked, but there were problems with connecting third-party libraries (raspicam).