abhiTronix/raspberry-pi-cross-compilers #77: Qt 5.15.2 configuration for raspberry 3B

ValentiWorkLearning, 2021-08-27:
I've decided to separate the issue from the previous one.

I guess, it hasn't been covered in the Qt building guide, but the default configuration for the `linux-rasp-pi-3-g++` produces the following result:
```
Build options:
  Mode ................................... release
  Optimize release build for size ........ no
  Building shared libraries .............. yes
  Using C standard ....................... C11
  Using C++ standard ..................... C++17
  Using ccache ........................... no
  Using new DTAGS ........................ no
  Relocatable ............................ yes
  Using precompiled headers .............. yes
  Using LTCG ............................. no
  Target compiler supports:
    NEON ................................. yes
  Build parts ............................ libs
Qt modules and options:
  Qt Concurrent .......................... yes
  Qt D-Bus ............................... yes
  Qt D-Bus directly linked to libdbus .... yes
  Qt Gui ................................. yes
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. yes
  Qt Widgets ............................. yes
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  udev ................................... yes
  Using system zlib ...................... yes
  Zstandard support ...................... no
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ yes
  GLib ................................... yes
  iconv .................................. no
  ICU .................................... yes
  Built-in copy of the MIME database ..... yes
  Tracing backend ........................ <none>
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  PCRE2 .................................. yes
    Using system PCRE2 ................... yes
Qt Network:
  getifaddrs() ........................... yes
  IPv6 ifname ............................ yes
  libproxy ............................... no
  Linux AF_NETLINK ....................... yes
  OpenSSL ................................ yes
    Qt directly linked to OpenSSL ........ no
  OpenSSL 1.1 ............................ yes
  DTLS ................................... yes
  OCSP-stapling .......................... yes
  SCTP ................................... no
  Use system proxies ..................... yes
  GSSAPI ................................. no
Qt Gui:
  Accessibility .......................... no
  FreeType ............................... yes
    Using system FreeType ................ yes
  HarfBuzz ............................... yes
    Using system HarfBuzz ................ yes
  Fontconfig ............................. yes
  Image formats:
    GIF .................................. yes
    ICO .................................. yes
[... 156 lines cut ...]
  DirectShow ............................. no
  Windows Media Foundation ............... no
Qt TextToSpeech:
  Flite .................................. no
  Flite with ALSA ........................ no
  Speech Dispatcher ...................... no
Qt Tools:
  Qt Assistant ........................... yes
  Qt Designer ............................ yes
  Qt Distance Field Generator ............ yes
  kmap2qmap .............................. yes
  Qt Linguist ............................ yes
  Mac Deployment Tool .................... no
  makeqpf ................................ yes
  pixeltool .............................. yes
  qdbus .................................. yes
  qev .................................... yes
  Qt Attributions Scanner ................ yes
  qtdiag ................................. yes
  qtpaths ................................ yes
  qtplugininfo ........................... yes
  Windows deployment tool ................ no
  WinRT Runner Tool ...................... no
Qt Tools:
  QDoc ................................... no

ValentiWorkLearning, 2021-08-27:
Configure was launched with the following options:

```shell

CROSS_COMPILER_LOCATION=$HOME/Downloads/cross-pi-gcc-8.3.0-2

../qt-everywhere-src-5.15.2/configure -device linux-rasp-pi3-g++ \
-opengl es2 -eglfs \
-opensource -confirm-license -release \
-nomake tests \
-pkg-config \
-no-use-gold-linker -no-gbm \
-skip qtwayland -skip qtlocation -skip qtwebengine -skip qtscript -no-feature-accessibility \
-device-option CROSS_COMPILE=$(echo $CROSS_COMPILER_LOCATION)/bin/arm-linux-gnueabihf- -sysroot ~/Development/rpi3_sysroot -make libs \
-prefix /usr/local/qt5.15 \
-hostprefix ~/Development/qt_5_15_2_completed_83gcc_build/tools \
-extprefix ~/Development/qt_5_15_2_completed_83gcc_build/binaries \
-v \
-recheck \
-L$HOME/Development/rpi3_sysroot/usr/lib/arm-linux-gnueabihf \
-I$HOME/Development/rpi3_sysroot/usr/include/arm-linux-gnueabihf

```

ValentiWorkLearning, 2021-08-27:
It seems, that the default configuration doesn't support hardware OpenGL. I guess, some parts of this guide should be taken into account :smile: 
https://wiki.qt.io/RaspberryPi2EGLFS#Step_by_step

abhiTronix, 2021-08-28:
> It seems, that the default configuration doesn't support hardware OpenGL. I guess, some parts of this guide should be taken 
> into account 😄
> https://wiki.qt.io/RaspberryPi2EGLFS#Step_by_step

@ValentiWorkLearning That's a well-documented guide that wasn't previously available. I guess we can copy the configurations  and properly credit the page in our guide. But due to limited time and more focus on my other projects, I'll not able make these changes as of now. However any related PRs are always welcomed.

ValentiWorkLearning, 2021-08-28:
@abhiTronix I'll try to figure out how to properly launch Opengl. I guess, there is just a few changes for RPi3

ValentiWorkLearning, 2021-08-30:
I found solution. It's necessary to specify the sysroot dir to vc libs.
```
include(../common/linux_device_pre.conf)

QMAKE_RPATHLINKDIR_POST += $$[QT_SYSROOT]/opt/vc/lib

QMAKE_LIBDIR_OPENGL_ES2 = $$[QT_SYSROOT]/opt/vc/lib
QMAKE_LIBDIR_EGL = $$QMAKE_LIBDIR_OPENGL_ES2
QMAKE_LIBDIR_OPENVG = $$QMAKE_LIBDIR_OPENGL_ES2

QMAKE_INCDIR_EGL = $$[QT_SYSROOT]/opt/vc/include \
$$[QT_SYSROOT]/opt/vc/include/interface/vcos/pthreads \
$$[QT_SYSROOT]/opt/vc/include/interface/vmcs_host/linux
QMAKE_INCDIR_OPENGL_ES2 = $${QMAKE_INCDIR_EGL}
QMAKE_INCDIR_OPENVG = $${QMAKE_INCDIR_EGL}

QMAKE_LIBS_EGL = -lEGL -lGLESv2
QMAKE_LIBS_OPENVG = -lEGL -lOpenVG -lGLESv2

QMAKE_INCDIR_BCM_HOST = $$[QT_SYSROOT]/opt/vc/include
QMAKE_LIBDIR_BCM_HOST = $$[QT_SYSROOT]/opt/vc/lib
QMAKE_LIBS_BCM_HOST = -lbcm_host

DISTRO_OPTS += deb-multi-arch
DISTRO_OPTS += hard-float

QMAKE_CFLAGS += -march=armv8-a -mtune=cortex-a53 -mfpu=crypto-neon-fp-armv8

QMAKE_CXXFLAGS = $$QMAKE_CFLAGS

EGLFS_DEVICE_INTEGRATION = eglfs_brcm

include(../common/linux_arm_device_post.conf)

load(qt_config)

```

The official qt qmake.conf is:
```
# I consider it a bug that this is required, but our EGL config.test _requires_ it
QMAKE_RPATHLINKDIR_POST += $$[QT_SYSROOT]/opt/vc/lib

VC_LIBRARY_PATH         = /opt/vc/lib
VC_INCLUDE_PATH         = =/opt/vc/include
```
which leads to the broken EGL drivers configuration

### Note. Everything above is related to the `linux-rasp-pi3-g++`