abhiTronix/raspberry-pi-cross-compilers #78: Errors when trying to build Hello World project

jteng7, 2021-09-08:
I'm seeing the following errors below when I try to build the Hello World project from here: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os

Do you know what could be wrong?

> test@ubuntu:\~/cmake-test/build$ cmake -DCMAKE_TOOLCHAIN_FILE=~/cmake-test/PI.cmake  -DCMAKE_BUILD_TYPE=Debug ..
> -- The C compiler identification is unknown
> -- The CXX compiler identification is unknown
> CMake Error at CMakeLists.txt:2 (project):
>   The CMAKE_C_COMPILER:
> 
>     ~/cmake-test/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-gcc
> 
>   is not a full path to an existing compiler tool.
> 
>   Tell CMake where to find the compiler by setting either the environment
>   variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
>   the compiler, or to the compiler name if it is in the PATH.
> 
> 
> CMake Error at CMakeLists.txt:2 (project):
>   The CMAKE_CXX_COMPILER:
> 
>     ~/cmake-test/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-g++
> 
>   is not a full path to an existing compiler tool.
> 
>   Tell CMake where to find the compiler by setting either the environment
>   variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
>   to the compiler, or to the compiler name if it is in the PATH.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/test/cmake-test/build/CMakeFiles/CMakeOutput.log".
> See also "/home/test/cmake-test/build/CMakeFiles/CMakeError.log".

abhiTronix, 2021-09-09:
> The CMAKE_C_COMPILER:
> 
> ~/cmake-test/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-gcc
> is not a full path to an existing compiler tool.

> The CMAKE_CXX_COMPILER:
> 
> ~/cmake-test/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-g++
> is not a full path to an existing compiler tool.

@jteng7 

Did you checked if this file paths `~/cmake-test/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-gcc` and `~/cmake-test/tools/cross-pi-gcc-8.3.0-2/bin/arm-linux-gnueabihf-g++` exists or not? Follow the docs carefully.

closed 2021-09-09 by abhiTronix

jteng7, 2021-09-09:
Both paths exist:
[image]

Also, I double-checked all dependencies are installed. 

One thing I noticed in the docs is I had to create a CMakeLists.txt file, not CMakeList.txt file (note the plural) -- is that an error in the docs?

jteng7, 2021-09-09:
An update: I got it working!

Changes I had to make:

- In PI.cmake, I had to change the tools and rootfs_dir paths to point to the full path (i.e. replace the tilde with the actual path with the home directory)
- In CMakeLists.txt, I had to change the second parameter of add_executable to main.cpp instead of foo.cpp, to match the file to compile

abhiTronix, 2021-09-09:
> In PI.cmake, I had to change the tools and rootfs_dir paths to point to the full path (i.e. replace the tilde with the actual path with the home directory). For example, set(tools /home/test/cmake-test/tools/cross-pi-gcc-8.3.0-2) works but set(tools ~/cmake-test/tools/cross-pi-gcc-8.3.0-2) does NOT work.

@jteng7 Aren't you running Linux?

> In CMakeLists.txt, I had to change the second parameter of add_executable to main.cpp instead of foo.cpp, to match the file to compile

Thanks I'll change it.

abhiTronix, 2021-09-09:
> One thing I noticed in the docs is I had to create a CMakeLists.txt file, not CMakeList.txt file (note the plural) -- is that an error in the docs?

It should be `CMakeLists. txt`.

abhiTronix, 2021-09-09:
@jteng7 Can you try set(tools $ENV{HOME}/cmake-test/tools/cross-pi-gcc-8.3.0-2) instead of `set(tools /home/test/cmake-test/tools/cross-pi-gcc-8.3.0-2)`?

abhiTronix, 2021-09-09:
Thanks, Docs updated.

jteng7, 2021-09-09:
> @jteng7 Aren't you running Linux?

Yeah, but one wrinkle is I'm running Ubuntu through VMWare, so that might be a reason for the issues I'm seeing?

> @jteng7 Can you try `set(tools $ENV{HOME}/cmake-test/tools/cross-pi-gcc-8.3.0-2)` instead of `set(tools /home/test/cmake-test/tools/cross-pi-gcc-8.3.0-2)`?
Thanks - using this works as well 👍