abhiTronix/raspberry-pi-cross-compilers #81: fatal error: sys/cdefs.h: No such file or directory

cooked, 2021-09-30:
Hi,
thanks for the nice framework. I've managed to cross compile and deploy Qt from linux PC to Raspberry Pi 4.
I'm trying to compile a test program for the RPi in QtCreator on the PC, but I get the following error. Any clue ?

In file included from /home/stefano/qtrpi4/tools/cross-pi-gcc-8.3.0-2/arm-linux-gnueabihf/include/c++/8.3.0/arm-linux-gnueabihf/bits/os_defines.h:39,
                 from /home/stefano/qtrpi4/tools/cross-pi-gcc-8.3.0-2/arm-linux-gnueabihf/include/c++/8.3.0/arm-linux-gnueabihf/bits/c++config.h:508,
                 from /home/stefano/qtrpi4/tools/cross-pi-gcc-8.3.0-2/arm-linux-gnueabihf/include/c++/8.3.0/type_traits:38,
                 from ../qtrpi4/qt5.15/include/QtCore/qglobal.h:45,
                 from ../qtrpi4/qt5.15/include/QtGui/qtguiglobal.h:43,
                 from ../qtrpi4/qt5.15/include/QtWidgets/qtwidgetsglobal.h:43,
                 from ../qtrpi4/qt5.15/include/QtWidgets/qmainwindow.h:43,
                 from ../qtrpi4/qt5.15/include/QtWidgets/QMainWindow:1,
                 from ../qt-test-2/mainwindow.h:4,
                 from ../qt-test-2/main.cpp:1:
/home/stefano/qtrpi4/sysroot/usr/include/features.h:424:12: fatal error: sys/cdefs.h: No such file or directory

abhiTronix, 2021-10-01:
@cooked Seems like you either failed to symlink `sys` directly properly or failed to rsync `/usr/include` directory from your raspberry pi. 

### Solution

1. First, Make symlinks properly on your raspberry pi:
  ```sh
 sudo ln -sf /usr/include/arm-linux-gnueabihf/asm/* /usr/include/asm/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/gnu/* /usr/include/gnu/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/bits/* /usr/include/bits/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/sys/* /usr/include/sys/
 sudo ln -sf /usr/include/arm-linux-gnueabihf/openssl/* /usr/include/openssl/
  ```

2. Then, Do [step-6](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md#6-sync-raspberry-pi-sysroot-most-important) and [step-7](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md#7-fix-symbolic-links) again.

3. Clear build directory:
```sh
  cd ~/rpi-qt/build
  sudo rm -rf *
 ```

4. Run [step-8](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md#8-configure-qt-build) and continue from there as shown in guide.

closed 2021-10-01 by abhiTronix

cooked, 2021-10-01:
Hi @abhiTronix , thanks that solved the problem. Weirdly enough in my command line history I could see that I had issued already all the simlinks exactly as per your instructions

abhiTronix, 2021-10-01:
> Weirdly enough in my command line history I could see that I had issued already all the simlinks exactly as per your instructions

@cooked Your're not wrong. Symlinks behave weirdly when there's already similar named folder/file present at destination, they sometime works and other times not. That's why I'm looking for alternatives to create symlinks safely.

abhiTronix, 2021-10-10:
**@cooked New [SSymlinker](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/utils/SSymlinker) bash script here to create symlinks more reliably. 😃**

cooked, 2021-10-24:
> **@cooked New [SSymlinker](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/utils/SSymlinker) bash script here to create symlinks more reliably. smiley**

that seems to work a treat !!!

in the meantime got stuck on another error that might be related and hopefully you can help me solve
when I try to crosscompile on PC my program I get:
`arm-linux-gnueabihf/bin/ld: warning: libmmal_core.so, needed by /home/stefano/qtrpi4/sysroot/usr/lib/arm-linux-gnueabihf/libavcodec.so.58, not found (try using -rpath or -rpath-link)`

I've qt built with opencv support (and opencv is installed on the rpi)... it looks like the /opt/vc folder is not copied or symlinked correctly...

any hints?