abhiTronix/raspberry-pi-cross-compilers #82: Cmake build fails for raspberry pi 4 target

vishnucool220, 2021-10-06:
I have followed your tutorial and everything works smooth till i select cmake instead of qmake. Qt projects based on qmake build system works fine but Qt projects based on cmake is not working. I get the following Cmake build errors:

[image]

What exactly is the problem? and how can i debug it?

thank you! 

Edit1 :
how can i set Cmake from crosscompiler version instead of host. ?

abhiTronix, 2021-10-06:
How your're using cmake, what's content of your cmake file, and on which system(linux, mac, WSL(windows)) you're using it?

abhiTronix, 2021-10-06:
> @abhiTronix its basic cmake file:
> 
> ```
> 
> cmake_minimum_required(VERSION 3.5)
> 
> project(cmake-example LANGUAGES CXX)
> 
> set(CMAKE_INCLUDE_CURRENT_DIR ON)
> 
> set(CMAKE_AUTOUIC ON)
> set(CMAKE_AUTOMOC ON)
> set(CMAKE_AUTORCC ON)
> 
> set(CMAKE_CXX_STANDARD 11)
> set(CMAKE_CXX_STANDARD_REQUIRED ON)
> 
> find_package(Qt5Core)
> 
> add_executable(cmake-example
>   main.cpp
> )
> target_link_libraries(cmake-example Qt5::Core)
> ```
> 
> I am on Linux Ubunu ( on VM (running on windows)). The software is Qt Creator.
> 
> How can i build my Qt project on my host system targeting raspberry pi 4 from termianl?

@vishnucool220 It will not work, you atleast need rsynced sysroot from raspberry pi to be able to build anything. See this doc: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os