abhiTronix/raspberry-pi-cross-compilers #84: Raspberry Pi GCC Cross-Compiler Toolchains (Bullseye)

distlibs, 2021-11-21:
When GCC cross-compiler toolchains will be available for Raspberry Pi OS Bullseye?

abhiTronix, 2021-11-26:
@distlibs @matzrm Can anyone list the version of `gcc` `glibc` `binutils` as follows:

```sh
dpkg -l | grep gcc
dpkg -l | grep g++
dpkg -l | grep binutils
dpkg -l | grep libc6
```

Paste the output of each line.

distlibs, 2021-11-27:
dpkg output:

```
||/ Name                                 Version                          Architecture Description
+++-====================================-================================-============-===============================================================================
ii  adduser                              3.118                            all          add and remove users and groups
ii  alsa-topology-conf                   1.2.4-1                          all          ALSA topology configuration files
ii  alsa-ucm-conf                        1.2.4-2                          all          ALSA Use Case Manager configuration files
ii  alsa-utils:armhf                     1.2.4-1+rpt1                     armhf        Utilities for configuring and using ALSA
ii  apt:armhf                            2.2.4                            armhf        commandline package manager
ii  apt-listchanges                      3.24                             all          package change history notification tool
ii  apt-utils:armhf                      2.2.4                            armhf        package management related utility programs
ii  avahi-daemon:armhf                   0.8-5                            armhf        Avahi mDNS/DNS-SD daemon
ii  base-files:armhf                     11.1+rpi1+deb11u1                armhf        Debian base system miscellaneous files
ii  base-passwd:armhf                    3.5.51                           armhf        Debian base system master password and group files
ii  bash:armhf                           5.1-2                            armhf        GNU Bourne Again SHell
ii  bash-completion                      1:2.11-2                         all          programmable completion for the bash shell
ii  bind9-host:armhf                     1:9.16.22-1~deb11u1              armhf        DNS Lookup Utility
ii  bind9-libs:armhf                     1:9.16.22-1~deb11u1              armhf        Shared Libraries used by BIND 9
ii  binutils:armhf                       2.35.2-2+rpi1                    armhf        GNU assembler, linker and binary utilities
ii  binutils-arm-linux-gnueabihf:armhf   2.35.2-2+rpi1                    armhf        GNU binary utilities, for arm-linux-gnueabihf target
ii  binutils-common:armhf                2.35.2-2+rpi1                    armhf        Common files for the GNU assembler, linker and binary utilities
ii  bluez:armhf                          5.55-3.1+rpt1                    armhf        Bluetooth tools and daemons
ii  bluez-firmware                       1.2-4+rpt8                       all          Firmware for Bluetooth devices
ii  bsdextrautils:armhf                  2.36.1-8                         armhf        extra utilities from 4.4BSD-Lite
ii  bsdutils:armhf                       1:2.36.1-8                       armhf        basic utilities from 4.4BSD-Lite
ii  build-essential:armhf                12.9                             armhf        Informational list of build-essential packages
ii  busybox:armhf                        1:1.30.1-6                       armhf        Tiny utilities for small and embedded systems
ii  bzip2:armhf                          1.0.8-4                          armhf        high-quality block-sorting file compressor - utilities
ii  ca-certificates                      20210119                         all          Common CA certificates
ii  cifs-utils:armhf                     2:6.11-3.1                       armhf        Common Internet File System utilities
ii  console-setup                        1.205                            all          console font and keymap setup program
ii  console-setup-linux                  1.205                            all          Linux specific part of console-setup
ii  coreutils:armhf                      8.32-4                           armhf        GNU core utilities
ii  cpio:armhf                           2.13+dfsg-4                      armhf        GNU cpio -- a program to manage archives of files
ii  cpp:armhf                            4:10.2.1-1+rpi1                  armhf        GNU C preprocessor (cpp)
ii  cpp-10:armhf                         10.2.1-6+rpi1                    armhf        GNU C preprocessor
ii  crda:armhf                           4.14+git20191112.9856751-1       armhf        wireless Central Regulatory Domain Agent
ii  cron:armhf                           3.0pl1-137                       armhf        process scheduling daemon
ii  curl:armhf                           7.74.0-1.3                       armhf        command line tool for transferring data with URL syntax
ii  dash:armhf                           0.5.11+git20200708+dd9ef66-5     armhf        POSIX-compliant shell
ii  dbus:armhf                           1.12.20-2                        armhf        simple interprocess messaging system (daemon and utilities)
ii  dc:armhf                             1.07.1-2+b1                      armhf        GNU dc arbitrary precision reverse-polish calculator
ii  debconf                              1.5.77                           all          Debian configuration management system
ii  debconf-i18n                         1.5.77                           all          full internationalization support for debconf
ii  debconf-utils                        1.5.77                           all          debconf utilities
[... cut ...]
ii  xkb-data                             2.29-2                           all          X Keyboard Extension (XKB) configuration data
ii  xxd:armhf                            2:8.2.2434-3                     armhf        tool to make (or reverse) a hex dump
ii  xz-utils:armhf                       5.2.5-2                          armhf        XZ-format compression utilities
ii  zlib1g:armhf                         1:1.2.11.dfsg-2                  armhf        compression library - runtime
ii  zlib1g-dev:armhf                     1:1.2.11.dfsg-2                  armhf        compression library - development
```

distlibs, 2021-11-27:
gcc 10.2.1
g++ 10.2.1
glibc 2.31
binutils 2.35.2

```
$ gcc --version
gcc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
$ g++ --version
g++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
$ ldd --version
ldd (Debian GLIBC 2.31-13+rpt2+rpi1) 2.31
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.
```

```
ld --version
GNU ld (GNU Binutils for Raspbian) 2.35.2
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

abhiTronix, 2021-11-27:
@distlibs Thank you so much. Much appreciated.

closed 2021-11-28 by abhiTronix

abhiTronix, 2021-11-28:
@distlibs @matzrm Bullseye support has been added. Please wait few hours before CI build finishes building the toolchains. 😃

matzrm, 2021-11-28:
Well done! Few days ago I have compiled  my GCC 10.2.0 aarch64 version. My
target is try to compile qt6 64bit with eglfs support, but I stopped
because GLdispatch.so lib is not found (bit the libs was here).

Do you will try to compile Qt6 with Kms (not fake Kms)? Can we work
together on it?

abhiTronix, 2021-11-29:
@matzrm Not a right place to ask this. But see issue #77

commit b5f40697 (2021-11-28): Initial support for Raspbian Bullseye OS(Debian 11) ARM/ARM64 Raspberry Pi Toolchains. (Fixes #84)

- Added GCC version `10.2.0` as Bullseye toolchains base.
- Added Binutils version `2.35.2` as Bullseye toolchains base.
- Added GLIBC version `2.31` as Bullseye toolchains base.
- Updated User and CI bash scripts.
- Updated Binary and Docs Sourceforge deployer.
- Updated complete Docs and links.
- Remove obsolete GCC versions from docs.
commit b8e9e918 (2021-11-28): Added CI workflows for Raspbian Bullseye OS(Debian 11) ARM/ARM64 Raspberry Pi Toolchains. (Fixes #84)

- Updated GitHub action workflows.
- Added support for GCC `10.2.0`.
- Added support for Bullseye OS.
- Added workflow to update Bullseye Toolchains docs.
- Added patch to disable GCC `10.2.0` builds for non-Bullseye OSes.