abhiTronix/raspberry-pi-cross-compilers #85: crtn.o crt1.o crti.o is not valid destination

SFMetec, 2021-11-26:
Hello,

I'm following your guide, I'm using raspberry pi3 b+ with Raspbian Stretch.

When I add symbolic link with SSymlinker I have the errors below:

```
./SSymlinker -s /usr/lib/arm-linux-gnueabihf/crtn.o -d /usr/lib/crtn.o
/usr/lib/crtn.o is not valid destination. Aborting!
```

```
./SSymlinker -s /usr/lib/arm-linux-gnueabihf/crt1.o -d /usr/lib/crt1.o
/usr/lib/crt1.o is not valid destination. Aborting!
```

```
./SSymlinker -s /usr/lib/arm-linux-gnueabihf/crti.o -d /usr/lib/crti.o
/usr/lib/crti.o is not valid destination. Aborting!
```

What am I doing wrong?

abhiTronix, 2021-11-26:
@SFMetec What's the value if you run `readlink -f /usr/lib/crti.o`

SFMetec, 2021-11-26:
> @SFMetec What's the value if you run `readlink -f /usr/lib/crti.o`

```
pi@raspberrypi:~ $ readlink -f /usr/lib/crti.o
/usr/lib/crti.o
```

abhiTronix, 2021-11-26:
@SFMetec OK I found the problem. Please wait.

SFMetec, 2021-11-26:
This the result of other link

```
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/asm -d /usr/include
'/usr/include/asm' -> '/usr/include/arm-linux-gnueabihf/asm'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/gnu -d /usr/include
'/usr/include/gnu' -> '/usr/include/arm-linux-gnueabihf/gnu'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/bits -d /usr/include
'/usr/include/bits' -> '/usr/include/arm-linux-gnueabihf/bits'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include
'/usr/include/sys/*' -> '/usr/include/arm-linux-gnueabihf/sys/*'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/openssl -d /usr/include
'/usr/include/openssl/*' -> '/usr/include/arm-linux-gnueabihf/openssl/*'

```

abhiTronix, 2021-11-26:
@SFMetec Re-clone and check if problem is fixed.

SFMetec, 2021-11-26:
> @SFMetec Re-clone and check if problem is fixed.

```
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/asm -d /usr/include
./SSymlinker: line 56: syntax error in conditional expression: unexpected token `;'
```

abhiTronix, 2021-11-26:
@SFMetec Hold on another bug is there.

abhiTronix, 2021-11-26:
@SFMetec Finally done, check again.

SFMetec, 2021-11-26:
> @SFMetec Finally done, check again.

Thank a lot, now I can go on

```
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/asm -d /usr/include
'/usr/include/asm' -> '/usr/include/arm-linux-gnueabihf/asm'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/gnu -d /usr/include
'/usr/include/gnu' -> '/usr/include/arm-linux-gnueabihf/gnu'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/bits -d /usr/include'/usr/include/bits' -> '/usr/include/arm-linux-gnueabihf/bits'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include
'/usr/include/sys/*' -> '/usr/include/arm-linux-gnueabihf/sys/*'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/openssl -d /usr/include
'/usr/include/openssl/*' -> '/usr/include/arm-linux-gnueabihf/openssl/*'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/lib/arm-linux-gnueabihf/crtn.o -d /usr/lib/crtn.o
'/usr/lib/crtn.o' -> '/usr/lib/arm-linux-gnueabihf/crtn.o'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/lib/arm-linux-gnueabihf/crt1.o -d /usr/lib/crt1.o
'/usr/lib/crt1.o' -> '/usr/lib/arm-linux-gnueabihf/crt1.o'
pi@raspberrypi:~ $ ./SSymlinker -s /usr/lib/arm-linux-gnueabihf/crti.o -d /usr/lib/crti.o
'/usr/lib/crti.o' -> '/usr/lib/arm-linux-gnueabihf/crti.o'
```

abhiTronix, 2021-11-26:
@SFMetec Thanks a lot. It was a very annoying bug.

SFMetec, 2021-11-26:
During compilation I have the error below
```

.obj/qharfbuzzng.o: In function `hb_qt_font_get_for_engine(QFontEngine*)':
qharfbuzzng.cpp:(.text+0x1220): undefined reference to `hb_font_set_ptem'
collect2: error: ld returned 1 exit status
```

SFMetec, 2021-11-26:
> During compilation I have the error below
> 
> ```
> 
> .obj/qharfbuzzng.o: In function `hb_qt_font_get_for_engine(QFontEngine*)':
> qharfbuzzng.cpp:(.text+0x1220): undefined reference to `hb_font_set_ptem'
> collect2: error: ld returned 1 exit status
> ```

After resync the build is successfully terminated but when I try to compile a new empty project I have:

```
/home/xxxx/rpi-qt/sysroot/usr/include/features.h:364: error: sys/cdefs.h: No such file or directory
 #  include <sys/cdefs.h>
```

abhiTronix, 2021-11-26:
> After resync the build is successfully terminated but when I try to compile a new empty project I have:
> 
> /home/xxxx/rpi-qt/sysroot/usr/include/features.h:364: error: sys/cdefs.h: No such file or directory
>  #  include <sys/cdefs.h>

Meaning rysnc didn't worked peoperly, resync each folder again and look for any error.

abhiTronix, 2021-11-26:
> During compilation I have the error below
> 
> 
> .obj/qharfbuzzng.o: In function `hb_qt_font_get_for_engine(QFontEngine*)':
> qharfbuzzng.cpp:(.text+0x1220): undefined reference to `hb_font_set_ptem'
> collect2: error: ld returned 1 exit status

and this is a unresolved QT bug: https://bugreports.qt.io/browse/QTBUG-72956

SFMetec, 2021-11-26:
> 

The rync works fine because the cdefs.h is available in ~/rpi-qt/sysroot/usr/include/arm-linux-gnueabihf/sys

But the error is still there

abhiTronix, 2021-11-26:
> The rync works fine because the cdefs.h is available in ~/rpi-qt/sysroot/usr/include/arm-linux-gnueabihf/sys

That's not important, Is it available at `~/rpi-qt/sysroot/usr/include/sys`?

SFMetec, 2021-11-26:
> 

No is not

abhiTronix, 2021-11-26:
> No is not

Then symlink failed. Check on Raspberry pi, does it have that?

abhiTronix, 2021-11-26:
Also, what's `'*'`?

SFMetec, 2021-11-26:
> 

`'*' -> '/usr/include/arm-linux-gnueabihf/sys/*'
`

SFMetec, 2021-11-26:
> > No is not
> 
> Then symlink failed. Check on Raspberry pi, does it have that?

`pi@raspberrypi:/usr/include/arm-linux-gnueabihf/sys $ ls -la | grep cdefs.h
-rw-r--r--  1 root root 15651 Feb  6  2019 cdefs.h
`

abhiTronix, 2021-11-26:
> `pi@raspberrypi:/usr/include/arm-linux-gnueabihf/sys $ ls -la | grep cdefs.h
> 
> -rw-r--r-- 1 root root 15651 Feb 6 2019 cdefs.h

Check in `/usr/include/sys`

abhiTronix, 2021-11-26:
Does `/usr/include/sys` even exist?

SFMetec, 2021-11-26:
> 

No it doesn't exist

abhiTronix, 2021-11-26:
SSymlinker didn't worked. Check other symlinks like: `asm` `gnu` `bits`?

abhiTronix, 2021-11-26:
/usr/include/asm
/usr/include/gnu 
/usr/include/bits

SFMetec, 2021-11-26:
> 

What I have to check?

abhiTronix, 2021-11-26:
Also, run `ls -la` in `/usr/include`

abhiTronix, 2021-11-26:
> What I have to check?

Check if following symlinks exists or not on Raspberry pi.
```sh
/usr/include/asm
/usr/include/gnu
/usr/include/bits
```

SFMetec, 2021-11-26:
> 

```
pi@raspberrypi:/usr/include $ ls -la asm
lrwxrwxrwx 1 root root 36 Nov 26 12:05 asm -> /usr/include/arm-linux-gnueabihf/asm
```
```
pi@raspberrypi:/usr/include $ ls -la gnu
lrwxrwxrwx 1 root root 36 Nov 26 12:05 gnu -> /usr/include/arm-linux-gnueabihf/gnu
```
```
pi@raspberrypi:/usr/include $ ls -la bits
lrwxrwxrwx 1 root root 37 Nov 26 12:05 bits -> /usr/include/arm-linux-gnueabihf/bits
```

abhiTronix, 2021-11-26:
> pi@raspberrypi:/usr/include $ ls -la asm
> lrwxrwxrwx 1 root root 36 Nov 26 12:05 asm -> /usr/include/arm-linux-gnueabihf/asm
> pi@raspberrypi:/usr/include $ ls -la gnu
> lrwxrwxrwx 1 root root 36 Nov 26 [...]

@SFMetec So this problem narrow downs to `sys` folder only.

abhiTronix, 2021-11-26:
> '*' -> '/usr/include/arm-linux-gnueabihf/sys/*'

@SFMetec This wrong. You have to remove `'*'` first and then make symlink.

Do this:
```sh
sudo rm /usr/include/'*' # beware '*' is in quotes
```
Then run SSymlinker

```sh
./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include
 ls -la /usr/include/sys #verify
```

SFMetec, 2021-11-26:
> 

`sudo rm /usr/include/'*' # beware '*' is in quotes`

or 

`sudo rm /usr/include/sys/'*' # beware '*' is in quotes`

?

abhiTronix, 2021-11-26:
`'*'` is in `/usr/include/sys` or `/usr/include`?

SFMetec, 2021-11-26:
> 

```
pi@raspberrypi:/usr/include $ ls -la sys

drwxr-xr-x   2 root root  4096 Nov 26 12:05 .
drwxr-xr-x 117 root root 20480 Nov 26 12:05 ..
lrwxrwxrwx   1 root root    38 Nov 26 12:05 * -> /usr/include/arm-linux-gnueabihf/sys/*
-rw-r--r--   1 root root  3755 Feb 12  2016 acl.h
-rw-r--r--   1 root root  7774 Feb 27  2018 apparmor.h
-rw-r--r--   1 root root  1090 Feb 27  2018 apparmor_private.h
-rw-r--r--   1 root root    96 Mar 25  2019 asoundlib.h
-rw-r--r--   1 root root  3297 May 24  2016 capability.h
```

abhiTronix, 2021-11-26:
Then do `sudo rm /usr/include/sys/'*' # beware '*' is in quotes`.

abhiTronix, 2021-11-26:
Leave quotes.

SFMetec, 2021-11-26:
```
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include
'/usr/include/sys/*' -> '/usr/include/arm-linux-gnueabihf/sys/*'
```

abhiTronix, 2021-11-26:
Now ` ls -la /usr/include/sys #verify`

abhiTronix, 2021-11-26:
What's the content of `/usr/include/sys`? Do `ls /usr/include/sys`.

SFMetec, 2021-11-26:
> Then do `sudo rm /usr/include/sys/*`.

NO! I removed all files with * without quote! I was stupid

abhiTronix, 2021-11-26:
> NO! I removed all files with * without quote! I was stupid

That can't happen, `sudo rm` only works for file.

SFMetec, 2021-11-26:
> What's the content of `/usr/include/sys`? Do `ls /usr/include/sys`.

```
pi@raspberrypi:~ $ ls -la /usr/include/sys

drwxr-xr-x   2 root root  4096 Nov 26 16:27 .
drwxr-xr-x 117 root root 20480 Nov 26 12:05 ..
lrwxrwxrwx   1 root root    38 Nov 26 16:27 * -> /usr/include/arm-linux-gnueabihf/sys/*
```

abhiTronix, 2021-11-26:
What all files btw?

abhiTronix, 2021-11-26:
> -rw-r--r--   1 root root  3755 Feb 12  2016 acl.h
> -rw-r--r--   1 root root  7774 Feb 27  2018 apparmor.h
> -rw-r--r--   1 root root  1090 Feb 27  2018 apparmor_private.h
> -rw-r--r--   1 root root    96 Mar 25  2019 asoundlib.h
> -rw-r--r--   1 root root  3297 May 24  2016 capability.h

These files deleted? You can easily find them on internet, they're just header files.

SFMetec, 2021-11-26:
> 

Yes, those

abhiTronix, 2021-11-26:
IDK, what's wrong, why it is creating `*` folder? Since you already have removed everything in `sys`, just do `sudo rm -r /usr/include/sys`. And then run `SSymlinker`.

SFMetec, 2021-11-26:
```
pi@raspberrypi:~ $ sudo rm -r /usr/include/sys

pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include

'/usr/include/sys' -> '/usr/include/arm-linux-gnueabihf/sys'

pi@raspberrypi:~ $  ls -la /usr/include/sys

lrwxrwxrwx 1 root root 36 Nov 26 16:35 /usr/include/sys -> /usr/include/arm-linux-gnueabihf/sys
```

abhiTronix, 2021-11-26:
Now search if `/usr/include/arm-linux-gnueabihf/sys/asoundlib.h` exists?

SFMetec, 2021-11-26:
No it not exist

abhiTronix, 2021-11-26:
The problem is resolved, I'll tell you how to bring back those files but first I need to see something. Run following commands:

```sh
sudo rm /usr/include/sys
sudo mkdir /usr/include/sys
```
Then run `SSymlinker` again.

And verify:
```sh
ls -la /usr/include/sys
```

SFMetec, 2021-11-26:
Tomorrow I retry from the beginning

abhiTronix, 2021-11-26:
> Tomorrow I retry from the beginning

Not beginning but my previous comment.

SFMetec, 2021-11-26:
> The problem is resolved, I'll tell you how to bring back those files but first I need to see something. Run following commands:
> 
> ```shell
> sudo rm /usr/include/sys
> sudo mkdir /usr/include/sys
> ```
> 
> Then run `SSymlinker` again.
> 
> And verify:
> 
> ```shell
> ls -la /usr/include/sys
> ```

```
pi@raspberrypi:~ $ sudo rm /usr/include/sys

pi@raspberrypi:~ $ sudo mkdir /usr/include/sys

pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include
'/usr/include/sys/*' -> '/usr/include/arm-linux-gnueabihf/sys/*'

pi@raspberrypi:~ $ ls -la /usr/include/sys
total 24
drwxr-xr-x   2 root root  4096 Nov 26 16:41 .
drwxr-xr-x 117 root root 20480 Nov 26 16:41 ..
lrwxrwxrwx   1 root root    38 Nov 26 16:41 * -> /usr/include/arm-linux-gnueabihf/sys/*
```

abhiTronix, 2021-11-26:
> pi@raspberrypi:~ $ ls -la /usr/include/sys
> total 24
> drwxr-xr-x   2 root root  4096 Nov 26 16:41 .
> drwxr-xr-x 117 root root 20480 Nov 26 16:41 ..
> lrwxrwxrwx   1 root root    38 Nov 26 16:41 * -> /usr/include/arm-linux-gnueabihf/sys/*

This is a bug, thank you, I'll work on it and get back to you tommorrow.

SFMetec, 2021-11-26:
> d

Thank a lot

abhiTronix, 2021-11-26:
@SFMetec OK I've solved the problem. Try these commands again after re-cloning to test it:

> Run following commands:
> sudo rm /usr/include/sys
> sudo mkdir /usr/include/sys

> Then run SSymlinker again.
> 
> And verify:
> 
> ls -la /usr/include/sys

closed 2021-11-26 by abhiTronix

SFMetec, 2021-11-29:
> 

I removed the sys folder and after I re-make the directory and resumed the deleted files:

```
pi@raspberrypi:/usr/include $ ls -la sys
total 48
drwxr-xr-x   2 root root  4096 Nov 29 07:23 .
drwxr-xr-x 117 root root 20480 Nov 29 07:17 ..
-rw-r--r--   1 root root  3755 Nov 29 07:23 acl.h
-rw-r--r--   1 root root  7774 Nov 29 07:23 apparmor.h
-rw-r--r--   1 root root  1090 Nov 29 07:23 apparmor_private.h
-rw-r--r--   1 root root    96 Nov 29 07:23 asoundlib.h
-rw-r--r--   1 root root  3297 Nov 29 07:23 capability.h
```

After that, I downloaded again the SSymlinker but I have the error below:

```
pi@raspberrypi:~ $ ./SSymlinker -s /usr/include/arm-linux-gnueabihf/sys -d /usr/include
ln: failed to create symbolic link '/usr/include/sys/acct.h': Permission denied
```

abhiTronix, 2021-11-29:
> ln: failed to create symbolic link '/usr/include/sys/acct.h': Permission denied

Did you tried using `sudo` as follows:

```sh
sudo chmod +x SSymlinker
```

abhiTronix, 2021-11-29:
@SFMetec Yes `sudo`  was missing in script, I've re-added it, now it will work normally.

[9 more replies]

commit 268f439f (2021-11-26): Fixed logic bug in validating non-existing destination paths. (Fixes #85)
commit 34c3bfb3 (2021-11-26): Fixed syntax bug and line endings (Fixed #85)
commit e61ca2de (2021-11-26): SSymlinker: Fixed critical globbing bug in bash script (Fixes #85)

- Since "*" is being used as the link name in `ls` command, and it's not getting expanded because of some unknown reason, thereby, it's not creating symlink for all files  in directory and is just adding '*' at destination. 
- Solution is to manually cycle through files in folder using loop and redirect them to `ls` command.