abhiTronix/raspberry-pi-cross-compilers #89: Cross compiler toolchains for Bullseye fail to compile simple program?

f-laurens, 2021-12-04:
I am attempting to use the cross compiler toolchains for Bullseye to build Python, and the Python build 'configure' step detects that the compiler is not able to compile a simple program:
```
configure:2844: checking build system type
configure:2858: result: x86_64-pc-linux-gnu
configure:2878: checking host system type
configure:2891: result: arm-unknown-linux-gnueabihf
configure:2921: checking for python3.9
configure:2948: result: python3.9
configure:2963: checking for python interpreter for cross build
configure:2976: result: python3.9
configure:3042: checking for --enable-universalsdk
configure:3089: result: no
configure:3113: checking for --with-universal-archs
configure:3128: result: no
configure:3284: checking MACHDEP
configure:3335: result: "linux"
configure:3589: checking for arm-linux-gnueabihf-gcc
configure:3605: found /opt/cross-pi-gcc-10.2.0-0/bin/arm-linux-gnueabihf-gcc
configure:3616: result: arm-linux-gnueabihf-gcc
configure:3885: checking for C compiler version
configure:3894: arm-linux-gnueabihf-gcc --version >&5
arm-linux-gnueabihf-gcc (GCC) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
configure:3894: arm-linux-gnueabihf-gcc -v >&5
Using built-in specs.
COLLECT_GCC=arm-linux-gnueabihf-gcc
COLLECT_LTO_WRAPPER=/opt/cross-pi-gcc-10.2.0-0/bin/../libexec/gcc/arm-linux-gnueabihf/10.2.0/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../configure --prefix= --target=arm-linux-gnueabihf --enable-languages=c,c++,fortran --with-sysroot=/arm-linux-gnueabihf/libc --with-build-sysroot=/tmp/cross-pi-gcc-10.2.0-0/arm-linux-gnueabihf/libc --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) 
configure:3925: checking whether the C compiler works
configure:3947: arm-linux-gnueabihf-gcc    conftest.c  >&5
/opt/cross-pi-gcc-10.2.0-0/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: /opt/cross-pi-gcc-10.2.0-0/bin/../arm-linux-gnueabihf/libc/usr/lib/crt1.o: in function `_start':
/tmp/build_toolchains/glibc-2.31/csu/../sysdeps/arm/start.S:124: undefined reference to `__libc_start_main'
/opt/cross-pi-gcc-10.2.0-0/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: /tmp/build_toolchains/glibc-2.31/csu/../sysdeps/arm/start.S:128: undefined reference to `abort'
/opt/cross-pi-gcc-10.2.0-0/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: /tmp/build_toolchains/glibc-2.31/csu/../sysdeps/arm/start.S:113: undefined reference to `__libc_csu_fini'
/opt/cross-pi-gcc-10.2.0-0/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: /tmp/build_toolchains/glibc-2.31/csu/../sysdeps/arm/start.S:120: undefined reference to `__libc_csu_init'
collect2: error: ld returned 1 exit status
configure:3989: result: no
configure: failed program was:
| /* confdefs.h */
| #define _GNU_SOURCE 1
| #define _NETBSD_SOURCE 1
| #define __BSD_VISIBLE 1
| #define _DARWIN_C_SOURCE 1
| #define _PYTHONFRAMEWORK ""
| #define _XOPEN_SOURCE 700
| #define _XOPEN_SOURCE_EXTENDED 1
| #define _POSIX_C_SOURCE 200809L
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:3996: error: C compiler cannot create executables
```
I do not have the problem when using the equivalent toolchains for Buster.

abhiTronix, 2021-12-04:
I think Bullseye toolchains are broken, I'm seeing plenty of issues regarding it. I'll look into it.

abhiTronix, 2021-12-04:
@f-laurens **Apparently there's a bug with GCC 10.2.0 when built with Glibc 2.31: https://sourceware.org/pipermail/glibc-cvs/2020q1/069150.html** and committed here: https://sourceware.org/pipermail/libc-alpha/2020-May/113553.html but never was backported to Glibc 2.31 branch.

abhiTronix, 2021-12-04:
Here's what author said:

> I also think that we need this patch backported to glibc 2.31 branch 
> because we are seeing failures for arm-none-linux-gnueabihf target when 
> building GCC 10 with glibc 2.31:

abhiTronix, 2021-12-04:
I think we need to apply patch directly to `math/Makefile`.

abhiTronix, 2021-12-04:
**Successfully resolved and merged in commit https://github.com/abhiTronix/raspberry-pi-cross-compilers/commit/e7353d20f6a134a44ab672afee4a301947b521b9**

abhiTronix, 2021-12-04:
@f-laurens Please wait till updated binaries are release. I'll inform you here.

closed 2021-12-05 by abhiTronix

abhiTronix, 2021-12-05:
@f-laurens There were way too many bugs with both GCC and Glibc. `v3.0.9` toolchains are in build and soon be available.

abhiTronix, 2021-12-05:
**This issue has been resolved in Toolchains version `v3.0.9`. Download it from here: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/**

f-laurens, 2021-12-05:
> This issue has been resolved in the latest Toolchains version `v3.0.9`.

The new version of the Bullseye cross toolchains indeed solves my problem.
Thanks!

commit e7353d20 (2021-12-04): [DEBUG] Critical bug-fix for Bullseye Toolchains (Fixes #89)

- Added patched `Makfile.patch` file that contains critical bug-fix for Glibc `2.31`.
- Bug can explained as follows:
  On platforms where long double has the same ABI as double, glibc-2.31
  defines long double functions as aliases for the corresponding double
  functions.  The declarations of those functions in <math.h> are
  disabled to avoid problems with aliases having incompatible types, but
  GCC 10 now gives errors for incompatible types when the long double
  function is known to GCC as a built-in function, not just when th [... cut]
commit c883af2a (2021-12-05): Added workflow to directy download patch Glibc 2.31 source from git (Fixes #89)