abhiTronix/raspberry-pi-cross-compilers #90: Bullseye Toolchain Error: selected processor does not support `vsdot.s8 q10,q9,d4[0]' in ARM mode

distlibs, 2021-12-11:
I was trying to compile tensorflow-lite 2.7 with Raspberry Pi GCC Native-Compiler Toolchains (Bullseye). I got error:

<pre>
...
[ 73%] Building C object _deps/xnnpack-build/CMakeFiles/XNNPACK.dir/src/qc8-igemm/gen/1x16c4-minmax-fp32-neondot.c.o
[ 73%] Building C object _deps/xnnpack-build/CMakeFiles/XNNPACK.dir/src/qc8-igemm/gen/4x8c4-minmax-fp32-neondot.c.o
[ 73%] Building C object _deps/xnnpack-build/CMakeFiles/XNNPACK.dir/src/qc8-igemm/gen/4x16c4-minmax-fp32-neondot.c.o
/tmp/ccLir1WO.s: Assembler messages:
/tmp/ccLir1WO.s:60: Error: selected processor does not support `vsdot.s8 q8,q11,d7[0]' in ARM mode
/tmp/ccLir1WO.s:62: Error: selected processor does not support `vsdot.s8 q10,q9,d7[0]' in ARM mode
/tmp/ccLir1WO.s:65: Error: selected processor does not support `vsdot.s8 q8,q9,d7[1]' in ARM mode
/tmp/ccLir1WO.s:69: Error: selected processor does not support `vsdot.s8 q10,q9,d7[1]' in ARM mode
/tmp/ccLir1WO.s:141: Error: selected processor does not support `vsdot.s8 q10,q9,d7[0]' in ARM mode
/tmp/ccLir1WO.s:144: Error: selected processor does not support `vsdot.s8 q8,q11,d7[0]' in ARM mode
make[2]: *** [_deps/xnnpack-build/CMakeFiles/XNNPACK.dir/build.make:5458: _deps/xnnpack-build/CMakeFiles/XNNPACK.dir/src/qc8-gemm/gen/1x8c4-minmax-fp32-neondot.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/tmp/ccpalmhZ.s: Assembler messages:
</pre>

Using Buster toolchain tensorflow-lite compiles without any errors.

It can be related with:
* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101723

Maybe you can apply patch like you did here https://github.com/abhiTronix/raspberry-pi-cross-compilers/issues/89

abhiTronix, 2021-12-11:
> Using Buster toolchain tensorflow-lite compiles without any errors.
> 
> It can be related with:
> 
> https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101723
> Cross-compilation for RPi3 (armv7) fails on assembly google/XNNPACK#1465
> CMake build error on armv7 with latest gcc google/XNNPACK#2003

@distlibs Thanks for bug report. What I'm understanding is that GCC `9-to-11` series is affected by this bug when Binutils `>2.33` is used. More specifically Binutils `2.35.1` used in Bullseye is affected is causing this bug. Moreover, this bug is already fixed in master branch of [GCC `>9`](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101723) but haven't been included in releases branches like [`10.3.0`](https://gcc.gnu.org/git/?p=gcc.git;a=tags) which was updated 8 months ago.

### Solution:
We can easily clone patched `releases/gcc-10` directly for Bullseye for compiling GCC `>10.2.0` binaries to resolve this bug.

distlibs, 2021-12-11:
Binaries cross-compiled with GCC 10.3.0 will be compatible with Raspberry Pi 4 which runs Bullseye with GCC 10.2.1?

abhiTronix, 2021-12-12:
Yes, aren't you using Raspberry Pi GCC-10.3.0 Native-Compiler Toolchains (Bullseye) right now? The GCC version doesn't matter, but only  GLIBC, KERNEL and BINUTILS version matters, which remains the same.

distlibs, 2021-12-12:
I am using right now GCC-**10.2.0** Raspberry Pi GCC **Cross-Compiler** Toolchains (Bullseye). I made mistake I updated original post.

abhiTronix, 2021-12-12:
Yes, you can use GCC-10.3.0 instead.

distlibs, 2021-12-12:
So I need to wait until you clone patched `releases/gcc-10` branch directly and pre-compile GCC 10.3.0 binaries for Bullseye? New Raspberry Pi GCC 10.3.0 Cross-Compiler Toolchains (Bullseye) binaries will be uploaded SourceForge where I can download?

abhiTronix, 2021-12-12:
You need to wait untill `v3.1.0` Toolchains are released, and thereby uploaded to Sourceforge after successful compiling. Thereby, You can `watch` to get notify when that happens.

abhiTronix, 2021-12-12:
@distlibs Patch has been added. Please wait for CI workflows to finish and then I'll make the release.

abhiTronix, 2021-12-12:
@distlibs `v3.10-alpha` toolchains are in build and soon be available on Sourceforge.

abhiTronix, 2021-12-13:
This issue has been resolved in the latest Toolchains version `v3.1.0-alpha` and GCC-10.3.0. Download it from here: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Bullseye/GCC%2010.3.0/

closed 2021-12-13 by abhiTronix

distlibs, 2021-12-13:
Thanks. Compilation finished successfully using GCC-10.3.0 Raspberry Pi Cross-Compiler Toolchains (Bullseye).

commit 3c96a72b (2021-12-12): Patched critical bug in GCC-10 series toolchains for bullseye  (Fixes #90)

- Implemented workflow to directly download patched GCC-10 source master branch from git when compiling bullseye toolchains.
- For more information on this bug, see: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101723