abhiTronix/raspberry-pi-cross-compilers #91: Raspberry pi zero W Illegal instruction

Iuliean, 2022-01-04:
I downloaded the 8.3.0 Buster toolchain and followed the instructions from C1 from [here](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler:-Installation-Instructions) 
and  after compiling a simple  hello world program i get  `Illegal instruction` when i run it on my pi zero

I compiled with `arm-linux-gnueabihf-g++ test.cpp -o test` and with `arm-linux-gnueabihf-g++ -march=armv6 -mfloat-abi=hard -mfpu=vfp test.cpp -o test`

abhiTronix, 2022-01-04:
> I downloaded the 8.3.0 Buster toolchain

@Iuliean Which toolchain, can you paste the exact URL of the toolchain you used? Also, what's your setup, which machine you're using?

Iuliean, 2022-01-04:
https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%208.3.0/

this is the toolchain i downloaded.  i am compiling it on a virtual box vm which runs manjaro for a raspberry pi zero w which runs raspbian (debian 10 )

abhiTronix, 2022-01-05:
> this is the toolchain i downloaded. i am compiling it on a virtual box vm which runs manjaro for a raspberry pi zero w which runs raspbian (debian 10 )

@Iuliean Well isn't that obvious that it will give error? You are trying to execute binary that is cross-compiled for ARM32(raspberry pi) on a AMD64(virtual box vm which runs manjaro), that's why it it is giving this error, instead you need to execute it on raspberry pi(zero w) after compiling. Learn more about Cross-compiling here: https://stackoverflow.com/a/897303/10158117

closed 2022-01-05 by abhiTronix

abhiTronix, 2022-01-05:
> after compiling a simple hello world program i get Illegal instruction when i run it on my pi zero

@Iuliean Sorry for previous comment, This is not it, you're definitely using wrong toolchain for your raspberry pi.  "Illegal instruction" means you're trying to run a program that was compiled for specific processor, and is then running on a processor that fails to meet those requirements. Are you sure you running it on  pi zero? 

> https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%208.3.0/

Also, this is incomplete, I need exact Toolchain URL you used, otherwise paste the name of compressed toolchain  `tar,gz` file you downloaded.

Iuliean, 2022-01-05:
Yeah i am so sorry, i was using the wrong toolchain i didn't notice that the link redirects to a place where multiple toolchains are available and i just downloaded whatever was at the top, i downloaded the correct toolchain now and the simple hello world program works now. Thanks a lot and i am sorry

abhiTronix, 2022-01-07:
> Edit: how do i install libs for it like boost do i have to manually put the files into the lib folder inside the toolchain?

No, you specify them in CMAKE file or you specify during configure/make command.