abhiTronix/raspberry-pi-cross-compilers #92: Potential mistake in the wiki for <Cross Compiler CMake Usage Guide with rsynced Raspberry Pi 64 bit OS>

Mins0o, 2022-01-10:
In the [|host: x86_64|target: arm64 raspOS| cross-compile wiki](https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-64-bit-OS#:~:text=sudo%20chmod%20%2Bx,usr/lib/crti.o), there seems to be a typo where all the directory contains an extra `/asm`.  
  
I am still trying to follow the instruction with my fix (removing it), to check if that succeeds. I'll let you know how it went.

Mins0o, 2022-01-10:
\+ openssl is not in `/usr/include/aarch64-linux-gnu/openssl`, giving error of
```/usr/include/aarch64-linux-gnu/openssl is not valid source file/directory. Aborting!```

 What should I do?

abhiTronix, 2022-01-10:
>+ openssl is not in /usr/include/aarch64-linux-gnu/openssl, giving error of
/usr/include/aarch64-linux-gnu/openssl is not valid source file/directory. Aborting!

@Mins0o leave it, it means openssl is not installed on your pi. It's not a critical dependency, so just skip it.

+ command rsync -avz --rsync-path="sudo rsync" --delete pi@192.168.1.47:/opt/vc rootfs/opt fails with the error:

> receiving incremental file list  
rsync: link_stat "/opt/vc" failed: No such file or directory (2)  
sent 8 bytes  received 76 bytes  15.27 bytes/sec  
total size is 0  speedup is 0.00  
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1668) [Receiver=3.1.2] 
rsync: [Receiver] write error: Broken pipe (32)

Maybe 64bit raspberry os lacks vc packages. So i think skip this also. What are you trying to compile exactly?

abhiTronix, 2022-01-10:
@Mins0o I've corrected the `asm` typo in docs. I'm closing this.

closed 2022-01-10 by abhiTronix

Mins0o, 2022-01-10:
Aha seems like 64bit OSes [does not have /opt/vc](https://raspberrytips.com/raspberry-pi-os-64-bits-vs-32-bits/#:~:text=libraspberrypi0%2C%20libraspberrypi%2Ddev%20and%20libraspberrypi%2Ddoc%20have%20been%20moved%20out%20of%20/opt/vc/*%20and%20into%20/usr/*%20instead%20(making%20it%20more%20standard).%20Any%20code%20built%20against%20these%20libraries%20will%20require%20changing%20to%20refer%20to%20a%20more%20standard%20location%20(/usr/lib/%20rather%20than%20/opt/vc/lib)) in the first place!