abhiTronix/raspberry-pi-cross-compilers #93: cross-pi-gcc-64 fails on debian-stretch on WSL2

Mins0o, 2022-01-10:
I was following the wiki and encountered this issue, with other version of gcc toolchains too (8.3.0, I believe).  
Using `sudo` did not help.

I tried the same process on ubuntu on WSL2, and that didn't have any issues until this point. (It has other issue...)  
  
This is the error I get at the last step, on Debian but not on Ubuntu.
```
cmake -DCMAKE_TOOLCHAIN_FILE=~/cmake-test/PI.cmake  -DCMAKE_BUILD_TYPE=Debug ..
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/aarch64-linux-gnu-gcc
-- Check for working C compiler: /home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/aarch64-linux-gnu-gcc - broken

CMake Error at /usr/local/share/cmake-3.21/Modules/CMakeTestCCompiler.cmake:69 (message):
The C compiler

"/home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/aarch64-linux-gnu-gcc"

is not able to compile a simple test program.

It fails with the following output:

    Change Dir: /home/minsoo/cmake-test/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/make -f Makefile cmTC_453e0/fast && /usr/bin/make  -f CMakeFiles/cmTC_453e0.dir/build.make CMakeFiles/cmTC_453e0.dir/build
    make[1]: Entering directory '/home/minsoo/cmake-test/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_453e0.dir/testCCompiler.c.o
    /home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/aarch64-linux-gnu-gcc --sysroot=/home/minsoo/cmake-test/rootfs   -fPIC -Wl,-rpath-link,/home/minsoo/cmake-test/rootfs/usr/lib/aarch64-linux-gnu -L/home/minsoo/cmake-test/rootfs/usr/lib/aarch64-linux-gnu  -o CMakeFiles/cmTC_453e0.dir/testCCompiler.c.o -c /home/minsoo/cmake-test/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_453e0
    /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_453e0.dir/link.txt --verbose=1
    /home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/aarch64-linux-gnu-gcc --sysroot=/home/minsoo/cmake-test/rootfs  -fPIC -Wl,-rpath-link,/home/minsoo/cmake-test/rootfs/usr/lib/aarch64-linux-gnu -L/home/minsoo/cmake-test/rootfs/usr/lib/aarch64-linux-gnu   -fPIC -Wl,-rpath-link,/home/minsoo/cmake-test/rootfs/usr/lib/aarch64-linux-gnu -L/home/minsoo/cmake-test/rootfs/usr/lib/aarch64-linux-gnu  CMakeFiles/cmTC_453e0.dir/testCCompiler.c.o -o cmTC_453e0
    /home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/../lib/gcc/aarch64-linux-gnu/10.3.0/../../../../aarch64-linux-gnu/bin/ld: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /home/minsoo/cmake-test/tools/cross-pi-gcc-10.3.0-64/bin/../lib/gcc/aarch64-linux-gnu/10.3.0/../../../../aarch64-linux-gnu/bin/ld)
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_453e0.dir/build.make:98: recipe for target 'cmTC_453e0' failed
    make[1]: *** [cmTC_453e0] Error 1
    make[1]: Leaving directory '/home/minsoo/cmake-test/build/CMakeFiles/CMakeTmp'
    Makefile:127: recipe for target 'cmTC_453e0/fast' failed
    make: *** [cmTC_453e0/fast] Error 2
    
    
    
    CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
    CMakeLists.txt:2 (project) 
```

I don't think fix is necessary since installing Ubuntu on WSL2 is a small extra cost, but I thought it might be worth a mention for anyone who bumps into same issue as I do.  
Maybe avoid Debian WSL2 for now and stick to Ubuntu WSL2?

abhiTronix, 2022-01-10:
I don't know what debian stretch does to files. Maybe some files are moved. Or packages are installed in other directories. All and all avoid debian for now.

abhiTronix, 2022-01-10:
@Mins0o Also while cross compiling complex libraries, avoid GCC 10.3.0 and use base toolchains like GCC 8.3.0 in buster, or GCC 10.2.0 in bullseye.