abhiTronix/raspberry-pi-cross-compilers #94: qmake built for x86

Nickellick, 2022-02-07:
I want to use the latest version of Qt5 for Raspberry Pi 4B for PySide2. However, when building the PySide2 library, there are problems running qmake.
After a bit of research, it turned out that qmake was somehow compiled for x86. All other binaries compiled correctly for ARM.

So, if I run (on target, of course)
`cd /usr/local/qt5.15/bin`
and try to check binary with
`file qmake`
It ouputs:
```qmake: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=781f60a8151971256f3ac25972223493fbdf2f4f, for GNU/Linux 3.2.0, with debug_info, not stripped```
I'm pretty sure that I used cross-compiler, because if I run
```file xmlpatterns``` or ```file qml```
It outputs 
```ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, stripped```

So my question is, how do I build qmake for a target?

abhiTronix, 2022-02-07:
How did you build qmake?

Nickellick, 2022-02-07:
> How did you build qmake?

I don't actually specify what to build. I just used your [guide](https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md) (BTW it was very helpful, thanks!)

Nickellick, 2022-02-07:
Important to mention:
I'm using Ubuntu 21.10 as "compiler machine" with gcc 11.2. Because of this, I had some problems with "numeric limits". I added the line
```CPP
#include <limits>
```
within the `#ifdef __cplusplus` block to the file
`~/rpi-qt/build/qtbase/src/corelib/global/qglobal.h`

and `-no-pch` parameter to configure, so it looked like this
`
../qt-everywhere-src-5.15.2/configure -release -opengl es2  -eglfs -device linux-rasp-pi4-v3d-g++ -device-option CROSS_COMPILE=$(echo $CROSS_COMPILER_LOCATION)/bin/arm-linux-gnueabihf- -sysroot ~/rpi-qt/sysroot/ -prefix /usr/local/qt5.15 -extprefix ~/rpi-qt/qt5.15 -opensource -confirm-license -skip qtscript -skip qtwayland -skip qtwebengine -nomake tests -make libs -pkg-config -no-use-gold-linker -L$HOME/rpi-qt/sysroot/usr/lib/arm-linux-gnueabihf -I$HOME/rpi-qt/sysroot/usr/include/arm-linux-gnueabihf -no-pch`
as it was mentioned in [this solution](https://stackoverflow.com/questions/68081481/how-can-i-build-qt-5-13-2-with-gcc-11-1-on-windows)

My `config.summary`:
```
Building on: linux-g++ (x86_64, CPU features: mmx sse sse2)
Building for: devices/linux-rasp-pi4-v3d-g++ (arm, CPU features: neon)
Target compiler: gcc 8.3.0
Configuration: cross_compile compile_examples largefile neon shared shared rpath release c++11 c++14 c++17 c++1z concurrent dbus reduce_exports stl
Build options:
  Mode ................................... release
  Optimize release build for size ........ no
  Building shared libraries .............. yes
  Using C standard ....................... C11
  Using C++ standard ..................... C++17
  Using ccache ........................... no
  Using new DTAGS ........................ no
  Relocatable ............................ yes
  Using precompiled headers .............. no
  Using LTCG ............................. no
  Target compiler supports:
    NEON ................................. yes
  Build parts ............................ libs
Qt modules and options:
  Qt Concurrent .......................... yes
  Qt D-Bus ............................... yes
  Qt D-Bus directly linked to libdbus .... yes
  Qt Gui ................................. yes
  Qt Network ............................. yes
  Qt Sql ................................. yes
  Qt Testlib ............................. yes
  Qt Widgets ............................. yes
  Qt Xml ................................. yes
Support enabled for:
  Using pkg-config ....................... yes
  udev ................................... yes
  Using system zlib ...................... yes
  Zstandard support ...................... no
Qt Core:
  DoubleConversion ....................... yes
    Using system DoubleConversion ........ yes
  GLib ................................... yes
  iconv .................................. no
  ICU .................................... yes
  Built-in copy of the MIME database ..... yes
  Tracing backend ........................ <none>
  Logging backends:
    journald ............................. no
    syslog ............................... no
    slog2 ................................ no
  PCRE2 .................................. yes
    Using system PCRE2 ................... yes
Qt Network:
  getifaddrs() ........................... yes
  IPv6 ifname ............................ yes
  libproxy ............................... no
  Linux AF_NETLINK ....................... yes
  OpenSSL ................................ yes
    Qt directly linked to OpenSSL ........ no
  OpenSSL 1.1 ............................ yes
[... 199 lines cut ...]
  kmap2qmap .............................. yes
  Qt Linguist ............................ yes
  Mac Deployment Tool .................... no
  makeqpf ................................ yes
  pixeltool .............................. yes
  qdbus .................................. yes
  qev .................................... yes
  Qt Attributions Scanner ................ yes
  qtdiag ................................. yes
  qtpaths ................................ yes
  qtplugininfo ........................... yes
  Windows deployment tool ................ no
  WinRT Runner Tool ...................... no
Qt Tools:
  QDoc ................................... no

Note: Also available for Linux: linux-clang linux-icc

WARNING: QDoc will not be compiled, probably because libclang could not be located. This means that you cannot build the Qt documentation.

Either ensure that llvm-config is in your PATH environment variable, or set LLVM_INSTALL_DIR to the location of your llvm installation.
On Linux systems, you may be able to install libclang by installing the libclang-dev or libclang-devel package, depending on your distribution.
On macOS, you can use Homebrew's llvm package.
On Windows, you must set LLVM_INSTALL_DIR to the installation path.
```

Nickellick, 2022-02-08:
> Also, Here in configuration you're using GCC 8.3.0 toolchains. Are you trying to compile QT binaries or something else? Help me understand the complete setup and steps you followed till now.

After configure, I just type `make`. My guess  is that it builds the tools for the host machine by compiler on the host machine, uses them, and then builds binaries for target. All other binaries except `qmake` are built for target.
Of course, I ran `make install` after the build.

abhiTronix, 2022-02-08:
But what are you using GCC 11.2 for?

Nickellick, 2022-02-08:
> But what are you using GCC 11.2 for?

I don't explicitly use it. As I wrote above, it is used automatically when I run the `make` command inside the `build` folder. The command for `configure` and its summary shown above.

All other binaries are compiled for target, so I'm pretty sure the cross-compiler works fine. Also:

> My guess is that it builds the tools for the host machine by compiler on the host machine, uses them, and then builds binaries for target

Nickellick, 2022-02-11:
> Linux Mint x64 with GCC 8.3.0 buster toolchains for Raspberry Pi 3B+. Also, It's not just me, hundreds of people everyday visit that specific guide. So the problem is definitely at your end, and still you're not convinced then feel free to try other toolchains.

Thanks a lot! But please tell me which version of GCC (regular, not for cross-compilation, not for RPi) is installed on your PC? I'll try building QT a few more times with a different environment to see what's going on.

And please listen to me. I don't blame you or your guide. I'm convinced it really works, and it works great. I see this has been tested many times by other people. I just want to know what's going on on my side. So I want to try again with a different environment - just like yours.

It's really strange that my computer only built one x86 binary and all the rest are for ARM. I just want to get it right.

abhiTronix, 2022-02-11:
Sorry but I've to close this. This not a forum.