abhiTronix/raspberry-pi-cross-compilers #98: Cross compile error with Raspberry Pi OS Lite Bullseye

WalterPrech, 2022-03-30:
Hi,
I'm following the step-by-step instructions for cross compiling Qt for Raspberry Pi.

I set up a Raspberry Pi 4 with 32bit Raspberry Pi OS Lite. Also I set up a virtual machine with Ubuntu 20.04 LTS as host to cross compile.

On the Pi 4 I can't enable GL (FAKE KMS) with sudo raspi-config (there is no Advanced Options -> A8 GL Driver -> G2 GL (Fake KMS) available).
Also the directory /opt/vc is not available.

On the host PC, after the build configure, EGLFS Raspberry Pi is set to yes.

The make command ends up with the following errors:
```
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp: In function ‘EGLNativeWindowType createDispmanxLayer(const QPoint&, const QSize&, int, DISPMANX_FLAGS_ALPHA_T)’:
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:74:5: error: ‘EGL_DISPMANX_WINDOW_T’ was not declared in this scope
   74 |     EGL_DISPMANX_WINDOW_T *eglWindow = new EGL_DISPMANX_WINDOW_T;
      |     ^~~~~~~~~~~~~~~~~~~~~
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:74:28: error: ‘eglWindow’ was not declared in this scope; did you mean ‘QWindow’?
   74 |     EGL_DISPMANX_WINDOW_T *eglWindow = new EGL_DISPMANX_WINDOW_T;
      |                            ^~~~~~~~~
      |                            QWindow
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:74:44: error: ‘EGL_DISPMANX_WINDOW_T’ does not name a type
   74 |     EGL_DISPMANX_WINDOW_T *eglWindow = new EGL_DISPMANX_WINDOW_T;
      |                                            ^~~~~~~~~~~~~~~~~~~~~
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp: In function ‘void destroyDispmanxLayer(EGLNativeWindowType)’:
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:5: error: ‘EGL_DISPMANX_WINDOW_T’ was not declared in this scope
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |     ^~~~~~~~~~~~~~~~~~~~~
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:28: error: ‘eglWindow’ was not declared in this scope; did you mean ‘QWindow’?
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |                            ^~~~~~~~~
      |                            QWindow
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:52: error: ‘EGL_DISPMANX_WINDOW_T’ does not name a type
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:74: error: expected ‘>’ before ‘*’ token
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |                                                                          ^
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:74: error: expected ‘(’ before ‘*’ token
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |                                                                          ^
      |                                                                          (
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:75: error: expected primary-expression before ‘>’ token
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |                                                                           ^
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:84:84: error: expected ‘)’ before ‘;’ token
   84 |     EGL_DISPMANX_WINDOW_T *eglWindow = static_cast<EGL_DISPMANX_WINDOW_T *>(window);
      |                                                                                    ^
      |                                                                                    )
/home/user/rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:88:12: error: type ‘<type error>’ argument given to ‘delete’, expected pointer
   88 |     delete eglWindow;
      |            ^~~~~~~~~
make[7]: *** [Makefile:1266: .obj/qeglfsbrcmintegration.o] Error 1
make[7]: Leaving directory '/home/user/rpi-qt/build/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm'
make[6]: *** [Makefile:152: sub-eglfs_brcm-make_first] Error 2
make[6]: Leaving directory '/home/user/rpi-qt/build/qtbase/src/plugins/platforms/eglfs/deviceintegration'
make[5]: *** [Makefile:107: sub-deviceintegration-make_first-ordered] Error 2
make[5]: Leaving directory '/home/user/rpi-qt/build/qtbase/src/plugins/platforms/eglfs'
make[4]: *** [Makefile:128: sub-eglfs-make_first] Error 2
make[4]: Leaving directory '/home/user/rpi-qt/build/qtbase/src/plugins/platforms'
make[3]: *** [Makefile:105: sub-platforms-make_first] Error 2
make[3]: Leaving directory '/home/user/rpi-qt/build/qtbase/src/plugins'
make[2]: *** [Makefile:723: sub-plugins-make_first] Error 2
make[2]: Leaving directory '/home/user/rpi-qt/build/qtbase/src'
make[1]: *** [Makefile:51: sub-src-make_first] Error 2
make[1]: Leaving directory '/home/user/rpi-qt/build/qtbase'
make: *** [Makefile:85: module-qtbase-make_first] Error 2
```
Are there special configurations to do for cross compiling with Raspberry Pi OS Lite Bullseye and Pi 4?

Can you please give me some hints?

WalterPrech, 2022-03-31:
I found a solution/patch to complete cross compiling the Qt libraries for Raspberry Pi4.

Using this patch, solves the error:
```
--- a/qtbase/src/gui/configure.json	2021-12-07 09:49:39.810299326 -0600
+++ b/qtbase/src/gui/configure.json	2021-12-07 09:49:20.546298594 -0600
@@ -862,7 +862,10 @@
             "type": "compile",
             "test": {
                 "include": [ "EGL/egl.h", "bcm_host.h" ],
-                "main": "vc_dispmanx_display_open(0);"
+                "main": [
+                    "vc_dispmanx_display_open(0);",
+                    "EGL_DISPMANX_WINDOW_T *eglWindow = new EGL_DISPMANX_WINDOW_T;"
+                ]
             },
             "use": "egl bcm_host"
         },
```

abhiTronix, 2022-03-31:
@WalterPrech Thanks that will be helpful for others.

closed 2022-03-31 by abhiTronix