abhiTronix/raspberry-pi-cross-compilers #102: Broken adress_sanitizer in gcc10.3 (+durty patch)

sleeperss, 2022-04-29:
Hi,

When cross-compilling with the address sanitizer I get the following error at link time :
```/opt/cross/tools/bin/../lib/gcc/aarch64-linux-gnu/10.3.0/../../../../aarch64-linux-gnu/bin/ld: program_name: hidden symbol `__aarch64_ldadd8_acq_rel' in /opt/cross/tools/bin/../lib/gcc/aarch64-linux-gnu/10.3.0/libgcc.a(ldadd_8_4.o) is referenced by DSO```

It seems to be an error from gcc-10 as mentionned in https://lists.freebsd.org/archives/freebsd-ports/2021-July/000431.html

I got it to work by replacing gcc-10.3.0 with gcc-10.3.1 and (I don't know why, possibly a libc dependency problem) removing the lines 1642 to 1649 of c++/10.3.1/aarch64-linux-gnu/bits/c++config.h.

Without the last modification I get multiple error like : 
```error: there are no arguments to 'pthread_cond_clockwait' that depend on a template parameter
error: there are no arguments to 'pthread_rwlock_clockrdlock' that depend on a template parameter
error: 'pthread_mutex_clocklock' was not declared in this scope; did you mean 'pthread_mutex_unlock'```

abhiTronix, 2022-04-30:
@sleeperss Thanks I'll look into this and make necessary changes.

abhiTronix, 2024-08-31:
@sleeperss This issue is no longer valid as we moved to newer patched version.

closed 2024-08-31 by abhiTronix