abhiTronix/raspberry-pi-cross-compilers #105: cannot find crt1.o: No such file or directory

Margelator, 2022-05-11:
> /home/dfyn/rpi-qt/tools/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crt1.o: No such file or directory
> /home/dfyn/rpi-qt/tools/cross-pi-gcc-10.2.0-2/bin/../lib/gcc/arm-linux-gnueabihf/10.2.0/../../../../arm-linux-gnueabihf/bin/ld: cannot find crti.o: No such file or directory
> collect2: error: ld returned 1 exit status
> make: *** [Makefile:68: verifyspec] Error 1
Note: Also available for Linux: linux-clang linux-icc

ERROR: Cannot compile a minimal program. The toolchain or QMakeSpec is broken.

abhiTronix, 2022-05-11:
Did you checked if all the folders rysnced properly before compiling? Also, did you make any changes to instructions given in our guide? And finally what's your environment(Host OS, Target OS, Host machine, Target machine)?

Margelator, 2022-05-11:
Yes,i checked. 
no change.
Host OS Ubuntu20.04

Margelator, 2022-05-11:
Yes,i checked.
no change.
Host OS Ubuntu20.04
Target OS Raspberry Pi OS with desktop bullseye

abhiTronix, 2022-05-11:
Target machine?

Margelator, 2022-05-11:
Yes,i checked.
no change.
Host OS:Ubuntu 20.04.4 LTS
Target OS:Raspberry Pi OS with desktop bullseye.
Host machine: x86_64 Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz
Target machine: Raspberry Pi 4B

abhiTronix, 2022-05-11:
Looks like sysroot issue. Can you paste the URL or Name of Toolchain you downloaded from sourceforge for compiling?

Margelator, 2022-05-11:
OK.
https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Bullseye/GCC%2010.2.0/Raspberry%20Pi%203A%2B%2C%203B%2B%2C%204/cross-gcc-10.2.0-pi_3%2B.tar.gz

abhiTronix, 2022-05-11:
You setup is fine, but sysroot is problematic. You either missed a step or sysroot isn't properly configured. Can you go to rootfs folder and run this command:

```sh
tree -H ./ > result.htm
````

Then paste result here. Aslo paste the content of PI. cmake file.

Margelator, 2022-05-11:
OK.
sysroot_result.html url is here:  https://t.wss.ink/f/8acc8co2ey5
home_user_result.html url is here:  https://t.wss.ink/f/8accrd68tzh 
sorry,i don't know PI.cmake

abhiTronix, 2022-05-11:
> sorry,i don't know PI.cmake

Aren't you following this guide? https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os

Margelator, 2022-05-11:
No, I only follow this guide: 
https://github.com/abhiTronix/raspberry-pi-cross-compilers/blob/master/QT_build_instructions.md

abhiTronix, 2022-05-11:
Then this issue is cannot be resolved currently, as QT is broken for new bullseye OS since FAKE KMS is no longer available for this OS, See #101 #88. But you can try bare-minimum example given in https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#cross-compiler-cmake-usage-guide-with-rsynced-raspberry-pi-32-bit-os to check if Toolchain works or not.

Margelator, 2022-05-11:
OK, Thank you very much for your reply.

abhiTronix, 2022-05-11:
I'm closing this now, and will be resolved with #101 and other related issues.

closed 2022-05-11 by abhiTronix