abhiTronix/raspberry-pi-cross-compilers #106: cmake can't find RT library

kingsimba, 2022-05-23:
I tried to install GCC 10 on 64 bit raspberry-pi(Debian Buster). And followed this instruction:

https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/64-Bit-Native-Compiler:-Installation-Instructions

I can compile a simple "hello_world.cpp". But ROS `catkin_make` and encountered the following error:

```
CMake Error at /opt/ros/noetic/share/catkin/cmake/assert.cmake:17 (message):

  Assertion failed: check for file existence, but filename
  (RT_LIBRARY-NOTFOUND) unset.  Message: RT Library

Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/tools/rt.cmake:44 (assert_file_exists)
  /opt/ros/noetic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:58 (find_package)
```

The cmake file contains:

```
    find_library(RT_LIBRARY rt)
    assert_file_exists(${RT_LIBRARY} "RT Library")
```

I guess some `*.cmake` files are not linked correctly. But I don't know how to fix it.

abhiTronix, 2022-05-23:
@kingsimba Are you sure this is Toolchain error? Check again: https://answers.ros.org/question/297453/rt_library-notfound-unset-message-rt-library/

closed 2022-05-23 by abhiTronix

kingsimba, 2022-05-23:
I tried to add some message in cmake.

```
message(STATUS "RT_LIBRARY: ${RT_LIBRARY}")
```

Before installing gcc. It shows:

```
RT_LIBRARY: /usr/lib/aarch64-linux-gnu/librt.so
```

After installing. It shows:

```
RT_LIBRARY: RT_LIBRARY-NOTFOUND
```

kingsimba, 2022-05-23:
I think I know what's going wrong. `librt.so`, `libthread.so`(and many more) should also be linked.

kingsimba, 2022-05-23:
By executing the following additional commands. `catkin_make` works as usual.

```
sudo ln -s /usr/lib/aarch64-linux-gnu/*.so  /usr/lib
sudo ln -s /usr/lib/aarch64-linux-gnu/cmake/*  /usr/lib/cmake
```