abhiTronix/raspberry-pi-cross-compilers #115: Raspberry Pi 4 B - Configuration and Build Issue

tlawrence2586, 2023-07-25:
I recently got a raspberry pi 4 B and want to install qt on it. Your steps are very clear and helpful but I come across an issue configuring. I am using wsl ubuntu 22.04.2 LTS on host machine. Raspbian Linux 11 bullseye on rpi4B with 32 bit OS image. Using 
cross compilin

QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    EGLFS EGLDevice ...................... yes
    EGLFS GBM ............................ yes
    EGLFS VSP2 ........................... no
    EGLFS Mali ........................... no
    **EGLFS Raspberry Pi ................... yes**
    EGLFS X11 ............................ yes
  LinuxFB ................................ yes
  VNC .................................... yes
  XCB:
    Using system-provided xcb-xinput ..... yes
    Native painting (experimental) ....... no
    GL integrations:
      GLX Plugin ......................... no
      EGL-X11 Plugin ..................... yes
      
 I see that EGLFS Raspberry Pi gets set to yes which is different from yours. Building the next step results in errors such as
      
     /rpi-qt/qt-everywhere-src-5.15.2/qtbase/src/plugins/platforms/eglfs/deviceintegration/eglfs_brcm/qeglfsbrcmintegration.cpp:74:5: 
    error: ‘EGL_DISPMANX_WINDOW_T’ was not declared in this scope
    74 |     EGL_DISPMANX_WINDOW_T *eglWindow = new EGL_DISPMANX_WINDOW_T;
      |     ^~~~~~~~~~~~~~~~~~~~~
      
      
  Any help would be appreciated, ive been spinning my wheels on this one. Thanks again.

tlawrence2586, 2023-07-25:
^ with cross compiler gcc 10.2.0-pi_3+

RuleNumber1, 2024-04-05:
i also met this problem,i`m going to use raspbian buster 32-bit OS instead of bullseye 32-bit