abhiTronix/raspberry-pi-cross-compilers #116: Raspberry Pi GCC Cross-Compiler Toolchains (Bookworm)

distlibs, 2023-10-15:
When GCC cross-compiler toolchains will be available for Raspberry Pi OS Bookworm?

cnbaker, 2023-11-17:
Yes please!

Givy, 2023-12-30:
Hi,
I tried to install the toolchain on RPi-3 64b;;
Linux RaXXXX 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 **aarch64 GNU/Linux**

It took a long time, but it finished with the following Error/s.  Any logs I need to look into?  Anything I am missing etc.
Also, can I install this on a RPi- 64b, bookworm?;;
Linux MiXXX 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) **aarch64 GNU/Linux**
__________________________________________________________________________________________________
/opt/raspberry-pi-cross-compilers/build-scripts# **./RTBuilder_64b -g 11.2.0 -o bullseye -V**
.
.
/tmp/gcc-10.2.0/gcc-10.2.0/gcc/gengtype-lex.l: In function ‘int yylex(const char**)’:
gengtype-lex.c:356:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
/tmp/gcc-10.2.0/gcc-10.2.0/gcc/gengtype-lex.l:119:1: note: in expansion of macro ‘YY_DO_BEFORE_ACTION’
/tmp/gcc-10.2.0/gcc-10.2.0/gcc/gengtype-lex.l:114:1: note: here
gengtype-lex.c:356:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
/tmp/gcc-10.2.0/gcc-10.2.0/gcc/gengtype-lex.l:136:1: note: in expansion of macro ‘YY_DO_BEFORE_ACTION’
/tmp/gcc-10.2.0/gcc-10.2.0/gcc/gengtype-lex.l:132:1: note: here
**_collect2: fatal error: ld terminated with signal 9 [Killed]_**
**_compilation terminated._**
### **make[1]: *** [../../gcc/c/Make-lang.in:85: cc1] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:4733: all-gcc] Error 2**
root@RaXXX:/opt/raspberry-pi-cross-compilers/build-scripts#

Thank you.

Givy, 2023-12-30:
If I stop here, is there anything to remove or delete or worry about?  If, then I'll go with the qt-cross-compile.

Thanks so much.

GK-Retech, 2024-05-31:
Has anyone able to cross compile for raspberry pi 5

abhiTronix, 2024-08-31:
@henrihallik @cnbaker @Givy @distlibs Support for both Bookworm OS and Raspberry Pi 5 is now available with `v3.1.0`

closed 2024-08-31 by abhiTronix