abhiTronix/raspberry-pi-cross-compilers #118: Sysroot of this toolchain is incompatible to sysroot in official Raspbian images

rweickelt, 2024-03-19:
I am trying to cross-build a complex project with CMake and VCPKG and a sysroot configuration. My sysroot is extracted from the official Raspbian RFS image.

Neither compilation nor linking works properly which is caused by a file layout deviation:

- Raspbian images follow the multi-arch layout where includes are found under `/usr/include/arm-linux-gnueabihf` and libs under `/usr/lib/arm-linux-gnueabihf`
- The sysroot delivered with these toolchains here does not have such an architecture subfolder, but puts everything directly under `/usr/include` or `/usr/lib` respectively.

Example: 

- Raspbian image: `/usr/include/arm-linux-gnueabihf/sys/cdefs.h`
- This toolchain: `/usr/include/sys/cfefs.h`

I cannot work around by simply passing additional include paths and rpaths to the compiler/linker driver. That is because VCPKG swallows additional include paths outside of the vcpkg build tree but passes `CMAKE_SYSROOT`.

I would expect the sysroot of this toolchain to match Rapsbians RFS 100%, except that it contains less. The [cross toolchain for Windows](https://gnutoolchains.com/raspberry/) actually does that.

rweickelt, 2024-03-19:
Actually, I think the problem is a missing `--enable-multiarch` in this toolchain's configuration.

abhiTronix, 2024-09-10:
@rweickelt No need to do anything, since you could easily make [symlinks](https://www.freecodecamp.org/news/symlink-tutorial-in-linux-how-to-create-and-remove-a-symbolic-link/) in Linux envs. Follow this step: https://github.com/abhiTronix/raspberry-pi-cross-compilers/wiki/Cross-Compiler-CMake-Usage-Guide-with-rsynced-Raspberry-Pi-32-bit-OS#8-setup-important-symlinks-as-follows-important

closed 2024-09-10 by abhiTronix